freebsd/drm-kmod #241: 5.15: Screenflickering on desktop

Alexander88207, 2023-03-11:
Hello,

when i do using only the desktop for browsing, videos etc.. then i do get a flickering, i can not describe it full looks for me more as a little fast flash or something.

Important to know (maybe): This doesnt happen when i do play a game.

FreeBSD Home 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n261233-005cca8361a4: Thu Mar  2 08:07:29 UTC 2023     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64

**PCI Info**

<details>
vgapci0@pci0:3:0:0:     class=0x030000 rev=0xc5 hdr=0x00 vendor=0x1002 device=0x73df subvendor=0x1043 subdevice=0x05cb
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT]'
    class      = display
    subclass   = VGA

</details>

**DRM KMOD version**
drm-515-kmod 5.15.25

**To Reproduce**

Just do normal desktop acitivites.

Note: I do have already tried an other cable, to make sure its not the cable.

evadot, 2023-03-29:
Does that happens with drm-510-kmod ?
This is typically the kind of issues hard to solve ...

Alexander88207, 2023-03-29:
Could be an refresh rate issue?

I do have just change to 60 from 240hz and it doesn't happen at all.

dumbbell, 2023-03-29:
> Its an Radeon 6700 XT, maybe could be something wrong on my end since this card was a also a test thing during porting from @dumbbell ?

Indeed, I'm using it on a daily basis at work/home without any problems.

dumbbell, 2023-03-29:
> Could be an refresh rate issue?
> 
> I do have just changed to 60 from 240hz and no flicker so far.

Do you mean that it works only when you set the refresh rate to 60 Hz?

Alexander88207, 2023-03-29:
Well when i do set to 60hz i get no flickering. This could explain while gaming it doesn't happen. Because i do use FreeSync/ VRR while gaming.

Alexander88207, 2023-03-29:
I think I have to apologize, I now have a 1080p 240hz connected (instead of 1440p) and there I now also no longer flickering.

I do close this issue for now and will reply if i still see anything in a long run etc..

ekhramtsov, 2023-04-21:
I had this with 5600G (1002:1638, Green Sardine) and 3840x2160 output in sway,
but with https://reviews.freebsd.org/D39628 (not yet in main) it was fixed here.

ltning, 2024-01-08:
I had the same problem with nvidia on an Apple 6k Studio Display. Was working fine for a long time, then suddenly started glitching and I could never figure out why. Am on AMD GPU now, and all is fine.