freebsd/drm-kmod #242: Unable to build drm-515-kmod (5.15.25) for aarch64 for lack of sufficient #include <linux/preempt.h> use

markmi, 2023-03-14:
**Describe the bug**
drivers/gpu/drm/drm_cache.c:362:10: error: call to undeclared function 'in_interrupt'
(Note: linux/preempt.h provides in_interrupt when included )

**FreeBSD version**
=>> Building graphics/drm-515-kmod
build started at Tue Mar  7 11:27:15 UTC 2023
port directory: /usr/ports/graphics/drm-515-kmod
package name: drm-515-kmod-5.15.25
building for: FreeBSD main-arm64-default-job-04 14.0-CURRENT FreeBSD 14.0-CURRENT 1400081 arm64
maintained by: x11@FreeBSD.org
Makefile ident: 
Poudriere version: 3.2.8-23-ga7f8d188
Host OSVERSION: 1400073
Jail OSVERSION: 1400081

(I'm referencing the official package-builder failure. My private build attempt got the same but the official build attempt is a better context to reference, avoiding questions about what might be different in my personal context.)

**PCI Info**
NA: Build time issue

**DRM KMOD version**
package name: drm-515-kmod-5.15.25

**To Reproduce**
Have poudriere(-devel) try to build graphics/drm-515-kmod on/for 14.0-CURRENT.

**Screenshots**
NA

**Additional context**
The official package-builder failure information was extracted from:
http://ampere2.nyi.freebsd.org/data/main-arm64-default/p64e3eb722c17_s7fc82fd1f8/logs/errors/drm-515-kmod-5.15.25.log

I do not know if completing the #include <linux/preempt.h> usage would be sufficient to allow the build vs. if there is some next issue also present.

I'll note that I was attempting to set up an arm64 context somewhat analogous to an amd64 FreeBSD bug report that seems to involve amdgpu having been loaded. (I've a serial console context for arm64 that I do not have for amd64 --and have other reasons to avoid messing with the amd64 context that I've access to.) I do not normally use any drm-*-kmod . I was just going to see if I got some replication(s) analogous to the amd64 bug report vs. not.

markmi, 2023-03-17:
Looks like there are later issues such as `drivers/gpu/drm/amd/amdgpu/amdgpu_ioc32.c:          return drm_compat_ioctl(filp, cmd, arg);` being in use with drm_compat_ioctl defined as NULL : `drivers/gpu/drm/amd/amdgpu/amdgpu_ioc32.c:43:26: error: called object type 'void *' is not a function or function pointer` . It suggests that far more is just  not present yet for aarch64 (from what I can tell).

So it appears my idea for trying to test reproducibility on aarch64 of what is reported on amd64 is just premature at this point.

It is not clear to me if this should be left open vs. closed (as aarch64-not-started-yet?).

chmeeedalf, 2025-02-20:
amdgpu_kms_compat_ioctl() is only used when CONFIG_COMPAT is defined.  Since it's not defined, drm_compat_ioctl is defined as NULL.  So the amdgpu_ioc32.c file should really only get built if CONFIG_COMPAT is defined (what defines it?)  If FreeBSD doesn't define it, just don't build the file.