freebsd/drm-kmod #244: drm-515-kmod: Loss of display signal with RX 6800 XT

old-unix-geek, 2023-03-25:
**Describe the bug**
Display loses signal after loading amdgpu via kldload under FreeBSD-CURRENT with RX 6800 XT with sienna_cichlid firmware installed.
The screen goes completely black and the on-screen dialog shows that it's lost signal from the PC.

**FreeBSD version**
```
FreeBSD zen.unixgeek.dev 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n261711-b5d43972e394: Thu Mar 23 08:12:31 UTC 2023     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
```
**PCI Info**
```
hostb0@pci0:0:0:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1450 subvendor=0x1462 subdevice=0x7b78
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:       class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1451 subvendor=0x1022 subdevice=0x1451
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) I/O Memory Management Unit'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib9@pci0:0:3:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:7:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib12@pci0:0:7:1:      class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1454 subvendor=0x1022 subdevice=0x1454
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B'
    class      = bridge
    subclass   = PCI-PCI
hostb6@pci0:0:8:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
[... 225 lines cut ...]
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_pfp.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_me.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_ce.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_rlc.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_mec.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_mec2.bin'
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_sdma.bin'
Mar 25 00:07:51 zen kernel: [drm] use_doorbell being set to: [true]
Mar 25 00:07:51 zen syslogd: last message repeated 3 times
Mar 25 00:07:51 zen kernel: drmn0: successfully loaded firmware image 'amdgpu/sienna_cichlid_vcn.bin'
Mar 25 00:07:51 zen kernel: [drm] Found VCN firmware Version ENC: 1.26 DEC: 2 VEP: 0 Revision: 0
Mar 25 00:07:51 zen kernel: drmn0: Will use PSP to load VCN firmware
Mar 25 00:07:51 zen kernel: [drm ERROR :psp_hw_start] PSP load sos failed!
Mar 25 00:07:51 zen kernel: [drm ERROR :psp_hw_init] PSP firmware loading failed
Mar 25 00:07:51 zen kernel: [drm ERROR :amdgpu_device_fw_loading] hw_init of IP block <psp> failed -22
Mar 25 00:07:51 zen kernel: drmn0: amdgpu_device_ip_init failed
Mar 25 00:07:51 zen kernel: drmn0: Fatal error during GPU init
Mar 25 00:07:51 zen kernel: drmn0: amdgpu: finishing device.
Mar 25 00:07:51 zen kernel: device_attach: drmn0 attach returned 22
```

**Additional Comments/Questions**

I've tried with single and multiple displays attached via HDMI and Display Port, Display Port only, and HDMI only. No changes.
What other information can help narrow the issue futher?

evadot, 2023-03-25:
Looks like you might need more firmwares.
Some AMD GPU needs firmware from other generation (usually for video decoding etc ...), can you test installing the meta package gpu-firmware-kmod (which will install all the firmwares) and try again ?
If that works please paste again dmesg as it will contain the name of the firmware loaded, I'm working on a script that will automatically install firmware packages based on the PCI id of the video card so I will need to patch it.

old-unix-geek, 2023-03-25:
I have installed the meta-package (gpu-firmware-kmod) which did install all firmware files in /boot/modules, but it did not change the behavior.

old-unix-geek, 2023-07-01:
Is there something else I can try?

phryk, 2023-09-24:
Thanks @evadot – I had a similar-ish problem where disconnecting a display would result in an empty TTY being displayed on the remaining displays.

After reading your comment I realized I only specifically installed the firmware for my RX 6600XT and installing the firmware metapackage fixed my particular issue. :)

old-unix-geek, 2023-10-27:
The same behavior occurs with FreeBSD 14.0-RC2 with drm-515-mod-5.15.118 and only drm-mod metapackage installed with my RX 6800 XT.

thedaemon, 2023-10-28:
I am getting loss of display and hard lock ups with drm-515-kmod-5.15.118 on my 6750XT, I even had a hard reset happen.

bonspry, 2024-08-04:
same issue on freebsd 14.1 on rx 6800

uber-perfect-cell, 2025-02-06:
same issue on freebsd 14.2 on rx6600

bonspry, 2025-02-08:
I fixed it by replacing my 2nd gen Ryzen CPU with 5th gen, possibly PCIE mismatch stuff