freebsd/drm-kmod #246: drm-kmod-515lts Hangs immediately when running vk benchmarks

adventureloop, 2023-04-05:
**Describe the bug**
Running drm-kmod from the 5.15-lts on FreeBSD -CURRENT I am able to run X and OpenGL applications/benchmarks and playback video in firefox, but I get an immediate hang if I run vkmark or vkcube-xcb.

**FreeBSD version**
`FreeBSD computer 14.0-CURRENT FreeBSD 14.0-CURRENT #2 main-n261806-d3a49f62a284: Mon Mar 27 16:15:25 UTC 2023     user@computer:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64`

**PCI Info**

```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x4621 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x46a6 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Integrated Graphics Controller'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x461d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake Innovation Platform Framework Processor Participant'
    class      = dasp
pcib1@pci0:0:6:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x464d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x4 Controller'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x466e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:7:1:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x463f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:7:2:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x462f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:7:3:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x461f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:8:0:	class=0x088000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x464f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor Gaussian & Neural Accelerator'
    class      = base peripheral
none2@pci0:0:10:0:	class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x467d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Platform Monitoring Technology'
    class      = dasp
xhci0@pci0:0:13:0:	class=0x0c0330 rev=0x02 hdr=0x00 vendor=0x8086 device=0x461e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 USB Controller'
    class      = serial bus
    subclass   = USB
none3@pci0:0:13:2:	class=0x0c0340 rev=0x02 hdr=0x00 vendor=0x8086 device=0x463e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
none4@pci0:0:13:3:	class=0x0c0340 rev=0x02 hdr=0x00 vendor=0x8086 device=0x466d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
xhci1@pci0:0:20:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ed subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH USB 3.2 xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none5@pci0:0:20:2:	class=0x050000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ef subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Shared SRAM'
    class      = memory
    subclass   = RAM
ig4iic0@pci0:0:21:0:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e8 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
ig4iic1@pci0:0:21:1:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e9 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
ig4iic2@pci0:0:21:3:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51eb subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
none6@pci0:0:22:0:	class=0x078000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e0 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH HECI Controller'
    class      = simple comms
none7@pci0:0:22:3:	class=0x070002 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e3 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake AMT SOL Redirection'
    class      = simple comms
    subclass   = UART
pcib6@pci0:0:29:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x51b0 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x5182 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:	class=0x040380 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51c8 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a3 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P SMBus Host Controller'
    class      = serial bus
    subclass   = SMBus
none8@pci0:0:31:5:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a4 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH SPI Controller'
    class      = serial bus
nvme0@pci0:1:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x15b7 device=0x5006 subvendor=0x15b7 subdevice=0x5006
    vendor     = 'Sandisk Corp'
    device     = 'WD Black SN750 / PC SN730 NVMe SSD'
    class      = mass storage
    subclass   = NVM
iwlwifi0@pci0:166:0:0:	class=0x028000 rev=0x1a hdr=0x00 vendor=0x8086 device=0x2725 subvendor=0x8086 subdevice=0x0024
    vendor     = 'Intel Corporation'
    device     = 'Wi-Fi 6 AX210/AX211/AX411 160MHz'
    class      = network

```

**DRM KMOD version**

`commit 4a74fc2b771b4bb5c0f297656a4545ce485b96bd`

**To Reproduce**

From an X session launch vkmark or vkcube-xcb, A frame or two will render and then the UI will lock up

adventureloop, 2023-04-05:
Re-testing after creating the issue I was able to run vkcube for a while, vkmark hung once it transitioned to the cube.

evadot, 2024-01-05:
Can you test again with 6.1-lts ? A lot of commits adds better supports for Alderlake.

adventureloop, 2024-01-07:
I am unable to load 6.1-lts I get:

```
link_elf_obj: symbol ___drm_dbg undefined
linker_load_file: ./i915/i915kms.ko - unsupported file type
```

However, the current drm-kmod in ports (I'm not sure how to determine what version this is) was able to run the full vkmark suite.

```
FreeBSD host 15.0-CURRENT FreeBSD 15.0-CURRENT #8 main-n267429-30409ecdb648: Sat Jan  6 20:53:43 GMT 2024     user@host:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
```

I think that resolves this issue. I can suspend and resume, but quickly after resume I get a hang.

closed 2024-01-07 by adventureloop