freebsd/drm-kmod #247: glamor causes stuttering on a specific KDE UI component - Kirigami ScrollBar

arrowd, 2023-04-11:
**Describe the bug**

1. Install the minimal KDE software required to reproduce the issue: `pkg install plasma5-kde-cli-tools plasma5-plasma-desktop`.
2. Run `kcmshell5 kcm_plasmasearch`.
3. Maximize the window for better reproducibility.
4. Move mouse cursor to the rightmost scrollbar.
5. Click and hold the scrollbar handle.
6. Without releasing the button move the cursor to the bottom of the scrollbar lane.

The scrolling animation is extremely stuttering. It takes only 2-3 frames for handle to reach the bottom. On some of our machines it is even worse - if you move the handle slowly but steady (without micro-stops) the window content isn't updated at all until you either release the button or stop the movement.

This behavior reproduces on `14-CURRENT`, `i915kms` and `drm-515-kmod` as well as `13.2-RC6`, `amdgpu` and `drm-510-kmod`.

Now, starting X server with the following config makes the problem go away:

```
Section "Device"
  Driver     "modesetting"
  Identifier "asdasd"
  Option "AccelMethod" "none"
EndSection
```

**FreeBSD version**
```
FreeBSD sbreeze 14.0-CURRENT FreeBSD 14.0-CURRENT #21 main-n261572-3451ed888e75: Tue Mar 21 20:33:52 +03 2023
root@sbreeze:/usr/obj/usr/src/amd64.amd64/sys/AMENITY amd64
```

**PCI Info**

<details>
  <summary>pciconf -lv</summary>

```
hostb0@pci0:0:0:0:      class=0x060000 rev=0x05 hdr=0x00 vendor=0x8086 device=0x5910 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:       class=0x060400 rev=0x05 hdr=0x01 vendor=0x8086 device=0x1901 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '6th-10th Gen Core Processor PCIe Controller (x16)'
    class      = bridge
    subclass   = PCI-PCI
vgapci1@pci0:0:2:0:     class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x591b subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics 630'
    class      = display
    subclass   = VGA
none0@pci0:0:8:0:       class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa12f subvendor=0x1043 subdevice=0x201f
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:  class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa131 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Thermal Subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:    class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa160 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:      class=0x078000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa13a subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family MEI Controller'
    class      = simple comms
ahci0@pci0:0:23:0:      class=0x010601 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa103 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'HM170/QM170 Chipset SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
pcib2@pci0:0:28:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa112 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:3:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa113 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:6:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa116 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa152 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'HM175 Chipset LPC/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none2@pci0:0:31:2:      class=0x058000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa121 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Power Management Controller'
    class      = memory
hdac0@pci0:0:31:3:      class=0x040380 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa171 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = 'CM238 HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa123 subvendor=0x1043 subdevice=0x15e0
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family SMBus'
    class      = serial bus
    subclass   = SMBus
vgapci0@pci0:1:0:0:     class=0x030200 rev=0xa1 hdr=0x00 vendor=0x10de device=0x1c8c subvendor=0x1043 subdevice=0x15e0
    vendor     = 'NVIDIA Corporation'
    device     = 'GP107M [GeForce GTX 1050 Ti Mobile]'
    class      = display
    subclass   = 3D
iwm0@pci0:2:0:0:        class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095a subvendor=0x8086 subdevice=0x5010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 7265'
    class      = network
re0@pci0:3:0:0: class=0x020000 rev=0x15 hdr=0x00 vendor=0x10ec device=0x8168 subvendor=0x1043 subdevice=0x200f
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller'
    class      = network
    subclass   = ethernet
rtsx0@pci0:4:0:0:       class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x5229 subvendor=0x1043 subdevice=0x202f
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS5229 PCI Express Card Reader'
```
</details>

**DRM KMOD version**
`drm-515-kmod 5.15.25_2`

**Screenshots**
Another window that has this problem is `kcmshell5 kcm_kwin_effects`. [Here is a video](https://arrowd.name/glamor_qml.mp4) of the debugging session for that window. We added a textbox that reacts on mouse event and displays its current position. It shows that stuttering is happening even without mouse clicks, just hovering over the scrollbar is enough. Note the cursor movements on 0:06-0:10 - this is the same freeze that prevents scrolling animation if the button is pressed.

**Additional context**
I have reports that it also works fine with NVIDIA proprietary driver.

arrowd, 2023-04-19:
I'll be happy do research/debug this on my own, but I need some starting point at least.

evadot, 2023-05-04:
I can't seem to reproduce this here on a valleyview machine with 5.10 (5.15 doesn't work yet on this one), will try on some other systems.

arrowd, 2023-05-23:
Hi @evadot , thanks for checking this out. Were you able to conduct testing on 5.15?

evadot, 2023-05-25:
Not yet, will try to find time.

kblc, 2023-08-09:
Hi there! ) any update?

arrowd, 2025-01-20:
Doesn't seem to be reproducible anymore.

closed 2025-01-20 by arrowd