freebsd/drm-kmod #249: Cannot get gpu acceleration on intel pentium N3510 using drm-510-kmod

monwarez, 2023-04-29:
**Describe the bug**
When attempting to launch a wayland compositor (Sway, Hyprland), the process abort at the crocus_dri.so level.
I am unable to get the crocus_dri built without stripping (I put STRIP_CMD to empty string, but still the lib is stripped). But here
an approximation of the back trace:
```
# 0 -> thr_kill at libc
# 1 -> raise at libc
# 2 -> abort at libc
# 3 -> at dri/crocus_dri.so
...
# 10 -> at libEGL_mesa.so
...
# 12 eglMakeCurrent at libEGL.so
# 13 wlr_egl_unset_current at libwlroots.so
# 14 wlr_gles2_renderer_create at libwlroots.so
# 15 wlr_gles2_renderer_create_with_drm_fd at libwlroots.so
```
**FreeBSD version**

FreeBSD redacted 13.2-RELEASE FreeBSD 13.2-RELEASE releng/13.2-n254617-525ecfdad597 GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f00 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f31 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx Series Graphics & Display'
    class      = display
    subclass   = VGA
ahci0@pci0:0:19:0:	class=0x010601 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f23 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor E3800 Series SATA AHCI Controller'
    class      = mass storage
    subclass   = SATA
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f35 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx, Celeron N2000 Series USB xHCI'
    class      = serial bus
    subclass   = USB
none0@pci0:0:26:0:	class=0x108000 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f18 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine'
    class      = encrypt/decrypt
hdac0@pci0:0:27:0:	class=0x040300 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f04 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx Series High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib1@pci0:0:28:0:	class=0x060400 rev=0x0a hdr=0x01 vendor=0x8086 device=0x0f48 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor E3800 Series PCI Express Root Port 1'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:1:	class=0x060400 rev=0x0a hdr=0x01 vendor=0x8086 device=0x0f4a subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor E3800 Series PCI Express Root Port 2'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f1c subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor Z36xxx/Z37xxx Series Power Control Unit'
    class      = bridge
    subclass   = PCI-ISA
ichsmb0@pci0:0:31:3:	class=0x0c0500 rev=0x0a hdr=0x00 vendor=0x8086 device=0x0f12 subvendor=0x103c subdevice=0x2188
    vendor     = 'Intel Corporation'
    device     = 'Atom Processor E3800/CE2700 Series SMBus Controller'
    class      = serial bus
    subclass   = SMBus
rtsx0@pci0:1:0:0:	class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x5227 subvendor=0x103c subdevice=0x2188
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS5227 PCI Express Card Reader'
iwm0@pci0:2:0:0:	class=0x028000 rev=0x73 hdr=0x00 vendor=0x8086 device=0x08b1 subvendor=0x8086 subdevice=0x4062
    vendor     = 'Intel Corporation'
    device     = 'Wireless 7260'
    class      = network

</details>

**DRM KMOD version**
drm-510-kmod 5.10.163_5

**To Reproduce**
Try to launch sway with an intel pentium N3510.
It will abort.

**Additional context**
When I was still on 13.1-RELEASE I could still use the removed drm-54-kmod, but since the upgrade it does not work.

evadot, 2023-04-29:
Yeah it's valleyview, I have the same problem and I still can't understand what's going on.
Hopefully I'll have some idea/fixes in the next few weeks.

evadot, 2023-05-04:
So, the offending commit is e1ca81bb6675d271841da91181305ccd70da4c1e, I'm still trying to make sense and also trying to find how to properly fix this in the master and 5.15-lts branch (code as changed a bit) but if in the meantime you could test reverting this commit to confirm that this fixes the issue for you that would be great.

monwarez, 2023-05-04:
I am unable to find this commit, what was the commit message ?

evadot, 2023-05-04:
> I am unable to find this commit, what was the commit message ?

Sorry, bad paste : https://github.com/freebsd/drm-kmod/commit/7bc2545f04fb7f76e3d03148eb55922babcec9ca

monwarez, 2023-05-04:
Thanks, I will test it latter this day.

commit e64430ea (2023-05-12): Revert "drm/i915/gt: Restore clear-residual mitigations for Ivybridge, Baytrail"

This reverts commit 7bc2545f04fb7f76e3d03148eb55922babcec9ca.

This causes problems on Baytrail/Valleyview hardware.
Closes #249
closed 2023-05-12 by evadot