freebsd/drm-kmod #252: Loadig firmware on alder lake freezes system

rkoberman, 2023-05-14:
**Describe the bug**
Loading drm-515-kmod with gpu-firmware-intel-kmod installed causes in instant freeze with no output or log messages. The system cannot be pinged and only a power cycle and reboot can get any response.  The port must be deleted to bring hte system up. Freeze occurs whether the driver is loaded during startup (kld_list or manual load after boot completion (no graphics startup.)

**FreeBSD version**
FreeBSD ptavv 14.0-CURRENT FreeBSD 14.0-CURRENT #12 main-n262908-c16e08e5f324-dirty: Thu May 11 17:01:29 PDT 2023     root@ptavv:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
**PCI Info**

<details>
  <summary>pciconf -lv</summary
  
Paste the output of `pciconf -vl`
hostb0@pci0:0:0:0:	class=0x060000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x4601 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x46a8 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-UP3 GT2 [Iris Xe Graphics]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x461d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake Innovation Platform Framework Processor Participant'
    class      = dasp
pcib1@pci0:0:6:0:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x464d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x4 Controller'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x466e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:7:2:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x462f subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:10:0:	class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x467d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Platform Monitoring Technology'
    class      = dasp
xhci0@pci0:0:13:0:	class=0x0c0330 rev=0x04 hdr=0x00 vendor=0x8086 device=0x461e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 USB Controller'
    class      = serial bus
    subclass   = USB
none2@pci0:0:13:2:	class=0x0c0340 rev=0x04 hdr=0x00 vendor=0x8086 device=0x463e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
none3@pci0:0:13:3:	class=0x0c0340 rev=0x04 hdr=0x00 vendor=0x8086 device=0x466d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
xhci1@pci0:0:20:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ed subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH USB 3.2 xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none4@pci0:0:20:2:	class=0x050000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ef subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Shared SRAM'
    class      = memory
    subclass   = RAM
iwlwifi0@pci0:0:20:3:	class=0x028000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51f0 subvendor=0x8086 subdevice=0x0090
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH CNVi WiFi'
    class      = network
none5@pci0:0:22:0:	class=0x078000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e0 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH HECI Controller'
    class      = simple comms
isab0@pci0:0:31:0:	class=0x060100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x5182 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:	class=0x040380 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51c8 subvendor=0x17aa subdevice=0x22e9
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a3 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P SMBus Host Controller'
    class      = serial bus
    subclass   = SMBus
none6@pci0:0:31:5:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a4 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH SPI Controller'
    class      = serial bus
em0@pci0:0:31:6:	class=0x020000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x1a1f subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (16) I219-V'
    class      = network
    subclass   = ethernet
nvme0@pci0:2:0:0:	class=0x010802 rev=0x01 hdr=0x00 vendor=0xc0a9 device=0x540a subvendor=0xc0a9 subdevice=0x5021
    vendor     = 'Micron/Crucial Technology'
    device     = 'P2 NVMe PCIe SSD'
    class      = mass storage
    subclass   = NVM

</details>

**DRM KMOD version**
drm-515-kmod 5.15.25_3
**To Reproduce**
Steps to reproduce the behavior:
Load i915kms.ko

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

decke, 2023-09-13:
Same here with 14.0-BETA1 on similar hardware. Any idea how to debug this?

rkoberman, 2023-09-13:
I will mention that the interface will work with drm-515-kmod if you
deinstall the gpu-firmware* package/port. The graphics will "work" fine,
but no hardware acceleration. Everything is software rendered, I can view
4K video with no issues, but the CPU is pretty busy. I note that Intel has
released several updates to the firmware package since the February 10
version, but I have not tried making one work, so far.

I can't think of any way to debug as it simply deadlocks the system with no
information. I don't know of a process to debug firmware load, but I assume
there is one and I'd be VERY happy to help if I had any clue how to proceed.
-- 
Kevin Oberman, Part time kid herder and retired Network Engineer
E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

cperciva, 2023-10-21:
Same here on 14.0-RC1.  Happy to help test patches.

rkoberman, 2023-10-21:
I see that there have been several updated versions of firmware released since the current.  Could one of those help? Can I just pull down the firmware and drop in the new files? Again, I have no real expertise on this, but would really like to get my Alder Lake system to have acceleration would really be great. Happy to do testing or gather any information that might help.

cperciva, 2023-10-21:
Per request from manu:

```
$ pciconf -lv vgapci0
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x46a6 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Integrated Graphics Controller'
    class      = display
    subclass   = VGA
```

djfiander, 2023-10-31:
I'm seeing the same problem on 14.0-RC3, but I **do**   see some messages in the log between the call to kldload and the system hanging:

Oct 28 16:32:41 victoria kernel: iic0: <I2C generic I/O> on iicbus0
Oct 28 16:32:42 victoria kernel: drmn0: <drmn> on vgapci0
Oct 28 16:32:42 victoria kernel: vgapci0: child drmn0 requested pci_enable_io
Oct 28 16:32:42 victoria syslogd: last message repeated 1 times
Oct 28 16:32:42 victoria kernel: drmn0: [drm] Incompatible option enable_guc=-1 - undocumented flag
Oct 28 16:32:42 victoria kernel: [drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
Oct 28 16:32:42 victoria kernel: [drm] Got stolen memory [base r](https://svnweb.freebsd.org/changeset/base/)x4b800000, size 0x4000000
Oct 28 16:32:42 victoria kernel: lkpi_iic0: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus1: <Philips I2C bus> on lkpi_iic0
Oct 28 16:32:42 victoria kernel: iic1: <I2C generic I/O> on iicbus1
Oct 28 16:32:42 victoria kernel: lkpi_iic1: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus2: <Philips I2C bus> on lkpi_iic1
Oct 28 16:32:42 victoria kernel: iic2: <I2C generic I/O> on iicbus2
Oct 28 16:32:42 victoria kernel: lkpi_iic2: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus3: <Philips I2C bus> on lkpi_iic2
Oct 28 16:32:42 victoria kernel: iic3: <I2C generic I/O> on iicbus3
Oct 28 16:32:42 victoria kernel: lkpi_iic3: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus4: <Philips I2C bus> on lkpi_iic3
Oct 28 16:32:42 victoria kernel: iic4: <I2C generic I/O> on iicbus4
Oct 28 16:32:42 victoria kernel: lkpi_iic4: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus5: <Philips I2C bus> on lkpi_iic4
Oct 28 16:32:42 victoria kernel: iic5: <I2C generic I/O> on iicbus5
Oct 28 16:32:42 victoria kernel: lkpi_iic5: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus6: <Philips I2C bus> on lkpi_iic5
Oct 28 16:32:42 victoria kernel: iic6: <I2C generic I/O> on iicbus6
Oct 28 16:32:42 victoria kernel: lkpi_iic6: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus7: <Philips I2C bus> on lkpi_iic6
Oct 28 16:32:42 victoria kernel: iic7: <I2C generic I/O> on iicbus7
Oct 28 16:32:42 victoria kernel: lkpi_iic7: <LinuxKPI I2C> on drmn0
Oct 28 16:32:42 victoria kernel: iicbus8: <Philips I2C bus> on lkpi_iic7
Oct 28 16:32:42 victoria kernel: iic8: <I2C generic I/O> on iicbus8
Oct 28 16:32:42 victoria kernel: lkpi_iic8: <LinuxKPI I2C>
Oct 28 16:32:42 victoria kernel:  on drmn0
Oct 28 16:32:42 victoria kernel: iicbus9: <Philips I2C bus> on lkpi_iic8
Oct 28 16:32:42 victoria kernel: iic9: <I2C generic I/O> on iicbus9
Oct 28 16:32:42 victoria kernel: drmn0: successfully loaded firmware image 'i915/adls_dmc_ver2_01.bin'
Oct 28 16:32:42 victoria kernel: drmn0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
Oct 28 16:32:42 victoria kernel: drmn0: successfully loaded firmware image 'i915/tgl_guc_62.0.0.bin'
Oct 28 16:32:42 victoria kernel: drmn0: successfully loaded firmware image 'i915/tgl_huc_7.9.3.bin'

ko56, 2023-11-08:
I have the same problem. 
14.0-RC4, Intel Alderlake-S graphics, and drm-515-kmod, version 5.15.118.
See the tail end of #232.

neelchauhan, 2023-11-08:
I had a similar issue back on 14-CURRENT. I gave up on FreeBSD because modern hardware support isn't a priority.

FreeBSD can't even support graphics on a 12th Gen HP Spectre. Linux has graphics on a M2 Mac, despite *reverse engineered* drivers. OpenBSD has graphics on 12th **and** 13th Gen Intel. Trust me, I've had them all.

bapt, 2023-11-08:
@neelchauhan first this is bullshit andgratuicious claim. There are people who spend a countless amount of time on drm drivers! and your comment is clearly inappropriate to them and also a totally useful comment, it helps noone here.

thewanderingtraderm, 2023-11-26:
Same problem here. When is it going to be fixed?

tdm4, 2023-12-08:
Same issue here:
```
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x4628 subvendor=0x1558 subdevice=0x4150
     vendor         = 'Intel Corporation'
     device         = 'Alder Lake-UP3 GT2 [UHD Graphics]'
     class          = display
     subclass       = VGA
```

Had to use the scfb driver in Xorg to get any kind of graphics.
Running `kldload i915kms` causes the system to completely lock up and freeze. Holding down the power button is the only way to bring the machine back.

rkoberman, 2023-12-09:
And, as in 252, deleting the firmware port will allow it to run... with
software rendering. Not great, but better than scfb.
Among other things, screen brightness management work.

tdm4, 2023-12-09:
@rkoberman - I copied away these files from /boot/modules: `i915_adlp_*.ko` and once I loaded i915kms, it worked and I could load X using modesetting, just no 3D acceleration as you mention. It's better than scfb (I can get redshift to work in it)

ko56, 2023-12-12:
I am also using scfb right now, and I have removed the drm-515 package entirely. But did you have any problems with DPMS? See [https://forums.freebsd.org/threads/xlib-extension-dpms-missing-on-display-0-0-14-0-release-xfce4.91257/](https://forums.freebsd.org/threads/xlib-extension-dpms-missing-on-display-0-0-14-0-release-xfce4.91257/). If so, did what you did help?

> @rkoberman - I copied away these files from /boot/modules: `i915_adlp_*.ko` and once I loaded i915kms, it worked and I could load X using modesetting, just no 3D acceleration as you mention. It's better than scfb (I can get redshift to work in it)

rkoberman, 2023-12-12:
scfb is extremely limited in its capabilities. It works, but that's about
it. Things like redshift and backlight all work. Pretty much all that is
lost is acceleration. For hte most part, not being a gamer, the only time I
notice this is when attempting to play an HD video while doing a big
compile (e.g. firefox or llvm). I have seen no downsides to using
drm-515-kmod. Before drm-515-knod I used scfb (no alternative) and am much
happier, now.

I just deleted the gpu-firmware-intel-kmod port and it worked.
-- 
Kevin Oberman, Part time kid herder and retired Network Engineer
E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

rkoberman, 2023-12-12:
Forgot to mention that I have not seen any issues with DPMS, though I have
only used the laptop's display. No errors logged.

ko56, 2023-12-12:
Doesn't work for me. Deleted the `gpu-firmware-intel-kmod-alderlake` package. Then added `drm-515-kmod`. Without a `20-scfb.conf` file in `/usr/local/etc/X11/xorg.conf.d`, the system freezes.

rkoberman, 2023-12-12:
> Doesn't work for me. Deleted the gpu-firmware-intel-kmod-alderlake
> package. Then added drm-515-kmod. Without a 20-scfb.conf file in
> /usr/local/etc/X11/xorg.conf.d, the system freezes.

Check your boot messages for a message about firmware load. 'dmesg -a'.
This should show any error prior to the freeze, though not the actual
trigger. Also, are you manually loading i915, /etc/rc.conf or other method

ko56, 2023-12-13:
I checked `dmesg -a` after the freeze, by powering down and then booting single-user. No error messages. 
I do not explicitly load anything in `/etc/rc.conf`.
The whole thing is very mysterious. Maybe the difference with you has something to do with my system not being a laptop?

rkoberman, 2023-12-16:
If you are not loading i915kms via rc.conf, how do you load it? Are you
loading it manually (kldload)? The only ways that load it safely are
manually or in rc.conf via the kld_load statement.

I am not surprised that you found no messages about the freeze, but I was
hoping for something that might either help or  point to a different issue.
It is also odd that you did not get a message about the missing firmware if
it is not present on your system. That should happen before the driver
actually loads.

ko56, 2023-12-16:
I don't load it manually either. Somehow the X server loads it by itself. Probably because of the `/usr/local/etc/X11/xorg.conf.d/20-scfb.conf`
What **is** on my system is `drm-510-kmod` (not 515), and 
`gpu-firmware-intel-kmod-alderlake20230210_1` and
`gpu-firmware-intel-kmod-tigerlake-20230210_1`.

rkoberman, 2023-12-16:
On Sat, Dec 16, 2023 at 6:15 AM ko56 ***@***.***> wrote:

> I don't load it manually either. Somehow the X server loads it by itself.
> Probably because of the /usr/local/etc/X11/xorg.conf.d/20-scfb.conf
> What *is* on my system is drm-510-kmod (not 515), and
> gpu-firmware-intel-kmod-alderlake20230210_1 and
> gpu-firmware-intel-kmod-tigerlake-20230210_1.
>
That explains it. drm-510-kmod does not support Alder Lake cards. You need
drm-515-kmod. You also must be running 14.0 or newer. I'm pretty sure that
older versions like the lkpi support for drm-515-kmod. This assumes that
your CPU is Alder Lake. You can confirm with "pciconf -lv | grep -B3 -A1
display".

E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

rkoberman, 2023-12-16:
On Sat, Dec 16, 2023 at 10:54 AM Kevin Oberman ***@***.***> wrote:

> On Sat, Dec 16, 2023 at 6:15 AM ko56 ***@***.***> wrote:
>
> That explains it. drm-510-kmod does not support Alder Lake cards. You need
> drm-515-kmod. You also must be running 14.0 or newer. I'm pretty [...]

Sorry. Typo. "older versions lack the lkpi support"

-- 
Kevin Oberman, Part time kid herder and retired Network Engineer
E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

ko56, 2023-12-17:
Yes, of course. See my original post (and some of the later ones) in this thread. Now I am at 14.0-RELEASE-p3 with the same problem: `drm-515-kmod` freezes the system, so I've reverted to `drm-510-kmod`.

rkoberman, 2023-12-26:
Sorry for the late response, but this has been a really busy week plus I
lot my system after a bad update to head.

Last suggestion... What does "locate i915_adlp" show?. Or try "ls
/boot/modules/i915_adlp-*". If it is not found, I am out of ises.
-- 
Kevin Oberman, Part time kid herder and retired Network Engineer
E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

ko56, 2023-12-26:
This is in my `/boot/modules`:
```
i915_adlp_dmc_bin.ko          i915_adlp_dmc_ver2_12_bin.ko
i915_adlp_dmc_ver2_10_bin.ko  i915_adlp_guc_62_0_3_bin.ko
```
As I said above, this is with the `drm-510-kmod` pkg, I've removed `drm-515-kmod`.

It would be helpful if you could send the following info, so I could compare with my system, and, hopefully, nail this down better:
1) CPU and GPU.
2) What drm-5xx-mod pkg you have.
3) What `/etc/rc.conf` loads, or you otherwise load.
4) What you have removed from `gpu-firmware-intel` by hand.
5) Anything else that you think may be relevant. 
Maybe smth. from `/var/log/Xorg.0.log` ?
6) What is the result of all this, i.e. what GPU does your OS think it is supporting. 
Maybe `Xorg.0.log ` has all that's needed.

IN any case, thanks for all your help.

rkoberman, 2023-12-29:
On Tue, Dec 26, 2023 at 1:27 PM ko56 ***@***.***> wrote:

> This is in my /boot/modules:
>
> i915_adlp_dmc_bin.ko          i915_adlp_dmc_ver2_12_bin.ko
> i915_adlp_dmc_ver2_10_bin.ko  i915_adlp_guc_62_0_3_bin.ko
>
> As I said above, this is with the drm-510-kmod pkg, I've removed
> drm-515-kmod.
>
> It would be helpful if you could send the following info, so I could
> compare with my system, and, hopefully, nail this down better:
>
>    1. CPU and GPU.
>
>  12th Gen Intel(R) Core(TM) i5-1235U (2496.00-MHz K8-class CPU)
Alder Lake-UP3 GT2 [Iris Xe Graphics

>
>    1. What drm-5xx-mod pkg you have.
>
>  drm-515-kmod-5.15.118_3

>
>    1. What /etc/rc.conf loads, or you otherwise load.
>
>  kld_list="cpuctl acpi_ibm acpi_video filemon" I am manually loading
drm-515-kmod due to issues with the iwlwifi driver.

>
>    1. What you have removed from gpu-firmware-intel by hand.
>
> Nothing. I remove the gpu-firmware-intel-kmod-alderlake (
***@***.***) completely.

>
>    1. Anything else that you think may be relevant.
>    Maybe smth. from /var/log/Xorg.0.log ?
>
> I keep hoping for something, but have never found anything.

>
>    1. What is the result of all this, i.e. what GPU does your OS think it
>    is supporting.
>    Maybe Xorg.0.log has all that's needed.'
>
> I run the modesetting driver. (I don't have the old Intel driver as it has
not been maintained for a long time.
[   175.846] (II) modeset(0): Refusing to try glamor on llvmpipe
[   175.866] (II) modeset(0): glamor initialization failed
[   175.866] (II) modeset(0): ShadowFB: preferred YES, enabled YES
[   175.866] (II) modeset(0): Double-buffered shadow updates: off
[   175.868] (II) modeset(0): Output eDP-1 has no monitor section
[   175.868] (II) modeset(0): Output DP-1 has no monitor section
[   175.870] (II) modeset(0): Output DP-2 has no monitor section
[   175.870] (II) modeset(0): Output DP-3 has no monitor section
[   175.872] (II) modeset(0): Output DP-4 has no monitor section
[   175.874] (II) modeset(0): EDID for output eDP-1
[   175.874] (II) modeset(0): Manufacturer: CMN  Model: 1614  Serial#: 0
[   175.874] (II) modeset(0): Year: 2021  Week: 16
[   175.874] (II) modeset(0): EDID Version: 1.4
[   175.874] (II) modeset(0): Digital Display Input
[   175.874] (II) modeset(0): 8 bits per channel
[   175.874] (II) modeset(0): Digital interface is DisplayPort
[   175.874] (II) modeset(0): Max Image Size [cm]: horiz.: 34  vert.: 22
[   175.874] (II) modeset(0): Gamma: 2.20
[   175.874] (II) modeset(0): No DPMS capabilities specified
[   175.874] (II) modeset(0): Supported color encodings: RGB 4:4:4
[   175.874] (II) modeset(0): First detailed timing is preferred mode
[   175.874] (II) modeset(0): Preferred mode is native pixel format and
[... 205 lines cut ...]
[   175.880] (II) UnloadModule: "scfb"
[   175.880] (II) Unloading scfb
[   175.880] (II) UnloadModule: "vesa"
[   175.880] (II) Unloading vesa
[   175.887] (==) modeset(0): Backing store enabled
[   175.887] (==) modeset(0): Silken mouse enabled
[   175.964] (II) modeset(0): Initializing kms color map for depth 24, 8
bpc.
[   175.966] (==) modeset(0): DPMS enabled

If you are running the Intel driver, that could be an issue as it won't
work with Alder Lake. It worked with my old Sandy Lake GPU that I bought in
2011, but Comet Lake worked badly with it and I suspect that Alder Lake
doesn't work at all.

I have only the stuff to enable evdev support (20-evdev-kbd.conf), and
libinput (20-libinput.conf) and the quirks file (10-quirks.conf) in
xorg.conf.d. I believe all three are defaults. Basically, I let Xorg
configure itself.
Message ID: ***@***.***>

> --
Kevin Oberman, Part time kid herder and retired Network Engineer
E-mail: ***@***.***
PGP Fingerprint: D03FB98AFA78E3B78C1694B318AB39EF1B055683

hirairaka5, 2023-12-30:
I have a Lenovo ThinkPad E15 Gen 4 Intel laptop (Alder Lake P) and I decided to install FreeBSD 14 on it.  The problem described here is pretty much exactly the same as what I've been seeing.  I spent some time with ddb and figured out what was going on, and now the laptop is running Xorg without any obvious problems.

The laptop wasn't actually hanging, it turns out.  The drm-515-kmod driver disabled writes to the console framebuffer because the efifb bios uses the same framebuffer mapping as the driver.  I diked out the code that disabled writes and all of a sudden I could see the logging that I had turned on, and ddb worked:

```Diff
commit 2bd315d861ceea3ab9c19aa1c2d2ef6adc2b281a (HEAD -> vt_freeze_main_vd-1)
Author: Donn Seeley <donn@xmission.com>
Date:   Sat Dec 30 15:22:13 2023 -0700

    fb: Don't suppress console writes with vt_freeze_main_vd()
    
    The vt_freeze_main_vd() function sets FB_FLAG_NOWRITE on the console
    framebuffer if the framebuffer is also used for graphics.  That
    suppresses debug logging (and ddb) on the console.  With modesetting, it
    appears to be unnecessary to disable the console framebuffer; dike out
    the code in vt_freeze_main_vd() until someone comes up with a better
    solution.

diff --git a/drivers/gpu/drm/linux_fb.c b/drivers/gpu/drm/linux_fb.c
index 5575676629..401006417f 100644
--- a/drivers/gpu/drm/linux_fb.c
+++ b/drivers/gpu/drm/linux_fb.c
@@ -61,6 +61,7 @@ static int __unregister_framebuffer(struct linux_fb_info *fb_info);
 static void
 vt_freeze_main_vd(struct apertures_struct *a)
 {
+#if 0
        struct fb_info *fb;
        int i;
        bool overlap = false;
@@ -87,6 +88,7 @@ vt_freeze_main_vd(struct apertures_struct *a)
                if (overlap == true)
                        fb->fb_flags |= FB_FLAG_NOWRITE;
        }
+#endif
 }
 
 void
```

After getting ddb to work, I was able to see that the kernel was segfaulting in the xarray code.  I made a small change to  that code to avoid the problem; my commit message describes the issue:

```Diff
commit 52d9683e62220c468b5928376b039975d3064d0d (HEAD -> xarray-1)
Author: Donn Seeley <donn@xmission.com>
Date:   Sat Dec 30 13:15:28 2023 -0700

    xa_destroy: leave the xarray re-usable, like Linux does
    
    When trying to get graphics to work on an Alder Lake P laptop, I got a
    crash.  I copied down the information from ddb:
    
      --- trap 0xc, rip=0xffffffff80b302ec, rsp=0xfffffe013acf44e0 ...
      __mtx_lock_sleep() at __mtx_lock_sleep+0xbc
      xa_load() at xa_load+0x73
      guc_lrc_desc_pin() at guc_lrc_desc_pin+0x49
      intel_guc_submission_enable() at intel_guc_submission_enable+0x7d
      __uc_init_hw() at __uc_init_hw+0x46f
      intel_gt_init_hw() at intel_gt_init_hw+0x481
      intel_gt_resume() at intel_gt_resume+0x5cc
      intel_gt_init() at intel_gt_init+0x213
      i915_gem_init() at i915_gem_init+0x92
      i915_driver_probe() at i915_driver_probe+0x40
      i915_pci_probe() at i915_pci_probe+0x40
      linux_pci_attach_device() at linux_pci_attach_device+0x420
      device_attach() at device_attach+0x3be
      bus_generic_driver_added() at bus_generif_driver_added+0xa1
      [etc.]
    
    We got a segfault in __mtx_lock_sleep() when trying to dereference a
    null thread pointer.  The sequence of events looks like this:
    
      intel_guc_submission_enable(guc)
        guc_init_lrc_mapping(guc)
          xa_destroy(&guc->context_lookup)
            mtx_destroy(&xa->mtx)
              _mtx_destroy(&(m)->mtx_lock)
                m->mtx_lock = MTX_DESTROYED
          guc_kernel_context_pin(guc, ce)
            guc_lrc_desc_pin(guc, loop=true)
              lrc_desc_registered(guc, id)
                __get_context(guc, id)
                  xa_load(xa=&guc->context_lookup, index=id)
                    xa_lock(xa)
                      mtx_lock(&(xa)->mtx)
                        mtx_lock_flags((m), 0)
                          _mtx_lock_flags((m), (opts), (file), (line))
                            __mtx_lock_flags(&(m)->mtx_lock, o, f, l)
                              _mtx_obtain_lock_fetch(m, &v, tid)
                                atomic_fcmpset_acq_ptr(&(mp)->mtx_lock, vp, (tid))
                              _mtx_lock_sleep(m, v, opts, file, line)
                                __mtx_lock_sleep(&(m)->mtx_lock, v)
                                  owner = lv_mtx_owner(v)
                                    ((struct thread *)((v) & ~MTX_FLAGMASK))
                                  TD_IS_RUNNING(owner)
                                    TD_GET_STATE(td) == TDS_RUNNING
                                      (td)->td_state
    
    We crash because v == MTX_DESTROYED, which means that td, the thread
    pointer part of the owner cookie, is NULL and we can't dereference it.
    
    The Linux version of xa_destroy() is careful to fix up the xarray so
    that it can be used again.  The FreeBSD version is simpler: we just need
    to maintain the xarray flags when we reinitialize the xarray structure.

diff --git a/sys/compat/linuxkpi/common/src/linux_xarray.c b/sys/compat/linuxkpi/common/src/linux_xarray.c
index 44900666242f..856a67ba7e3c 100644
--- a/sys/compat/linuxkpi/common/src/linux_xarray.c
+++ b/sys/compat/linuxkpi/common/src/linux_xarray.c
@@ -352,6 +352,9 @@ xa_destroy(struct xarray *xa)
        radix_tree_for_each_slot(ppslot, &xa->root, &iter, 0)
                radix_tree_iter_delete(&xa->root, &iter, ppslot);
        mtx_destroy(&xa->mtx);
+
+       /* Linux leaves the xarray re-usable after xa_destroy().  */
+       xa_init_flags(xa, xa->flags);
 }
 
 /*
```

I don't know whether the change to `vt_freeze_main_vd()` will cause problems for other systems, but it doesn't seem to be an issue on my system.  I'd love to hear whether these changes fix other folks' problems.

ko56, 2023-12-31:
@rkoberman Thanks. I duplicated what you did as much as I could, including the `modesetting` driver, but no luck. My CPU is i5-13600 and the GPU Intel 770 UHD (Alderlake S).
However, the next post by @hirairaka5 seems hopeful.

decke, 2023-12-31:
i5-13600 is Raptor Lake so 13. Generation.

ko56, 2023-12-31:
Is your comment referring to the discussion with @rkoberman , or to the one with @hirairaka5 ?

hirairaka5, 2024-01-01:
I submitted the xarray kernel bug to FreeBSD BugZilla as bug 276057.

FWIW, I thought that it was annoying that Xorg wouldn't default to using the modesetting driver for my recent Intel chipset.  I don't have a fix for that, but I thought that I'd provide my trivial `/usr/local/etc/X11/xorg.conf` file:

```
  Section "Screen"
      Identifier "LaptopScreen"
      Device     "LaptopDevice"
  EndSection

  Section "Device"
      Identifier "LaptopDevice"
      Driver     "modesetting"
  EndSection
```

ko56, 2024-01-01:
I looked at the bug 276057 on Bugzilla.  I am interested in how this turns out, as I've been having the "freezing"problem for a while, with no resolution.

nenjan72, 2024-01-02:
I have got AlderLake-P graphics working on my Thinkpad X1 Extreme gen 5 with @hirairaka5 patch on Freebsd 14 and drm-515-kmod  and gpu-firmware-intel-kmod from /usr/ports. No more kernel panic/freezing :)

evadot, 2024-01-05:
Alderlake support was recent in 5.15, for Alderlake (-P or -S) systems please use 6.1-lts

closed 2024-01-05 by evadot

tdm4, 2024-04-21:
I have an Alder Lake - U (i7-1255U) in my laptop and I get this lockup. Do I need to wait for FreeBSD 14.1 and use 6.1-LTS or will 5.15 work with it?

evadot, 2024-04-21:
> I have an Alder Lake - U (i7-1255U) in my laptop and I get this lockup. Do I need to wait for FreeBSD 14.1 and use 6.1-LTS or will 5.15 work with it?

Alder Lake will not work with 5.15 so you need to wait.

hirairaka5, 2024-04-21:
I've been using graphics with Alder Lake-P on 14.0 without a problem for months, after applying the kernel patch that I provided above.  I don't know whether the fix would help Alder Lake-U, but it might be worth a shot.

nwildner, 2024-07-03:
Sup folks. 

Still reproducible on FreeBSD 14.1. Asus Dash TUF15, i7 12650 + Intel 630

If I `kdload i915kms` without the `gpu-firmware-intel-kmod-alderlake` firmware package installed, it works.

But, if I try to load this module with the firmware package installed, laptop will hang for 5 seconds and reset completely and power cycle. No video output, no log messages, nothing.

evadot, 2024-07-03:
> Sup folks.
> 
> Still reproducible on FreeBSD 14.1. Asus Dash TUF15, [i7 12650H](https://www.intel.com/content/www/us/en/products/sku/226066/intel-core-i712650h-processor-24m-cache-up-to-4-70-ghz/specifications.html) + Intel 630
> 
> If I `kdload i915kms` without the `gpu-firmware-intel-kmod-alderlake` firmware package installed, it works.
> 
> But, if I try to load this module with the firmware package installed, laptop will hang for 5 seconds and reset completely and power cycle. No video output, no log messages, nothing.

With which drm-kmod ? You need 6.1 so on 14.1 you will need to compile/install the port directly, no package is available.

nwildner, 2024-07-03:
> With which drm-kmod ? You need 6.1 so on 14.1 you will need to compile/install the port directly, no package is available.

Hi @evadot . Using the default `drm-kmod` which will point to `drm-515-kmod`. 

Ok. I'll compile the port and let you know here as soon as I get my laptop.

rkoberman, 2024-07-04:
Just confirmed that drm-61-kmod is default for 14.1. 515 will fail to load
firmware forever. It will not be fixed as support was not ready in 5.15 of
the Linux kernel.

You probably need to build the port and will need kernel sources as
packages are built for the oldest supported minor release of a major
release. That is still 14.0.

ko56, 2024-07-04:
OK, I finally built the port from source, and everything works fine.
    At last my Intel UHD 770 graphics card is detected!
    
    On 07/03/24 20:23, Kevin Oberman wrote:
    
    
      
      Just confirmed that drm-61-kmod is default for 14.1. 515 will fail
      to load
      
      firmware forever. It will not be fixed as support was not ready in
      5.15 of
      
      the Linux kernel.
      
      
      You probably need to build the port and will need kernel sources
      as
      
      packages are built for the oldest supported minor release of a
      major
      
      release. That is still 14.0.

      
      

      
      
<https://www.intel.com/content/www/us/en/products/sku/226066/intel-core-i712650h-processor-24m-cache-up-to-4-70-ghz/specifications.html>
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
      
"target": "https://github.com/freebsd/drm-kmod/issues/252#issuecomment-2207638886",
"url": "https://github.com/freebsd/drm-kmod/issues/252#issuecomment-2207638886",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]