freebsd/drm-kmod #258: drm-515-kmod amdgpu memory leak

phryk, 2023-10-01:
**Describe the bug**
`drm-515-kmod` leaks memory when driving an RX 6600XT through `amdgpu`.

**FreeBSD version**
FreeBSD aether 14.0-BETA3 FreeBSD 14.0-BETA3 #0 releng/14.0-n265111-c429a0425dc4: Fri Sep 22 05:52:05 UTC 2023     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400097 1400097
**PCI Info**

<details>
  <summary>pciconf -lv</summary><pre>
hostb0@pci0:0:0:0:	class=0x060000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x0108 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 Processor Family DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x09 hdr=0x01 vendor=0x8086 device=0x0101 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200/2nd Generation Core Processor Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
none0@pci0:0:22:0:	class=0x078000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1c3a subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family MEI Controller'
    class      = simple comms
ehci0@pci0:0:26:0:	class=0x0c0320 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c2d subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family USB Enhanced Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:0:27:0:	class=0x040300 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c20 subvendor=0x1043 subdevice=0x8410
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib4@pci0:0:28:0:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c10 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 1'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:28:1:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c12 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 2'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:28:2:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c14 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 3'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:28:3:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c16 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 4'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:28:4:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c18 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 5'
    class      = bridge
    subclass   = PCI-PCI
pcib9@pci0:0:28:5:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c1a subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 6'
    class      = bridge
    subclass   = PCI-PCI
pcib10@pci0:0:28:6:	class=0x060401 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x244e subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '82801 PCI Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib12@pci0:0:28:7:	class=0x060400 rev=0xb5 hdr=0x01 vendor=0x8086 device=0x1c1e subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family PCI Express Root Port 8'
    class      = bridge
    subclass   = PCI-PCI
ehci1@pci0:0:29:0:	class=0x0c0320 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c26 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family USB Enhanced Host Controller'
    class      = serial bus
    subclass   = USB
isab0@pci0:0:31:0:	class=0x060100 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c46 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = 'P67 Express Chipset LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
ahci1@pci0:0:31:2:	class=0x010601 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c02 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family 6 port Desktop SATA AHCI Controller'
    class      = mass storage
    subclass   = SATA
ichsmb0@pci0:0:31:3:	class=0x0c0500 rev=0x05 hdr=0x00 vendor=0x8086 device=0x1c22 subvendor=0x1043 subdevice=0x844d
    vendor     = 'Intel Corporation'
    device     = '6 Series/C200 Series Chipset Family SMBus Controller'
    class      = serial bus
    subclass   = SMBus
pcib2@pci0:1:0:0:	class=0x060400 rev=0xc1 hdr=0x01 vendor=0x1002 device=0x1478 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 10 XL Upstream Port of PCI Express Switch'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:2:0:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x1479 subvendor=0x1002 subdevice=0x1479
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 10 XL Downstream Port of PCI Express Switch'
    class      = bridge
    subclass   = PCI-PCI
vgapci0@pci0:3:0:0:	class=0x030000 rev=0xc1 hdr=0x00 vendor=0x1002 device=0x73ff subvendor=0x1043 subdevice=0x05d3
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 23 [Radeon RX 6600/6600 XT/6600M]'
    class      = display
    subclass   = VGA
hdac0@pci0:3:0:1:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0xab28 subvendor=0x1002 subdevice=0xab28
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 21/23 HDMI/DP Audio Controller'
    class      = multimedia
    subclass   = HDA
xhci0@pci0:5:0:0:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1b21 device=0x1042 subvendor=0x1043 subdevice=0x8488
    vendor     = 'ASMedia Technology Inc.'
    device     = 'ASM1042 SuperSpeed USB Host Controller'
    class      = serial bus
    subclass   = USB
xhci1@pci0:8:0:0:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1b21 device=0x1042 subvendor=0x1043 subdevice=0x8488
    vendor     = 'ASMedia Technology Inc.'
    device     = 'ASM1042 SuperSpeed USB Host Controller'
    class      = serial bus
    subclass   = USB
re0@pci0:9:0:0:	class=0x020000 rev=0x06 hdr=0x00 vendor=0x10ec device=0x8168 subvendor=0x1043 subdevice=0x8432
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller'
    class      = network
    subclass   = ethernet
pcib11@pci0:10:0:0:	class=0x060401 rev=0x01 hdr=0x01 vendor=0x1b21 device=0x1080 subvendor=0x1b21 subdevice=0x1080
    vendor     = 'ASMedia Technology Inc.'
    device     = 'ASM1083/1085 PCIe to PCI Bridge'
    class      = bridge
    subclass   = PCI-PCI
ahci0@pci0:12:0:0:	class=0x010601 rev=0x11 hdr=0x00 vendor=0x1b4b device=0x9172 subvendor=0x1043 subdevice=0x8477
    vendor     = 'Marvell Technology Group Ltd.'
    device     = '88SE9172 SATA 6Gb/s Controller'
    class      = mass storage
    subclass   = SATA</pre>
</details>

**DRM KMOD version**
drm-515-kmod 5.15.25_3

**To Reproduce**

* boot system with `amdgpu` in kld_list
* log into X11/awesome (via slim)
* run `vmstat -m | grep lkpikmalloc` in a loop
* watch `lkpikmalloc` grow forever

**Additional context**

I already tried locally built `drm-515-kmod` as well as all flavors of  `gpu-firmware-amd-kmod`
from the main branch of the ports (cloned today), built against the `releng/14.0` branch checked
out in `/usr/src`. No change in behavior.

When `lkpikmalloc` was already over 6G, trying to start firefox lead to both displays switching off,
switching to a TTY didn't work. Had to reboot the system, but during shutdown, the displays
switched back on and full-res KMS VT was visible.

Szowisz, 2023-10-10:
Hello, same issue on RX 6650 XT. The problem appears on FreeBSD version 14 as well 15. I have that issue, no matter which desktop i tried. 
**Some debug info**

<details>
  <summary>pciconf -lvbce</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0xa703 subvendor=0x1458 subdevice=0x5000
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0xa70d subvendor=0x1458 subdevice=0x5000
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-PCI
    cap 10[40] = PCI-Express 2 root port max data 256(256) ARI disabled
                 max read 128
                 link x8(x16) speed 16.0(32.0)
                 slot 1 power limit 75000 mW
    cap 05[80] = MSI supports 1 message, 64 bit 
    cap 0d[98] = PCI Bridge subvendor=0x1458 subdevice=0x5000
    cap 01[a0] = powerspec 3  supports D0 D3  current D0
    ecap 0001[100] = AER 1 0 fatal 0 non-fatal 0 corrected
    ecap 000d[220] = ACS 1 Source Validation disabled, Translation Blocking disabled
                     P2P Req Redirect disabled, P2P Cmpl Redirect disabled
                     P2P Upstream Forwarding disabled, P2P Egress Control unavailable
                     P2P Direct Translated unavailable, Enhanced Capability unavailable
    ecap 001f[150] = Precision Time Measurement 1
    ecap 0019[a30] = PCIe Sec 1 lane errors 0
    ecap 0025[a90] = Data Link Feature 1
    ecap 0026[a9c] = Physical Layer 16.0 GT/s 1
    ecap 0027[edc] = Lane Margining at Receiver 1
    ecap 002a[adc] = Physical Layer 32.0 GT/s 1
pcib4@pci0:0:6:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0xa74d subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    device     = 'Raptor Lake PCIe 4.0 Graphics Port'
    class      = bridge
    subclass   = PCI-PCI
    cap 10[40] = PCI-Express 2 root port max data 256(256) ARI enabled
                 max read 128
                 link x4(x4) speed 8.0(16.0)
                 slot 0 power limit 75000 mW
    cap 05[80] = MSI supports 1 message 
    cap 0d[90] = PCI Bridge subvendor=0x0000 subdevice=0x0000
    cap 01[a0] = powerspec 3  supports D0 D3  current D0
    ecap 0001[100] = AER 1 0 fatal 0 non-fatal 0 corrected
    ecap 000d[220] = ACS 1 Source Validation disabled, Translation Blocking disabled
                     P2P Req Redirect disabled, P2P Cmpl Redirect disabled
                     P2P Upstream Forwarding disabled, P2P Egress Control unavailable
                     P2P Direct Translated unavailable, Enhanced Capability unavailable
    ecap 001e[200] = L1 PM Substates 1
    ecap 001f[150] = Precision Time Measurement 1
    ecap 0002[280] = VC 1 max VC1
    ecap 0019[a30] = PCIe Sec 1 lane errors 0
    ecap 0025[a90] = Data Link Feature 1
    ecap 0026[a9c] = Physical Layer 16.0 GT/s 1
    ecap 0027[edc] = Lane Margining at Receiver 1
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7ae0 subvendor=0x1458 subdevice=0x5007
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH USB 3.2 Gen 2x2 XHCI Controller'
    class      = serial bus
    subclass   = USB
    bar   [10] = type Memory, range 64, base 0x4000000000, size 65536, enabled
    cap 01[70] = powerspec 2  supports D0 D3  current D0
    cap 05[80] = MSI supports 8 messages, 64 bit enabled with 1 message
    cap 09[90] = vendor (length 20) Intel cap 15 version 0
    cap 09[b0] = vendor (length 0) Intel cap 0 version 1
none0@pci0:0:20:2:	class=0x050000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7aa7 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH Shared SRAM'
    class      = memory
[... 1276 lines cut ...]
  
Section "InputClass"
        Identifier "system-keyboard"
        MatchIsKeyboard "on"
        Option "XkbLayout" "pl"
EndSection

</details>

<details>
  <summary>amdgpu.conf</summary>
  
Section "Device"
Identifier "Card0"
Driver "amdgpu"
Option "VariableRefresh" "true"
EndSection

</details>

**FreeBSD version**
FreeBSD FreeBSD-GIGABYTE 15.0-CURRENT FreeBSD 15.0-CURRENT #0 main-n265864-23ef25d25d98: Tue Oct 10 19:13:53 CEST 2023     root@FreeBSD-GIGABYTE:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1500001 1500001

**DRM KMOD version**
drm-515-kmod 5.15.25_3

wulf7, 2023-10-13:
Try this patch:
```
diff --git a/drivers/dma-buf/dma-fence-chain.c b/drivers/dma-buf/dma-fence-chain.c
index ae84a72cb2..dc3618011b 100644
--- a/drivers/dma-buf/dma-fence-chain.c
+++ b/drivers/dma-buf/dma-fence-chain.c
@@ -153,6 +153,7 @@ dma_fence_chain_init(struct dma_fence_chain *chain,
 	spin_lock_init(&chain->lock);
 	chain->fence = fence;
 	chain->prev = prev;
+	chain->prev_seqno = 0;
 	prev_chain = to_dma_fence_chain(prev);
 	if (prev_chain != NULL &&
 	    __dma_fence_is_later(seqno, prev->seqno, prev->ops)) {
```

wulf7, 2023-10-15:
OOPS. Wrong patch. Proper one is:
```
diff --git a/drivers/dma-buf/dma-fence-chain.c b/drivers/dma-buf/dma-fence-chain.c
index ae84a72cb2..c13760fe8c 100644
--- a/drivers/dma-buf/dma-fence-chain.c
+++ b/drivers/dma-buf/dma-fence-chain.c
@@ -219,6 +219,7 @@ dma_fence_chain_walk(struct dma_fence *fence)
 		dma_fence_put(prev);
 	}
 
+	dma_fence_put(fence);
 	return (prev);
 }
 
```

Szowisz, 2023-10-15:
> OOPS. Wrong patch. Proper one is:
>
> ```
>
> diff --git a/drivers/dma-buf/dma-fence-chain.c b/drivers/dma-buf/dma-fence-chain.c
>
> index ae84a72cb2..c13760fe8c 100644
>
> --- [...]

Okay thanks, I will try that patch

evadot, 2023-10-16:
> OOPS. Wrong patch. Proper one is:
>
> ```
> diff --git a/drivers/dma-buf/dma-fence-chain.c b/drivers/dma-buf/dma-fence-chain.c
> index ae84a72cb2..c13760fe8c 100644
> --- a/drivers/dma-buf/dma-fence-chain.c [...]

Hi Vladimir,

I've tested this patch since I got the problem too and never took time to look at it, it solves the problems, thanks a lot !
Can you commit this to master and cherry-pick it to 5.15-lts ? I'll make a new release after that.

phryk, 2023-10-16:
How do I properly apply this patch? just put it into `files/patch-foo` in the ports directory?

wulf7, 2023-10-16:
> I've tested this patch since I got the problem too and never took time to look at it, it solves the problems, thanks a lot !
> Can you commit this to master and cherry-pick it to 5.15-lts ? I'll make a new release after that.

Done. I cherry-picked couple of other commits too. I hope they don't break anything.
The patch revealed one more bug in dma-buf. Previously leaking memory chunks are allocated with M_DMABUF type but freed with M_LKPI type that brokes memory accounting for this types. I will try to fix that.

wulf7, 2023-10-16:
> How do I properly apply this patch? just put it into `files/patch-foo` in the ports directory?

Just build and install current 5.15-lts branch from sources

evadot, 2023-10-16:
> Done. I cherry-picked couple of other commits too. I hope they don't break anything. The patch revealed one more bug in dma-buf. Previously leaking memory chunks are allocated with M_DMABUF type but [...]

Thanks,

Will wait a bit before tagging a release so you have time to fix this.

evadot, 2023-10-16:
> Thanks,
>
> Will wait a bit before tagging a release so you have time to fix this.

I did a release anyway, we can always make a new one after :)

evadot, 2023-10-16:
Issue is fixed, thanks @wulf7

closed 2023-10-16 by evadot