freebsd/drm-kmod #259: System freezes when firmware is loaded

rkoberman, 2023-10-03:
**Describe the bug**
A clear and concise description of what the bug is.
When the firmware is loaded on 
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0$
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-UP3 GT2 [Iris Xe Graphics]'
the system instantly freezes with no logging or messages. System must be power cycled to restore operation and the firmware uninstalled before loading dm-515-kmod. System operates with no firmware load, but lacks any acceleration.

**FreeBSD version**
FreeBSD ptavv 14.0-CURRENT FreeBSD 14.0-CURRENT amd64 1400094 #6 n264565-dedc71fd3890: Wed Sep 27 07:00:26 PDT 2023     root@ptavv:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400094 1400094

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  hostb0@pci0:0:0:0:	class=0x060000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x4601 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x46a8 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-UP3 GT2 [Iris Xe Graphics]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x461d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake Innovation Platform Framework Processor Participant'
    class      = dasp
pcib1@pci0:0:6:0:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x464d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x4 Controller'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x466e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:7:2:	class=0x060400 rev=0x04 hdr=0x01 vendor=0x8086 device=0x462f subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:10:0:	class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x467d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Platform Monitoring Technology'
    class      = dasp
xhci0@pci0:0:13:0:	class=0x0c0330 rev=0x04 hdr=0x00 vendor=0x8086 device=0x461e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 USB Controller'
    class      = serial bus
    subclass   = USB
none2@pci0:0:13:2:	class=0x0c0340 rev=0x04 hdr=0x00 vendor=0x8086 device=0x463e subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
none3@pci0:0:13:3:	class=0x0c0340 rev=0x04 hdr=0x00 vendor=0x8086 device=0x466d subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
xhci1@pci0:0:20:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ed subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH USB 3.2 xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none4@pci0:0:20:2:	class=0x050000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ef subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Shared SRAM'
    class      = memory
    subclass   = RAM
iwlwifi0@pci0:0:20:3:	class=0x028000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51f0 subvendor=0x8086 subdevice=0x0090
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH CNVi WiFi'
    class      = network
none5@pci0:0:22:0:	class=0x078000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e0 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH HECI Controller'
    class      = simple comms
isab0@pci0:0:31:0:	class=0x060100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x5182 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:	class=0x040380 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51c8 subvendor=0x17aa subdevice=0x22e9
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a3 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P SMBus Host Controller'
    class      = serial bus
    subclass   = SMBus
none6@pci0:0:31:5:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a4 subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH SPI Controller'
    class      = serial bus
em0@pci0:0:31:6:	class=0x020000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x1a1f subvendor=0x17aa subdevice=0x22e8
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (16) I219-V'
    class      = network
    subclass   = ethernet
nvme0@pci0:2:0:0:	class=0x010802 rev=0x01 hdr=0x00 vendor=0xc0a9 device=0x540a subvendor=0xc0a9 subdevice=0x5021
    vendor     = 'Micron/Crucial Technology'
    device     = 'P2 [Nick P2] / P3 / P3 Plus NVMe PCIe SSD (DRAM-less)'
    class      = mass 

</details>

**DRM KMOD version**
drm-515-kmod 5.15.25_3

**To Reproduce**
Steps to reproduce the behavior:
install gpu-firmware-intel-kmod@alderlake
Boot system to single user. kldload i915kms or add i915kms to kld_list in /etc/rc.conf and boot normally.
**Screenshots**
No screnshots
**Additional context**

I note that Intel has issued several newer firmware packages since February. No idea if a newer one might help.

decke, 2023-10-03:
Is this a duplicate of #252 ?

rkoberman, 2023-10-03:
Dup of 252. Sorry!

rkoberman, 2023-10-03:
Oh, crap! I was sure that I had only opened a bug in bugzilla, but, yres,
it is the same bug. So sorry. I just closed it.

closed 2023-10-03 by rkoberman