freebsd/drm-kmod #261: memory usage drops below zero

magicalskye, 2023-10-24:
hello lovely video card people,

i am having major gremlins in the machine.

i’m sorry that this will not be a succinct and to-the-point bug report but this is because i have no idea what i am doing. i cannot stress enough how much i have no idea. this is literally my first day on freeBSD. i have plenty of experience with hacky stuff on linux but i am 100% new to this ecosystem.

so, i have a Radeon RX 6500 XT. i installed freeBSD 13.2 and it obviously could not make heads or tails of the card. so i found out how to upgrade to 14.0 and then pkg install-ed the drm-515-kmod and i was very happy to finally have X running. then a helpful person in IRC told me to check for a memory leak and sure enough, it was leaking. and leaking more the more i was doing things. glxgears made it get bigger faster. no surprise there.

so i figured out how to use ports, and got the ports git repos fresh per the handbook, and because this is my first day, i just did portmaster -a and let it do its thing. i watched it compile the 515 and at the end it even gave me a helpful message about the package. so i know it updated that from ports.

i did a reboot and all was fine, so i turned on my fluxbox and ran the glxgears again. and at first i was happy because i saw that it was not getting bigger but smaller. but it kept getting smaller. and smaller. and then it reached zero and overflowed (underflowed?) into an impossibly large number. it keeps getting smaller, and the more i run the glxgears the smaller it gets, and if i do nothing it stays mostly the same size.

thankfully, i have not yet figured out which command replaces the 'watch' command i know from linux, therefore i was just doing 'vmstat -m | grep lkpikmalloc' repeatedly, which also means i could simply copy and paste the output (and leave out the parts where i was frantically reading man pages to find out what the columns actually mean, which they unfortunately do not explain)

now i may not know exactly what i did today but i feel like something here is very, very wrong and i am expecting either godzilla or little green men any minute now

~~~
root@blfred:~ # vmstat -m | grep lkpikmalloc
  lkpikmalloc   365  2552K    22058  16,32,64,128,256,384,512,1024,2048,4096,8192,16384,32768,65536
root@blfred:~ # vmstat -m | grep lkpikmalloc
  lkpikmalloc 18446744073709551461  2374K    25262  16,32,64,128,256,384,512,1024,2048,4096,8192,16384,32768,65536
root@blfred:~ # vmstat -m | grep lkpikmalloc
  lkpikmalloc 18446744073709542505   153K    82662  16,32,64,128,256,384,512,1024,2048,4096,8192,16384,32768,65536
root@blfred:~ # vmstat -m | grep lkpikmalloc
  lkpikmalloc 18446744073709541809 18014398509481965K    87142  16,32,64,128,256,384,512,1024,2048,4096,8192,16384,32768,65536
root@blfred:~ # vmstat 
 procs    memory    page                      disks     faults       cpu
 r  b  w  avm  fre  flt  re  pi  po   fr   sr ad0 ad1   in   sy   cs us sy id
 0  0  0 514G  30G 4.0K   0   9   0 4.4K   37   0   0 1198  13K 3.9K  0  0 100
~~~

wulf7, 2023-10-24:
Try latest drm-515-kmod from sources. It should (about) fix memory accounting

magicalskye, 2023-10-24:
i thought the ports version was the newest one! but apparently not. the github version works indeed much better with fewer reality bends, thank you very much

evadot, 2023-10-24:
> Try latest drm-515-kmod from sources. It should (about) fix memory accounting

Thanks wulf !
I'll make a new release of ports tomorrow.

magicalskye, 2023-10-25:
appreciate it! thank you for your good work :)

closed 2023-10-25 by magicalskye