freebsd/drm-kmod #263: Panic & Hard Reset 6750XT

thedaemon, 2023-11-07:
**Describe the bug**
Panic / Hard Reset

**FreeBSD version**
FreeBSD thevoid 14.0-RC4 FreeBSD 14.0-RC4 #0 releng/14.0-n265374-d59533eaf942: Fri Nov  3 05:51:43 UTC 2023     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400097 1400097

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1450 subvendor=0x1022 subdevice=0x1450
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) Root Complex'
    class      = bridge
    subclass   = HOST-PCI
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib9@pci0:0:3:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1453 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:7:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib12@pci0:0:7:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1454 subvendor=0x1022 subdevice=0x1454
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B'
    class      = bridge
    subclass   = PCI-PCI
hostb6@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib13@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1454 subvendor=0x1022 subdevice=0x1454
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B'
    class      = bridge
[... 158 lines cut ...]
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
hdac1@pci0:13:0:3:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1457 subvendor=0x1849 subdevice=0x1202
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-0fh) HD Audio Controller'
    class      = multimedia
    subclass   = HDA

</details>

**DRM KMOD version**
Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`
drm-515-kmod 5.15.118_1

**To Reproduce**
Steps to reproduce the behavior: use picom, unplug Wacom One Tablet Monitor. I can play games and use Blender 3d software without any issues. Picom crashes me seemingly randomly and unplugging the Wacom One doesn't always crash me. Let me know what else you need from me.

**Screenshots**
None

**Additional context**
[Add any other context about the problem here.][https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=274770](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=274770)
Crash reported here with Core Dump Logs.

thedaemon, 2023-11-18:
I am testing 5.15-lts to see if that solves my problems.

evadot, 2024-01-05:
> I am testing 5.15-lts to see if that solves my problems.

Did it ?

thedaemon, 2024-02-09:
Yes, sorry, this
https://github.com/wulf7/drm-kmod/tree/5.15-lts

Is stable on my 6750XT.

the-sweeney-au, 2024-07-03:
The problem still persists as of drm-kmod version 5.15.160.
The bug will trigger randomly if left at default screens resolution, but triggers instantly if one try to change screen refresh setting (ie from 60Hz to 75Hz).
Problem is not present in drm-kmod 5.10

the-sweeney-au, 2024-07-20:
The issue is not present in drm-kmod 6.1.92. It appears to be a bug to be effecting drm-kmod 5.15.x only.