freebsd/drm-kmod #265: Kernel panic when unloading i915drm

khorben, 2023-11-13:
**Describe the bug**
When unloading the `i915drm` kernel module on a machine with supported Intel-based graphics, the system crashes and instantly reboots. (Black screen and then POST)

**FreeBSD version**

```
# uname -aKU
FreeBSD. 14.0-RC4 FreeBSD 14.0-RC4 #0 releng/14.0-n265374-d59533eaf942: Fri Nov  3 05:51:43 UTC 2023    root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400097 1400097
```

**PCI Info**

Intel HD Graphics 610 on an [ASRock H110M-ITX](https://www.asrock.com/mb/Intel/H110M-ITX/)

**DRM KMOD version**
Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`

**To Reproduce**
Steps to reproduce the behavior:

```
# pkg install drm-kmod
[...]
# kldload i915drm
# kldunload i915drm
```

thesunexpress, 2023-11-24:
errrmm.... 

For Intel: kld_list="i915kms"

evadot, 2024-01-05:
Yes this is known, drm unloading never really worked reliably for us and yes we should fix that one day (tm).

huanghwh, 2024-06-11:
so suspend/resume should be work?

evadot, 2024-06-11:
> so suspend/resume should be work?

Yes, suspend/resume have nothing to do with unloading.

wulf7, 2024-06-11:
> so suspend/resume should be work?

Suspend/resume generally does not work on modern hardware due to missing S0ix support

huanghwh, 2024-06-11:
I have macbookpro 2017, supend/resume works on drm-5.15, I try to make MateBook X Pro works under drm-6.1, but it is not, when suspend, the Caps LED is always on, never resume:(

grahamperrin, 2024-12-29:
> … when suspend, the Caps LED is always on, never resume:(

That sounds like a separate issue, not a kernel panic (system crash).