freebsd/drm-kmod #276: drm-515-kmod 5.15.118_3 panic on STABLE-14/amd64 built 27-Dec-2023

daemonblade, 2023-12-31:
**Describe the bug**

I'm getting a system panic when `kldload i915kms` on a version of STABLE-14 refreshed on 27-Dec-2023.

**FreeBSD version**

FreeBSD stormbringer 14.0-STABLE FreeBSD 14.0-STABLE #0 stable/14-n266056-70025e767f28: Wed Dec 27 09:56:50 NZDT 2023     root@stormbringer:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400502 1400502

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x4660 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x460d subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x16 Controller'
    class      = bridge
    subclass   = PCI-PCI
vgapci1@pci0:0:2:0:	class=0x038000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x4680 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'AlderLake-S GT1'
    class      = display
none0@pci0:0:4:0:	class=0x118000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x461d subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake Innovation Platform Framework Processor Participant'
    class      = dasp
pcib2@pci0:0:6:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x464d subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x4 Controller'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:8:0:	class=0x088000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x464f subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor Gaussian & Neural Accelerator'
    class      = base peripheral
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7ae0 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH USB 3.2 Gen 2x2 XHCI Controller'
    class      = serial bus
    subclass   = USB
none2@pci0:0:20:2:	class=0x050000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7aa7 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH Shared SRAM'
    class      = memory
    subclass   = RAM
ig4iic0@pci0:0:21:0:	class=0x0c8000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7acc subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH Serial IO I2C Controller'
    class      = serial bus
none3@pci0:0:22:0:	class=0x078000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7ae8 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH HECI Controller'
    class      = simple comms
none4@pci0:0:22:3:	class=0x070002 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7aeb subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    class      = simple comms
    subclass   = UART
ahci0@pci0:0:23:0:	class=0x010601 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7ae2 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-S PCH SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
isab0@pci0:0:31:0:	class=0x060100 rev=0x11 hdr=0x00 vendor=0x8086 device=0x7a83 subvendor=0x1028 subdevice=0x0ac1
    vendor     = 'Intel Corporation'
    class      = bridge
[... 70 lines cut ...]
Dec 28 07:54:47 stormbringer kernel: r10: 0000000000530020 r11: ffffffffffffa9c7 r12: fffffe01a5568580
Dec 28 07:54:47 stormbringer kernel: r13: fffff800184d0000 r14: 0000000000000000 r15: fffffe01a95bcda8
Dec 28 07:54:47 stormbringer kernel: trap number        = 12
Dec 28 07:54:47 stormbringer kernel: panic: page fault
Dec 28 07:54:47 stormbringer kernel: cpuid = 8
Dec 28 07:54:47 stormbringer kernel: time = 1703703168
Dec 28 07:54:47 stormbringer kernel: KDB: stack backtrace:
Dec 28 07:54:47 stormbringer kernel: #0 0xffffffff80b9208d at kdb_backtrace+0x5d
Dec 28 07:54:47 stormbringer kernel: #1 0xffffffff80b45122 at vpanic+0x132
Dec 28 07:54:47 stormbringer kernel: #2 0xffffffff80b44fe3 at panic+0x43
Dec 28 07:54:47 stormbringer kernel: #3 0xffffffff8101294c at trap_fatal+0x40c
Dec 28 07:54:47 stormbringer kernel: #4 0xffffffff8101299f at trap_pfault+0x4f
Dec 28 07:54:47 stormbringer kernel: #5 0xffffffff80fe8dc8 at calltrap+0x8
Dec 28 07:54:47 stormbringer kernel: #6 0xffffffff80dc0d9c at xa_next+0xbc
Dec 28 07:54:47 stormbringer kernel: #7 0xffffffff84335011 at intel_guc_submission_cancel_requests+0x31
Dec 28 07:54:47 stormbringer kernel: #8 0xffffffff8431df59 at __intel_gt_set_wedged+0x389
Dec 28 07:54:47 stormbringer kernel: #9 0xffffffff8431db64 at intel_gt_set_wedged+0x74
Dec 28 07:54:47 stormbringer kernel: #10 0xffffffff84314a73 at intel_gt_resume+0x5f3
Dec 28 07:54:47 stormbringer kernel: #11 0xffffffff84312113 at intel_gt_init+0x213
Dec 28 07:54:47 stormbringer kernel: #12 0xffffffff8420c652 at i915_gem_init+0x92
Dec 28 07:54:47 stormbringer kernel: #13 0xffffffff84208bf6 at i915_driver_probe+0xd56
Dec 28 07:54:47 stormbringer kernel: #14 0xffffffff84219cd0 at i915_pci_probe+0x40
Dec 28 07:54:47 stormbringer kernel: #15 0xffffffff80db4b40 at linux_pci_attach_device+0x420
Dec 28 07:54:47 stormbringer kernel: #16 0xffffffff80b81a7e at device_attach+0x3be
Dec 28 07:54:47 stormbringer kernel: #17 0xffffffff80b837c1 at bus_generic_driver_added+0xa1

evadot, 2024-01-05:
Is this a new issue ? Did it worked before ?

daemonblade, 2024-01-05:
I only upgraded to STABLE-14 in the past week or so, and it's the first time I've tried it.
Yesterday, I updated to: **drm-kmod-20220907_1** and **drm-515-kmod-5.15.118_3**, and the logs this time are just a _bit_ different:
```
Jan  4 08:18:34 stormbringer kernel: iic0: <I2C generic I/O> on iicbus0
Jan  4 08:18:34 stormbringer kernel: drmn1: <drmn> on vgapci1
Jan  4 08:18:34 stormbringer kernel: vgapci1: child drmn1 requested pci_enable_io
Jan  4 08:18:34 stormbringer syslogd: last message repeated 1 times
Jan  4 08:18:34 stormbringer kernel: drmn1: [drm] Incompatible option enable_guc=-1 - undocumented flag
Jan  4 08:18:34 stormbringer kernel: [drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
Jan  4 08:18:34 stormbringer kernel: [drm] Got stolen memory base 0x6c800000, size 0x3c00000
Jan  4 08:18:34 stormbringer kernel: lkpi_iic0: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus1: <Philips I2C bus> on lkpi_iic0
Jan  4 08:18:34 stormbringer kernel: iic1: <I2C generic I/O> on iicbus1
Jan  4 08:18:34 stormbringer kernel: lkpi_iic1: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus2: <Philips I2C bus> on lkpi_iic1
Jan  4 08:18:34 stormbringer kernel: iic2: <I2C generic I/O> on iicbus2
Jan  4 08:18:34 stormbringer kernel: lkpi_iic2: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus3: <Philips I2C bus> on lkpi_iic2
Jan  4 08:18:34 stormbringer kernel: iic3: <I2C generic I/O> on iicbus3
Jan  4 08:18:34 stormbringer kernel: lkpi_iic3: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus4: <Philips I2C bus> on lkpi_iic3
Jan  4 08:18:34 stormbringer kernel: iic4: <I2C generic I/O> on iicbus4
Jan  4 08:18:34 stormbringer kernel: lkpi_iic4: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus5: <Philips I2C bus> on lkpi_iic4
Jan  4 08:18:34 stormbringer kernel: iic5: <I2C generic I/O> on iicbus5
Jan  4 08:18:34 stormbringer kernel: lkpi_iic5: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus6: <Philips I2C bus> on lkpi_iic5
Jan  4 08:18:34 stormbringer kernel: iic6: <I2C generic I/O> on iicbus6
Jan  4 08:18:34 stormbringer kernel: lkpi_iic6: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus7: <Philips I2C bus> on lkpi_iic6
Jan  4 08:18:34 stormbringer kernel: iic7: <I2C generic I/O> on iicbus7
Jan  4 08:18:34 stormbringer kernel: lkpi_iic7: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus8: <Philips I2C bus> on lkpi_iic7
Jan  4 08:18:34 stormbringer kernel: iic8: <I2C generic I/O> on iicbus8
Jan  4 08:18:34 stormbringer kernel: lkpi_iic8: <LinuxKPI I2C> on drmn1
Jan  4 08:18:34 stormbringer kernel: iicbus9: <Philips I2C bus> on lkpi_iic8
Jan  4 08:18:34 stormbringer kernel: iic9: <I2C generic I/O> on iicbus9
Jan  4 08:18:34 stormbringer kernel: drmn1: successfully loaded firmware image 'i915/adls_dmc_ver2_01.bin'
Jan  4 08:18:34 stormbringer kernel: drmn1: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
Jan  4 08:18:34 stormbringer kernel: drmn1: successfully loaded firmware image 'i915/tgl_guc_62.0.0.bin'
Jan  4 08:18:34 stormbringer kernel: drmn1: successfully loaded firmware image 'i915/tgl_huc_7.9.3.bin'
Jan  4 08:19:21 stormbringer syslogd: kernel boot file is /boot/kernel/kernel
Jan  4 08:19:21 stormbringer kernel: [drm ERROR :intel_huc_auth] HuC: Firmware not verified -60
Jan  4 08:19:21 stormbringer kernel: drmn1: 0xfffffe01aa2ce608Vdrmn1: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
Jan  4 08:19:21 stormbringer kernel: 
Jan  4 08:19:21 stormbringer kernel: Fatal trap 12: page fault while in kernel m
```

evadot, 2024-11-12:
So this is a duplicate of https://github.com/freebsd/drm-kmod/issues/252 and this is fixed with 6.1-lts. Closing this one.

closed 2024-11-12 by evadot