freebsd/drm-kmod #277: graphics/drm-61-kmod amdgpu panic: Unregistered use of FPU in kernel

jownit, 2024-01-10:
**Describe the bug**
Panics when loading amdgpu.ko

**FreeBSD version**
FreeBSD joxan 15.0-CURRENT FreeBSD 15.0-CURRENT #123 main-n267479-13720136fbf9: Wed Jan 10 09:26:10 CET 2024     root@joxan:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64 1500008 1500008

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:1:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:1:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x5126 subdevice=0x17aa
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
[... 194 lines cut ...]
dcn_bw_sync_calcs_and_dml() at dcn_bw_sync_calcs_and_dml+0xb/frame 0xfffffe0100b3b2f0
dcn10_create_resource_pool() at dcn10_create_resource_pool+0x6a6/frame 0xfffffe0100b3b490
dc_create_resource_pool() at dc_create_resource_pool+0x4c/frame 0xfffffe0100b3b4b0
dc_create() at dc_create+0x330/frame 0xfffffe0100b3b4f0
dm_hw_init() at dm_hw_init+0x3d9/frame 0xfffffe0100b3b6e0
amdgpu_device_ip_hw_init_phase2() at amdgpu_device_ip_hw_init_phase2+0x5a/frame 0xfffffe0100b3b710
amdgpu_device_ip_init() at amdgpu_device_ip_init+0x370/frame 0xfffffe0100b3b790
amdgpu_device_init() at amdgpu_device_init+0x1cdb/frame 0xfffffe0100b3b850
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0x16/frame 0xfffffe0100b3b880
amdgpu_pci_probe() at amdgpu_pci_probe+0x283/frame 0xfffffe0100b3b8c0
linux_pci_attach_device() at linux_pci_attach_device+0x478/frame 0xfffffe0100b3b910
device_attach() at device_attach+0x3b5/frame 0xfffffe0100b3b960
bus_generic_driver_added() at bus_generic_driver_added+0xa1/frame 0xfffffe0100b3b990
devclass_driver_added() at devclass_driver_added+0x39/frame 0xfffffe0100b3b9d0
devclass_add_driver() at devclass_add_driver+0x11e/frame 0xfffffe0100b3ba10
_linux_pci_register_driver() at _linux_pci_register_driver+0xcc/frame 0xfffffe0100b3ba40
amdgpu_evh() at amdgpu_evh+0x80/frame 0xfffffe0100b3ba50
module_register_init() at module_register_init+0x85/frame 0xfffffe0100b3ba80
linker_load_module() at linker_load_module+0xbf9/frame 0xfffffe0100b3bd70
kern_kldload() at kern_kldload+0x16a/frame 0xfffffe0100b3bdd0
sys_kldload() at sys_kldload+0x5c/frame 0xfffffe0100b3be00
amd64_syscall() at amd64_syscall+0x109/frame 0xfffffe0100b3bf30
fast_syscall_common() at fast_syscall_common+0xf8/frame 0xfffffe0100b3bf30
--- syscall (304, FreeBSD ELF64, kldload), rip = 0x32d5ff8d767a, rsp = 0x32d5fe406be8, rbp = 0x32d5fe407160 ---
KDB: enter: panic

wulf7, 2024-02-01:
Test this patch:
```
diff --git a/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c b/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
index e73f089c84..a4c1e94f79 100644
--- a/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
+++ b/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
@@ -1561,6 +1561,7 @@ void dcn_bw_notify_pplib_of_wm_ranges(
 
 void dcn_bw_sync_calcs_and_dml(struct dc *dc)
 {
+	DC_FP_START();
 	DC_LOG_BANDWIDTH_CALCS("sr_exit_time: %f ns\n"
 			"sr_enter_plus_exit_time: %f ns\n"
 			"urgent_latency: %f ns\n"
@@ -1697,6 +1698,7 @@ void dcn_bw_sync_calcs_and_dml(struct dc *dc)
 			dc->dcn_ip->can_vstartup_lines_exceed_vsync_plus_back_porch_lines_minus_one,
 			dc->dcn_ip->bug_forcing_luma_and_chroma_request_to_same_size_fixed,
 			dc->dcn_ip->dcfclk_cstate_latency);
+	DC_FP_END();
 
 	dc->dml.soc.sr_exit_time_us = dc->dcn_soc->sr_exit_time;
 	dc->dml.soc.sr_enter_plus_exit_time_us = dc->dcn_soc->sr_enter_plus_exit_time;
```

jownit, 2024-02-02:
I tried the patch, but unfortunately it does not make any difference.

panic: Unregistered use of FPU in kernel
cpuid = 5
time = 1706869368
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe0100aaff60
vpanic() at vpanic+0x135/frame 0xfffffe0100ab0090
panic() at panic+0x43/frame 0xfffffe0100ab00f0
trap() at trap+0x8dc/frame 0xfffffe0100ab0210
calltrap() at calltrap+0x8/frame 0xfffffe0100ab0210
--- trap 0x16, rip = 0xffffffff84001001, rsp = 0xfffffe0100ab02e8, rbp = 0xfffffe0100ab02f0 ---
dcn_bw_sync_calcs_and_dml() at dcn_bw_sync_calcs_and_dml+0x31/frame 0xfffffe0100ab02f0
dcn10_create_resource_pool() at dcn10_create_resource_pool+0x6a6/frame 0xfffffe0100ab0490
dc_create_resource_pool() at dc_create_resource_pool+0x4c/frame 0xfffffe0100ab04b0
dc_create() at dc_create+0x330/frame 0xfffffe0100ab04f0
dm_hw_init() at dm_hw_init+0x3d9/frame 0xfffffe0100ab06e0
amdgpu_device_ip_hw_init_phase2() at amdgpu_device_ip_hw_init_phase2+0x5a/frame 0xfffffe0100ab0710
amdgpu_device_ip_init() at amdgpu_device_ip_init+0x370/frame 0xfffffe0100ab0790
amdgpu_device_init() at amdgpu_device_init+0x1cdb/frame 0xfffffe0100ab0850
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0x16/frame 0xfffffe0100ab0880
amdgpu_pci_probe() at amdgpu_pci_probe+0x283/frame 0xfffffe0100ab08c0
linux_pci_attach_device() at linux_pci_attach_device+0x478/frame 0xfffffe0100ab0910
device_attach() at device_attach+0x3b5/frame 0xfffffe0100ab0960
bus_generic_driver_added() at bus_generic_driver_added+0xa1/frame 0xfffffe0100ab0990
devclass_driver_added() at devclass_driver_added+0x39/frame 0xfffffe0100ab09d0
devclass_add_driver() at devclass_add_driver+0x11e/frame 0xfffffe0100ab0a10
_linux_pci_register_driver() at _linux_pci_register_driver+0xcc/frame 0xfffffe0100ab0a40
amdgpu_evh() at amdgpu_evh+0x80/frame 0xfffffe0100ab0a50
module_register_init() at module_register_init+0x85/frame 0xfffffe0100ab0a80
linker_load_module() at linker_load_module+0xbf9/frame 0xfffffe0100ab0d70
kern_kldload() at kern_kldload+0x16a/frame 0xfffffe0100ab0dd0
sys_kldload() at sys_kldload+0x5c/frame 0xfffffe0100ab0e00
amd64_syscall() at amd64_syscall+0x109/frame 0xfffffe0100ab0f30
fast_syscall_common() at fast_syscall_common+0xf8/frame 0xfffffe0100ab0f30
--- syscall (304, FreeBSD ELF64, kldload), rip = 0x331be343b69a, rsp = 0x331be1d001c8, rbp = 0x331be1d00740 ---
KDB: enter: panic

wulf7, 2024-02-02:
Ok. Then let`s increase DC_FP scope:
```
diff --git a/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c b/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
index e73f089c84..22613780c3 100644
--- a/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
+++ b/drivers/gpu/drm/amd/display/dc/dml/calcs/dcn_calcs.c
@@ -1561,6 +1561,7 @@ void dcn_bw_notify_pplib_of_wm_ranges(
 
 void dcn_bw_sync_calcs_and_dml(struct dc *dc)
 {
+	DC_FP_START();
 	DC_LOG_BANDWIDTH_CALCS("sr_exit_time: %f ns\n"
 			"sr_enter_plus_exit_time: %f ns\n"
 			"urgent_latency: %f ns\n"
@@ -1749,4 +1750,5 @@ void dcn_bw_sync_calcs_and_dml(struct dc *dc)
 	dc->dml.ip.bug_forcing_LC_req_same_size_fixed =
 		dc->dcn_ip->bug_forcing_luma_and_chroma_request_to_same_size_fixed == dcn_bw_yes;
 	dc->dml.ip.dcfclk_cstate_latency = dc->dcn_ip->dcfclk_cstate_latency;
+	DC_FP_END();
 }
```

jownit, 2024-02-02:
That made a difference!
No panic, and my external display works.
Looks good so far, X11 starts and looks like before, when using drm-510-kmod

alfonsosiciliano, 2024-02-08:
Also runs for:
Device: PCI 1002:15d8 Advanced Micro Devices, Inc. [AMD/ATI] Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series]

Thanks!

lbartoletti, 2024-02-09:
(Maybe a different problem? But,) I also have a panic using drm-61-kmod on FreeBSD 15.0

[image]

It's an APU (raphael) on aAMD Ryzen 9 7900X 

b.f.o issue for reference: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=268394#c10

alfonsosiciliano, 2024-02-09:
> (Maybe a different problem? But,) I also have a panic using drm-61-kmod on FreeBSD 15.0
> 
> ![amd_panic](https://private-user-images.githubusercontent.com/7521540/303616312-350f18f5-83b0-46d1-967c-0729c0cd34ce.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0Nzc2NTYsIm5iZiI6MTcwNzQ3NzM1NiwicGF0aCI6Ii83NTIxNTQwLzMwMzYxNjMxMi0zNTBmMThmNS04M2IwLTQ2ZDEtOTY3Yy0wNzI5YzBjZDM0Y2UuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMTExNTU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZiMDJlZDJmNzYxYzk4MmJjNDhjOTA0YjMzMjA3NzdjOTIwNDk5NGM5ZGRkOGNhZjU5MzExNThhMzQwNDZjMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.rPnmnrTH3xb-UNf7oDFIwN018FubGtIaLpEgrUhp3RQ)
> 
> It's an APU (raphael) on aAMD Ryzen 9 7900X
> 
> b.f.o issue for reference: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=268394#c10

Sorry, unfortunately I can't "read" the image.

Is the problem:  "panic: Unregistered use of FPU"?
Do core file and kgdb backtrace refer to the "dcn_bw_sync_calcs_and_dml()" function?
Does the problem occur after the patch "DC_FP_START\END()"?

wulf7, 2024-02-11:
I pushed slightly different version of the patch to 6.1-lts branch. If it still works we can update drm-61-kmod port

wulf7, 2024-02-11:
> (Maybe a different problem? But,) I also have a panic using drm-61-kmod on FreeBSD 15.0
> 
> It's an APU (raphael) on aAMD Ryzen 9 7900X
> 
> b.f.o issue for reference: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=268394#c10

That is different panic.

What you see is a driver unloading bug (misplaced vm_phys_fictitious_unreg_range() call). It is triggered by a fatal error (not by the driver crash!) during initialization. You should check message buffer content rather than backtrace in that case. Most probably the reason of error is unsupported GPU or missing firmware module.
Try to install all FLAVORs of graphics/gpu-firmware-amd-kmod and if it does not help you may try WIP 6.6 branch.

lbartoletti, 2024-03-19:
> That is different panic.
>
> What you see is a driver unloading bug (misplaced vm_phys_fictitious_unreg_range() call). It is triggered by a fatal error (not by the driver crash!) during initialization. [...]

Thanks, and sorry for my late answer.

Different problem now with 61-lts, I can load amdgpu, but now, I have black screen.

I was unable to compile the 6.6 branch due to a missing header:

evadot, 2024-06-04:
Closing as the original bug is fixed.

closed 2024-06-04 by evadot