freebsd/drm-kmod #281: 6.1-lts destroys my dmesg

kevans91, 2024-01-23:
**Describe the bug**
I didn't check if 5.10-lts was doing the same, but I get hundreds of this spew per minute:

```
WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const volatile __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) *)&(((volatile const unsigned long *)(&ce->flags))[((12) / 64)])); }); __asm__ __volatile__("": : :"memory"); __var; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:217
```

This one is also seemingly reported with some frequently:

```
Jan 23 19:38:27 ifrit kernel: r; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:217 
```

To be precise, here's a ~1-minute window from syslogd:
```
1172 Jan 23 19:36:44 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1173 Jan 23 19:37:02 ifrit syslogd: last message repeated 351 times                  
1174 Jan 23 19:37:02 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1175 Jan 23 19:37:02 ifrit kernel: 12) / 64)]) *)&(((volatile const unsigned long *)(&ce->flags))[((12) / 64)])); }); __asm__ __volatile__("": : :"memory"); __var; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/m
1176 Jan 23 19:37:03 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1177 Jan 23 19:37:20 ifrit syslogd: last message repeated 78 times                   
1178 Jan 23 19:37:20 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) 
1179 Jan 23 19:37:20 ifrit kernel: __var = ({ __asm__ __volatile__("": : :"memory"); (*(const volatile __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) *)&(((volatile const unsigned long *)(&ce->flags))
1180 Jan 23 19:37:20 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1181 Jan 23 19:37:29 ifrit syslogd: last message repeated 595 times                  
1182 Jan 23 19:37:29 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1183 Jan 23 19:37:29 ifrit kernel: r; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:217
1184 Jan 23 19:37:29 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1185 Jan 23 19:37:41 ifrit syslogd: last message repeated 282 times                  
1186 Jan 23 19:37:41 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1187 Jan 23 19:37:41 ifrit kernel: 12) / 64)]) *)&(((volatile const unsigned long *)(&ce->flags))[((12) / 64)])); }); __asm__ __volatile__("": : :"memory"); __var; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/m
1188 Jan 23 19:37:41 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1189 Jan 23 19:37:43 ifrit syslogd: last message repeated 142 times                  
1190 Jan 23 19:37:43 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1191 Jan 23 19:37:43 ifrit kernel: r; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:217
1192 Jan 23 19:37:43 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1193 Jan 23 19:37:50 ifrit syslogd: last message repeated 308 times                  
1194 Jan 23 19:37:50 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1195 Jan 23 19:37:50 ifrit kernel: r; }) & (1UL << ((12) & (64 - 1)))) failed at /usr/local/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:217
1196 Jan 23 19:37:50 ifrit kernel: WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const vol
1197 Jan 23 19:38:02 ifrit syslogd: last message repeated 388 times                  
```

**FreeBSD version**
FreeBSD 15.0-CURRENT FreeBSD 15.0-CURRENT kbsd/next-n267365-8aa2734d89d4 GENERIC amd64 1500008 1500008

**PCI Info**

```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a14 subvendor=0xf111 subdevice=0x0001
    vendor     = 'Intel Corporation'
    device     = '11th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a49 subvendor=0xf111 subdevice=0x0001
    vendor     = 'Intel Corporation'
    device     = 'TigerLake-LP GT2 [Iris Xe Graphics]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a03 subvendor=0xf111 subdevice=0x0001
    vendor     = 'Intel Corporation'
    device     = 'TigerLake-LP Dynamic Tuning Processor Participant'
    class      = dasp
pcib1@pci0:0:6:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a09 subvendor=0xf111 subdevice=0x0001
    vendor     = 'Intel Corporation'
    device     = '11th Gen Core Processor PCIe Controller'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a23 subvendor=0xf111 subdevice=0x0001
    vendor     = 'Intel Corporation'
[... 94 lines cut ...]
    device     = 'Tiger Lake-LP SPI Controller'
    class      = serial bus
nvme0@pci0:1:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x15b7 device=0x5006 subvendor=0x15b7 subdevice=0x5006
    vendor     = 'Sandisk Corp'
    device     = 'SanDisk Extreme Pro / WD Black SN750 / PC SN730 / Red SN700 NVMe SSD'
    class      = mass storage
    subclass   = NVM
iwlwifi0@pci0:170:0:0:	class=0x028000 rev=0x1a hdr=0x00 vendor=0x8086 device=0x2725 subvendor=0x8086 subdevice=0x0024
    vendor     = 'Intel Corporation'
    device     = 'Wi-Fi 6 AX210/AX211/AX411 160MHz'
    class      = network
```

**DRM KMOD version**
```
commit b66c0a09276885836524a9724bab5d053891ddd2 (HEAD -> 6.1-lts, tag: drm_v6.1.69, origin/6.1-lts)
Author: Ville Syrjälä <ville.syrjala@linux.intel.com>
Date:   Tue Dec 5 20:03:08 2023 +0200
```

**To Reproduce**
Boot my laptop

**Additional context**
This laptop can't boot with i915kms on Linux, so I've no path for trying to bisect along upstream drm to see if it happens / has happened there / where it might be fixed

emaste, 2024-02-01:
I also get it on the 6.6 WIP

`WARNING !lockdep_is_held(&ce->timeline->mutex) || !!(({ __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) __var = ({ __asm__ __volatile__("": : :"memory"); (*(const volatile __typeof(((volatile const unsigned long *)(&ce->flags))[((12) / 64)]) *)&(((volatile const unsigned long *)(&ce->flags))[((12) / 64)])); }); __asm__ __volatile__("": : :"memory"); __var; }) & (1UL << ((12) & (64 - 1)))) failed at /root/freebsd/sys/modules/drm-kmod/drivers/gpu/drm/i915/gt/intel_context.h:221`

emaste, 2024-06-05:
Note that `lockdep_assert` are only under `INVARIANTS`, if building for a non-`INVARIANTS` kernel will not print these warnings.

emaste, 2024-06-06:
Fixed by https://github.com/freebsd/freebsd-src/commit/21f4cf5ccf87fd15d4d71f89d4c3ecab13f26146

emaste, 2024-06-06:
(Issue should be closed -- I don't have permission.)

closed 2024-06-07 by kevans91