freebsd/drm-kmod #284: drm errors with drm-61-kmod and i915 Inter Raptor Lake P

rosorio, 2024-02-05:
**Describe the bug**
Even if the video management is quite acceptable, I still having a lot of errors like this
drmn0: [drm] *ERROR* Fault errors on pipe A: 0x00000080
drmn0: [drm] *ERROR* Fault errors on pipe B: 0x00000080
drmn0: [drm] *ERROR* Fault errors on pipe C: 0x00000080
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.

**FreeBSD version**
FreeBSD 15.0-CURRENT amd64 (1500011 c4ded83c196b91e65b20210b99ca54208ca5d20e)

**DRM KMOD version**
drm-61-kmod-6.1.69

cperciva, 2024-04-02:
I have screen flickering when running Chrome and the same `Timed out waiting for DSB workload completion.` error, but I don't have the "Fault errors on pipe" messages.

Running drm-61-kmod-6.1.69_2 on FreeBSD 14.0-STABLE (n267029-0cdc80d26615).

rosorio, 2024-04-07:
Since a week I start seeing new behaviors with firefox having the config gfx.webrender.all at true
Firefox start to freeze from time to time and the console report swap/memory/drm issues.
The memory/swap issues are quite strange because my laptop was never in memory or swap starvation

For the record I'm currently running FreeBSD 15.0-CURRENT main-n269057-dc831e93bad6 GENERIC amd64

drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
witness_lock_list_get: witness exhausted
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[101670] context reset due to GPU hang
drmn0: [drm] GPU HANG: ecode 12:1:002b5fff, in Renderer [101670]
drmn0: [drm] GPU HANG: ecode 12:1:0061ffff, in Renderer [113955]
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[113955] context reset due to GPU hang
swap_pager: out of swap space
swp_pager_getswapspace(20): failed
swp_pager_getswapspace(7): failed
swap_pager: out of swap space
swp_pager_getswapspace(9): failed
pid 12091 (firefox), jid 0, uid 1001, was killed: failed to reclaim memory
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] GPU HANG: ecode 12:1:86d79ff8, in Renderer [190053]
drmn0: [drm] Resetting rcs0 for stopped heartbeat on rcs0
drmn0: [drm] Renderer[190053] context reset due to GPU hang
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[199330] context reset due to GPU hang
drmn0: [drm] GPU HANG: ecode 12:1:00201001, in Renderer [199330]
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[214106] context reset due to GPU hang
drmn0: [drm] GPU HANG: ecode 12:1:00201001, in Renderer [214106]
drmn0: [drm] GPU HANG: ecode 12:1:00251fff, in Renderer [220660]
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[220660] context reset due to GPU hang
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] *ERROR* Timed out waiting for DSB workload completion.
drmn0: [drm] Resetting rcs0 for CS error
drmn0: [drm] Renderer[221033] context reset due to GPU hang
drmn0: [drm] GPU HANG: ecode 12:1:00233fff, in Renderer [221033]

adventureloop, 2024-05-02:
```
FreeBSD computer 15.0-CURRENT FreeBSD 15.0-CURRENT main-n269723-4ba444de708b GENERIC amd64
```

I have these errors as well on my Framework 13 12th generation Intel. 

I get the blanking when triggering full draw events (like a drop down terminal) or moving my mouse to the edge of the screen quickly in kde. It is worse on the internal panel than on an external display.

hodong-kim, 2024-05-05:
I am also experiencing the same bug.
I use Asus PRIME N100I-D D4-CSM motherboard.

```
hodong@:~ $ pkg info drm-61-kmod
drm-61-kmod-6.1.69_2
Name           : drm-61-kmod
Version        : 6.1.69_2
Installed on   : Sun May  5 07:00:23 2024 KST
Origin         : graphics/drm-61-kmod
Architecture   : FreeBSD:14:amd64
Prefix         : /usr/local
Categories     : kld graphics
Licenses       : MIT and GPLv2 and BSD2CLAUSE
Maintainer     : x11@FreeBSD.org
WWW            : https://github.com/freebsd/drm-kmod/
Comment        : DRM drivers modules
Annotations    :
    FreeBSD_version: 1401000
Flat size      : 17.0MiB
Description    :
amdgpu, i915, and radeon DRM drivers modules.
Currently corresponding to Linux 6.1 DRM.
This version is for FreeBSD 14-STABLE 1400508
and above.

hodong@:~ $ uname -a
FreeBSD  14.1-BETA1 FreeBSD 14.1-BETA1 #0 releng/14.1-n267603-3f5f38755b8f: Fri May  3 13:20:07 UTC 2024     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
```

rosorio, 2024-06-11:
Last drm-kmod drivers are astonishing, thanks for your hard work!

closed 2024-06-11 by rosorio

hodong-kim, 2024-06-19:
The same bug occurs in version 6.1.92. So I ended up replacing the motherboard with a J5040.

freebsdcross, 2024-06-26:
I am seeing this as well; I think this needs a re-open

huanghwh, 2024-06-27:
after switch to wanland+sway and use updated mesa, no errors any more

% pkg info -x drm
drm-61-kmod-6.1.92
libdrm-2.4.121,1
% pkg info -x mesa
libosmesa-24.0.9_1
mesa-dri-24.0.9
mesa-libs-24.0.9

freebsdcross, 2024-06-27:
For me this was with a local poudriere build with ports tree checked out June 21:
drm-61-kmod-6.1.92
libdrm-2.4.121,1
libosmesa-24.0.9
mesa-dri-24.0.9
mesa-libs-24.0.9

on 14.1-RELEASE, running gnome under X.

I know a number of people have pointed to 'it doesn't happen under wayland, use wayland'.

Wayland uses the same underlying drivers, it may be (for now) avoiding the problematic code path for various reasons, that doesn't mean it still isn't there and won't come up.

A number of the desktop environments just don't work in wayland currently (gnome under freebsd doesn't work at all, I've tried repeatedly to get KDE5 to go to no success, and KDE6 is beta currently).

Wayland needs to rely on the x emulated server for a large number of apps, and it doesn't work right for many (especially those relying on color profiles).

freebsdcross, 2024-06-27:
I am re-opening this under a new issue

freebsdcross, 2024-06-27:
https://github.com/freebsd/drm-kmod/issues/310