freebsd/drm-kmod #287: kernel panic with drm-515-kmod

Aminavy, 2024-02-17:
During play a video in firefox or use accelerated graphics panic occurred

**FreeBSD version**
FreeBSD genesis 14.0-RELEASE-p5 FreeBSD 14.0-RELEASE-p5 #0: Tue Feb 13 23:37:36 UTC 2024     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400097 1400097

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x17aa subdevice=0x3814
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x1022 subdevice=0x1234
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:5:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x1022 subdevice=0x1234
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x1022 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x3838
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1022 subdevice=0x790e
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e9 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 1'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:24:2:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 2'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:24:3:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15eb subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 3'
    class      = bridge
    subclass   = HOST-PCI
hostb7@pci0:0:24:4:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ec subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 4'
    class      = bridge
    subclass   = HOST-PCI
hostb8@pci0:0:24:5:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ed subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 5'
    class      = bridge
    subclass   = HOST-PCI
hostb9@pci0:0:24:6:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ee subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 6'
    class      = bridge
    subclass   = HOST-PCI
hostb10@pci0:0:24:7:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ef subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 7'
    class      = bridge
    subclass   = HOST-PCI
none1@pci0:1:0:0:	class=0x028000 rev=0x32 hdr=0x00 vendor=0x168c device=0x003e subvendor=0x17aa subdevice=0x0827
    vendor     = 'Qualcomm Atheros'
    device     = 'QCA6174 802.11ac Wireless Network Adapter'
    class      = network
nvme0@pci0:2:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x144d device=0xa809 subvendor=0x144d subdevice=0xa801
    vendor     = 'Samsung Electronics Co Ltd'
    device     = 'NVMe SSD Controller 980 (DRAM-less)'
    class      = mass storage
    subclass   = NVM
vgapci0@pci0:3:0:0:	class=0x030000 rev=0xde hdr=0x00 vendor=0x1002 device=0x15d8 subvendor=0x17aa subdevice=0x380a
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series]'
    class      = display
    subclass   = VGA
hdac0@pci0:3:0:1:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0x15de subvendor=0x17aa subdevice=0x3804
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Raven/Raven2/Fenghuang HDMI/DP Audio Controller'
    class      = multimedia
    subclass   = HDA
none2@pci0:3:0:2:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15df subvendor=0x17aa subdevice=0x3807
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) Platform Security Processor'
    class      = encrypt/decrypt
xhci0@pci0:3:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e5 subvendor=0x17aa subdevice=0x3809
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven2 USB 3.1'
    class      = serial bus
    subclass   = USB
none3@pci0:3:0:5:	class=0x048000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e2 subvendor=0x17aa subdevice=0x3804
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'ACP/ACP3X/ACP6x Audio Coprocessor'
    class      = multimedia
hdac1@pci0:3:0:6:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x17aa subdevice=0x3803
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h/19h HD Audio Controller'
    class      = multimedia
    subclass   = HDA

</details>

**DRM KMOD version**
drm-510-kmod 5.10.163_8

evadot, 2024-02-17:
Having the panic message and backtrace will help.

Aminavy, 2024-02-18:
I can't saw panic message because X server was running when kernel panic occurred.

NorwegianRockCat, 2024-02-25:
I can say that I'm seeing this too with 5.15 and an 5700 XT. I will say it doesn't panic. The screen goes blank and the just reboots. So, no core is generated. Is some way to catch this crash and dump core? I could try as it seems to relatively simple to reproduce.

Went back to the 5.10 kmod and it doesn't have this problem with video.

evadot, 2024-04-23:
> I can't saw panic message because X server was running when kernel panic occurred.

Please configure your system so crash dumps are saved.