freebsd/drm-kmod #288: ERROR :drm_atomic_helper_wait_for_flip_done, primary screen complete lockup, no updates, system reboots randomly

nerozero, 2024-02-26:
**Describe the bug**
After installing Intel Display drivers notebook become absolutely unstable, random screen freeze even during boot process, system lockup, random reboot, extremely slow response to keyboard (cli mode), still possible to access system via ssh

**FreeBSD version**
FreeBSD latnz 14.0-RELEASE-p5 FreeBSD 14.0-RELEASE-p5 #0: Tue Feb 13 23:37:36 UTC 2024     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400097 1400097

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
ig4iic1@pci0:0:21:1:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d61 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
ig4iic2@pci0:0:21:2:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d62 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
ig4iic3@pci0:0:21:3:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d63 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:	class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:2:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d12 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:29:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d18 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d4e subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point LPC Controller/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none2@pci0:0:31:2:	class=0x058000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d21 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PMC'
    class      = memory
hdac0@pci0:0:31:3:	class=0x040380 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d71 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP HD Audio'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d23 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:	class=0x020000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x15d7 subvendor=0x1028 subdevice=0x081c
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (4) I219-LM'
    class      = network
    subclass   = ethernet
rtsx0@pci0:1:0:0:	class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x1028 subdevice=0x081c
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS525A PCI Express Card Reader'
iwm0@pci0:2:0:0:	class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095a subvendor=0x8086 subdevice=0x5010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 7265'
    class      = network
nvme0@pci0:3:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x1c5c device=0x1327 subvendor=0x1c5c subdevice=0x0000
    vendor     = 'SK hynix'
    device     = 'BC501 NVMe Solid State Drive'
    class      = mass storage
    subclass   = NVM

</details>

**DRM KMOD version**
drm-515-kmod 5.15.118_3

**To Reproduce**

install 
```sh
pkg install drm-515-kmod gpu-firmware-intel-kmod-kabylake
sysrc kld_list+=i915kms
shutdown -r now
```
( cpu:  [i5-8250U](https://ark.intel.com/content/www/us/en/ark/products/124967/intel-core-i5-8250u-processor-6m-cache-up-to-3-40-ghz.html) )

**Additional context**

content of /var/log/messages:
```
Feb 26 06:37:02 latnz kernel: drmn0: successfully loaded firmware image 'i915/kbl_dmc_ver1_04.bin'
Feb 26 06:37:02 latnz kernel: drmn0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
Feb 26 06:37:02 latnz kernel: drmn0: [drm] [ENCODER:102:DDI B/PHY B] is disabled/in DSI mode with an ungated DDI clock, gate it
Feb 26 06:37:02 latnz kernel: drmn0: [drm] [ENCODER:109:DDI C/PHY C] is disabled/in DSI mode with an ungated DDI clock, gate it
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_atomic_helper_wait_for_flip_done] [CRTC:51:pipe A] flip_done timed out
Feb 26 06:37:02 latnz kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
Feb 26 06:37:02 latnz kernel: lkpi_iic3: <LinuxKPI I2C> on drm1
Feb 26 06:37:02 latnz kernel: iicbus3: <Philips I2C bus> on lkpi_iic3
Feb 26 06:37:02 latnz kernel: iic3: <I2C generic I/O> on iicbus3
Feb 26 06:37:02 latnz kernel: lkpi_iic4: <LinuxKPI I2C> on drm3
Feb 26 06:37:02 latnz kernel: iicbus4: <Philips I2C bus> on lkpi_iic4
Feb 26 06:37:02 latnz kernel: iic4: <I2C generic I/O> on iicbus4
Feb 26 06:37:02 latnz kernel: [drm] Initialized i915 1.6.0 20201103 for drmn0 on minor 0
Feb 26 06:37:02 latnz kernel: VT: Replacing driver "efifb" with new "fb".
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_crtc_commit_wait] flip_done timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_atomic_helper_wait_for_dependencies] [CRTC:51:pipe A] commit wait timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_crtc_commit_wait] flip_done timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_atomic_helper_wait_for_dependencies] [CONNECTOR:95:eDP-1] commit wait timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_crtc_commit_wait] flip_done timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_atomic_helper_wait_for_dependencies] [PLANE:31:plane 1A] commit wait timed out
Feb 26 06:37:02 latnz kernel: [drm ERROR :drm_atomic_helper_wait_for_flip_done] [CRTC:51:pipe A] flip_done timed out
Feb 26 06:37:02 latnz kernel: start FB_INFO:
Feb 26 06:37:02 latnz kernel: type=11 height=1080 width=1920 depth=32
Feb 26 06:37:02 latnz kernel: pbase=0xd0000000 vbase=0xfffff800d0000000
Feb 26 06:37:02 latnz kernel: name=drmn0 flags=0x0 stride=7680 bpp=32
Feb 26 06:37:02 latnz kernel: end FB_INFO
```

torrented-roms, 2026-09-26:
For those who encounter this issue, try adding `hw.i915kms.nuclear_pageflip="0"` to `/boot/loader.conf`.

Note that you will encounter some visual tearing (for example, of moving windows).

torrented-roms, 2026-09-29:
To those who encounter this or similar problems in the future, try the following tunables.

In `/boot/loader.conf`:
- `hw.i915kms.nuclear_pageflip="0"`
  - This forces legacy non-atomic pageflip instead of the atomic (or nuclear) KMS commit path.
- `hw.pci.enable_msi="0"`
  - This disables legacy MSI system-wide.
- `hw.pci.enable_msix="0"`
  - Disables MSI-X system-wide. In my case, this solved a startup error (`Root mount waiting for: CAM`) that wasn't obviously related to system lockups. But it's mentioned here due to its relationship with `hw.pci.enable_msi="0"`.

Troubleshooting Environment:
OS: FreeBSD 15.1
Hardware: Dell Latitude 3310
CPU: i5-8265u
Display Driver: `drm-latest-kmod` (6.9)