freebsd/drm-kmod #290: [amdgpu] vulkan is not available but llvmpipe

nxjosephofficial, 2024-03-08:
**Describe the bug**
My system uses llvmpipe for rendering instead of vulkan. Tested with `glxgears` and `vkcube`.

**FreeBSD version**
`FreeBSD desktop.home.lan 14.0-STABLE FreeBSD 14.0-STABLE #1 stable/14-n266940-275aee513b58: Fri Mar  8 07:11:22 +03 2024
  root@desktop.home.lan:/usr/obj/usr/src/amd64.amd64/sys/CUSTOM amd64 1400509 1400509`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
amdiommu0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:2:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib8@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb4@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1448 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1449 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
[... 98 lines cut ...]
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h/19h HD Audio Controller'
    class      = multimedia
    subclass   = HDA
```

</details>

**DRM KMOD version**
`drm-61-kmod 6.1.69_1`

**To Reproduce**
```
% sudo kldload linux64
% sudo kldload nullfs
% sudo sysctl security.bsd.unprivileged_chroot=1
% sudo sysctl vfs.usermount=1
% fetch https://gist.githubusercontent.com/shkhln/99ed7076d981a8eee39801bb6634caed/raw/c6f6d38688e02f8131930d475fb5f4f4a51e4893/linux-chroot
% chmod +x linux-chroot
% ./linux-chroot /usr/bin/glxgears -info | grep RENDER
% ./linux-chroot /usr/bin/vkcube
```

**Screenshots**
[image]

evadot, 2024-03-08:
What about native vkcube ? You seems to be using a linux version.

nxjosephofficial, 2024-03-08:
Native vkcube:
```
Selected GPU 0: AMD Radeon Graphics (RADV RENOIR), type: IntegratedGpu
```

nxjosephofficial, 2024-03-09:
native glxgears shows `llvmpipe`
```
% glxgears -info | grep RENDER
failed to authenticate magic 1
failed to load driver: radeonsi
GL_RENDERER   = llvmpipe (LLVM 15.0.7, 256 bits)
```

evadot, 2024-03-09:
failed to authenticate magic 1
Are you a member of the 'video' group ?
failed to load driver: radeonsi
Do you have mesa-dri correctly installed ?

nxjosephofficial, 2024-03-09:
Yes, my users are member of the `video` group. I installed `mesa-dri` and `mesa-libs` from `ports`.
```
$ pkg info -a | grep mesa
mesa-dri-24.0.2                OpenGL hardware acceleration drivers for DRI2+
mesa-libs-24.0.2               OpenGL libraries that support GLX and EGL clients
```
```
$ glxgears -info | grep RENDER
GL_RENDERER   = AMD Radeon Graphics (radeonsi, renoir, LLVM 15.0.7, DRM 3.49, 14.0-STABLE)

$ ./linux-chroot /usr/bin/glxgears -info | grep RENDER
mount -o nocover -t tmpfs tmpfs /home/steam/umnt
mount: tmpfs: Device busy
Assuming /home/steam/umnt is already set up.
amdgpu: os_same_file_description couldn't determine if two DRM fds reference the same file description.
If they do, bad things may happen!
GL_RENDERER   = RENOIR (renoir, LLVM 15.0.7, DRM 3.49, 5.15.0)
```

evadot, 2024-03-09:
I'm lost, please clarify the problem, are you seeing problem with native applications ? https://github.com/freebsd/drm-kmod/issues/290#issuecomment-1986865390 seems to indicate that you have issue but your last comment seems to say that everything is fine.

Is the *only* problem that you are seeing is vulkan using llvmpipe under linux compat ?

nxjosephofficial, 2024-03-09:
I saw minecraft that i run natively was using `llvmpipe` but now after upgrading `mesa-dri, mesa-libs` to latest version, it seems fine. I can't test minecraft again but i am going to test vulkan in steam thats running in linux emulation soon.

My problem was "not being able to use vulkan under linux compatibility". 

`vkcube` under linux compatibility still shows `llvmpipe`.

jbeich, 2024-03-21:
> native glxgears shows llvmpipe

Probably https://github.com/freebsd/drm-kmod/issues/175. Try restarting Xorg next time.

> `vkcube` under linux compatibility still shows `llvmpipe` while native is ok..

[linux-c7-dri](https://www.freshports.org/graphics/linux-c7-dri) (unlike [linux-nvidia-libs](https://www.freshports.org/x11/linux-nvidia-libs)) doesn't support modern GPUs. Workaround: install a whole Linux distro under chroot. See also [Ubuntu LTS testing thread](https://forums.freebsd.org/threads/linuxulator-vs-intel-amd-gpus.92445/).

shkhln, 2024-04-08:
I think I identified the issue (with linsysfs). Try the updated gist at https://gist.githubusercontent.com/shkhln/99ed7076d981a8eee39801bb6634caed/raw/33f0c8be3e8d65cb03641f47947504c65216e33d/linux-chroot.

nxjosephofficial, 2024-04-10:
This time `vkcube` shows my gpu, it seems to work. The cube doesn't turn like crazy. I don't have Dota 2 installed now but i can try soon.
```
% ./linux-chroot /usr/bin/vkcube
mount -o nocover -t tmpfs tmpfs /home/yusuf/umnt
mount: tmpfs: Device busy
Assuming /home/yusuf/umnt is already set up.
Selected GPU 0: AMD Unknown (RADV RENOIR), type: 1
```

nxjosephofficial, 2024-04-10:
https://github.com/shkhln/linuxulator-steam-utils/issues/129#issuecomment-2048526002

evadot, 2024-04-23:
So everything wrt drm-kmod is good ?

closed 2024-04-23 by evadot