freebsd/drm-kmod #293: radeonkms blank screen with FreeBSD 13.3 and RV370 GPU

andrewhotlab, 2024-03-26:
After upgrading FreeBSD from 13.2 to 13.3, the screen goes blank when radeonkms module loads. Running on iMac10,1 with Radeon RV730 GPU.

**FreeBSD version**
FreeBSD imac 13.3-RELEASE FreeBSD 13.3-RELEASE releng/13.3-n257428-80d2b634ddf0 GENERIC amd64 1303001 1303001

**PCI Info**
hostb0@pci0:0:0:0:	class=0x060000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0a84 subvendor=0x0000 subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:1:	class=0x050000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0a88 subvendor=0x0000 subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Memory Controller'
    class      = memory
    subclass   = RAM
isab0@pci0:0:3:0:	class=0x060100 rev=0xb3 hdr=0x00 vendor=0x10de device=0x0aac subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
none1@pci0:0:3:1:	class=0x050000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa4 subvendor=0x0000 subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Memory Controller'
    class      = memory
    subclass   = RAM
none2@pci0:0:3:2:	class=0x0c0500 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa2 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 SMBus'
    class      = serial bus
    subclass   = SMBus
none3@pci0:0:3:3:	class=0x050000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0a89 subvendor=0x0000 subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Memory Controller'
    class      = memory
    subclass   = RAM
none4@pci0:0:3:4:	class=0x050000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0a98 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Memory Controller'
    class      = memory
    subclass   = RAM
none5@pci0:0:3:5:	class=0x0b4000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa3 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Co-processor'
    class      = processor
ohci0@pci0:0:4:0:	class=0x0c0310 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa5 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 OHCI USB 1.1 Controller'
    class      = serial bus
    subclass   = USB
ehci0@pci0:0:4:1:	class=0x0c0320 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa6 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 EHCI USB 2.0 Controller'
    class      = serial bus
    subclass   = USB
ohci1@pci0:0:6:0:	class=0x0c0310 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa7 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 OHCI USB 1.1 Controller'
    class      = serial bus
    subclass   = USB
ehci1@pci0:0:6:1:	class=0x0c0320 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0aa9 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 EHCI USB 2.0 Controller'
    class      = serial bus
    subclass   = USB
hdac0@pci0:0:8:0:	class=0x040300 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0ac0 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 High Definition Audio'
    class      = multimedia
    subclass   = HDA
pcib1@pci0:0:9:0:	class=0x060401 rev=0xb1 hdr=0x01 vendor=0x10de device=0x0aab subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 PCI Bridge'
    class      = bridge
    subclass   = PCI-PCI
nfe0@pci0:0:10:0:	class=0x020000 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0ab0 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 Ethernet'
    class      = network
    subclass   = ethernet
ahci0@pci0:0:11:0:	class=0x010601 rev=0xb1 hdr=0x00 vendor=0x10de device=0x0ab9 subvendor=0x10de subdevice=0xcb79
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 AHCI Controller'
    class      = mass storage
    subclass   = SATA
pcib2@pci0:0:12:0:	class=0x060400 rev=0xb1 hdr=0x01 vendor=0x10de device=0x0ac4 subvendor=0x10de subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 PCI Express Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:21:0:	class=0x060400 rev=0xb1 hdr=0x01 vendor=0x10de device=0x0ac6 subvendor=0x10de subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 PCI Express Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:22:0:	class=0x060400 rev=0xb1 hdr=0x01 vendor=0x10de device=0x0ac7 subvendor=0x10de subdevice=0x0000
    vendor     = 'NVIDIA Corporation'
    device     = 'MCP79 PCI Express Bridge'
    class      = bridge
    subclass   = PCI-PCI
vgapci0@pci0:2:0:0:	class=0x030000 rev=0x00 hdr=0x00 vendor=0x1002 device=0x9488 subvendor=0x106b subdevice=0x00b6
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RV730/M96-XT [Mobility Radeon HD 4670]'
    class      = display
    subclass   = VGA
hdac1@pci0:2:0:1:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0xaa38 subvendor=0x106b subdevice=0xaa38
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RV710/730 HDMI Audio [Radeon HD 4000 series]'
    class      = multimedia
    subclass   = HDA
ath0@pci0:3:0:0:	class=0x028000 rev=0x01 hdr=0x00 vendor=0x168c device=0x002a subvendor=0x106b subdevice=0x008f
    vendor     = 'Qualcomm Atheros'
    device     = 'AR928X Wireless Network Adapter (PCI-Express)'
    class      = network
pcib5@pci0:4:0:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x104c device=0x823e subvendor=0x0000 subdevice=0x0000
    vendor     = 'Texas Instruments'
    device     = 'XIO2213A/B/XIO2221 PCI Express to PCI Bridge [Cheetah Express]'
    class      = bridge
    subclass   = PCI-PCI
none6@pci0:5:0:0:	class=0x0c0010 rev=0x01 hdr=0x00 vendor=0x104c device=0x823f subvendor=0x0000 subdevice=0x0000
    vendor     = 'Texas Instruments'
    device     = 'XIO2213A/B/XIO2221 IEEE-1394b OHCI Controller [Cheetah Express]'
    class      = serial bus
    subclass   = FireWire

**DRM KMOD version**
drm-510-kmod 5.10.163_8
drm-kmod 20220907_1

**To Reproduce**
Load radeonkms.ko on FreeBSD 13.3-RELEASE with a Radeon RV730 GPU.

**Additional context**
I obviously rebuilt all DRM related Ports, `graphics/gpu-firmware-kmod` and `x11-drivers/xf86-video-ati`. Thanks to Boot Environments, I tried the same steps with FreeBSD 14.0 (which brings `graphics/drm-515-kmod`), and the screen freezes when loading radeonkms.ko. The system is reachable via SSH with both releases. No problem with FreeBSD 13.2

I'm not very experienced with this type of issue... there is a way to make the module logging debug info when it loads?

andrewhotlab, 2024-03-26:
I've found some additional info in `/var/log/messages`... here are the relevant lines from FreeBSD 13.3:
<details><summary>messages_13.3.txt</summary>
<pre>
<code>Mar 26 10:08:43 imac kernel: [drm] radeon kernel modesetting enabled.
Mar 26 10:08:43 imac kernel: drmn0: <drmn> on vgapci0
Mar 26 10:08:43 imac kernel: vgapci0: child drmn0 requested pci_enable_io
Mar 26 10:08:43 imac syslogd: last message repeated 1 times
Mar 26 10:08:43 imac kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
Mar 26 10:08:43 imac kernel: [drm] initializing kernel modesetting (RV730 0x1002:0x9488 0x106B:0x00B6 0x00).
Mar 26 10:08:43 imac kernel: [drm ERROR :radeon_atombios_init] Unable to find PCI I/O BAR; using MMIO for ATOM IIO
Mar 26 10:08:43 imac kernel: ATOM BIOS: 113
Mar 26 10:08:43 imac kernel: drmn0: VRAM: 256M 0x0000000000000000 - 0x000000000FFFFFFF (256M used)
Mar 26 10:08:43 imac kernel: drmn0: GTT: 1024M 0x0000000010000000 - 0x000000004FFFFFFF
Mar 26 10:08:43 imac kernel: [drm] Detected VRAM RAM=256M, BAR=256M
Mar 26 10:08:43 imac kernel: [drm] RAM width 128bits DDR
Mar 26 10:08:43 imac kernel: [TTM] Zone  kernel: Available graphics memory: 8364582 KiB
Mar 26 10:08:43 imac kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 KiB
Mar 26 10:08:43 imac kernel: [TTM] Initializing pool allocator
Mar 26 10:08:43 imac kernel: [drm] radeon: 256M of VRAM memory ready
Mar 26 10:08:43 imac kernel: [drm] radeon: 1024M of GTT memory ready.
Mar 26 10:08:43 imac kernel: [drm] Loading RV730 Microcode
Mar 26 10:08:43 imac kernel: drmn0: successfully loaded firmware image 'radeon/RV730_pfp.bin'
Mar 26 10:08:43 imac kernel: drmn0: successfully loaded firmware image 'radeon/RV730_me.bin'
Mar 26 10:08:43 imac kernel: drmn0: successfully loaded firmware image 'radeon/R700_rlc.bin'
Mar 26 10:08:43 imac kernel: drmn0: successfully loaded firmware image 'radeon/RV730_smc.bin'
Mar 26 10:08:43 imac kernel: [drm] External GPIO thermal controller with fan control
Mar 26 10:08:43 imac kernel: [drm] radeon: dpm initialized
Mar 26 10:08:43 imac kernel: drmn0: successfully loaded firmware image 'radeon/RV710_uvd.bin'
Mar 26 10:08:43 imac kernel: [drm] GART: num cpu pages 262144, num gpu pages 262144
Mar 26 10:08:43 imac kernel: [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
Mar 26 10:08:43 imac kernel: [drm] PCIE GART of 1024M enabled (table at 0x000000000014C000).
Mar 26 10:08:43 imac kernel: drmn0: WB enabled
Mar 26 10:08:43 imac kernel: drmn0: fence driver on ring 0 use gpu addr 0x0000000010000c00
Mar 26 10:08:43 imac kernel: drmn0: fence driver on ring 3 use gpu addr 0x0000000010000c0c
Mar 26 10:08:43 imac kernel: drmn0: fence driver on ring 5 use gpu addr 0x000000000005c598
Mar 26 10:08:43 imac kernel: drmn0: radeon: MSI limited to 32-bit
Mar 26 10:08:43 imac kernel: drmn0: radeon: using MSI.
Mar 26 10:08:43 imac kernel: [drm] radeon: irq initialized.
Mar 26 10:08:43 imac kernel: [drm] ring test on 0 succeeded in 1 usecs
Mar 26 10:08:43 imac kernel: [drm] ring test on 3 succeeded in 2 usecs
Mar 26 10:08:44 imac kernel: [drm] ring test on 5 succeeded in 1 usecs
Mar 26 10:08:44 imac kernel: [drm] UVD initialized successfully.
Mar 26 10:08:44 imac kernel: [drm] ib test on ring 0 succeeded in 0 usecs
Mar 26 10:08:44 imac kernel: [drm] ib test on ring 3 succeeded in 0 usecs
Mar 26 10:08:44 imac kernel: [drm] ib test on ring 5 succeeded
Mar 26 10:08:44 imac kernel: lkpi_iicbb0: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb0: <I2C bit-banging driver> on lkpi_iicbb0
Mar 26 10:08:44 imac kernel: iicbus0: <Philips I2C bus> on iicbb0 addr 0xff
Mar 26 10:08:44 imac kernel: iic0: <I2C generic I/O> on iicbus0
Mar 26 10:08:44 imac kernel: lkpi_iicbb1: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb1: <I2C bit-banging driver> on lkpi_iicbb1
Mar 26 10:08:44 imac kernel: iicbus1: <Philips I2C bus> on iicbb1 addr 0xff
Mar 26 10:08:44 imac kernel: iic1: <I2C generic I/O> on iicbus1
Mar 26 10:08:44 imac kernel: lkpi_iicbb2: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb2: <I2C bit-banging driver> on lkpi_iicbb2
Mar 26 10:08:44 imac kernel: iicbus2: <Philips I2C bus> on iicbb2 addr 0xff
Mar 26 10:08:44 imac kernel: iic2: <I2C generic I/O> on iicbus2
Mar 26 10:08:44 imac kernel: lkpi_iicbb3: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb3: <I2C bit-banging driver> on lkpi_iicbb3
Mar 26 10:08:44 imac kernel: iicbus3: <Philips I2C bus> on iicbb3 addr 0xff
Mar 26 10:08:44 imac kernel: iic3: <I2C generic I/O> on iicbus3
Mar 26 10:08:44 imac kernel: lkpi_iicbb4: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb4: <I2C bit-banging driver> on lkpi_iicbb4
Mar 26 10:08:44 imac kernel: iicbus4: <Philips I2C bus> on iicbb4 addr 0xff
Mar 26 10:08:44 imac kernel: iic4: <I2C generic I/O> on iicbus4
Mar 26 10:08:44 imac kernel: lkpi_iicbb5: <LinuxKPI I2CBB> on drmn0
Mar 26 10:08:44 imac kernel: iicbb5: <I2C bit-banging driver> on lkpi_iicbb5
Mar 26 10:08:44 imac kernel: iicbus5: <Philips I2C bus> on iicbb5 addr 0xff
Mar 26 10:08:44 imac kernel: iic5: <I2C generic I/O> on iicbus5
Mar 26 10:08:44 imac kernel: lkpi_iicbb6: <LinuxKPI I2CBB> on drmn0
[... 132 lines cut ...]
Mar 26 15:26:55 imac kernel: [drm]   eDP-1
Mar 26 15:26:55 imac kernel: [drm]   HPD2
Mar 26 15:26:55 imac kernel: [drm]   DDC: 0x7e50 0x7e50 0x7e54 0x7e54 0x7e58 0x7e58 0x7e5c 0x7e5c
Mar 26 15:26:55 imac kernel: [drm]   Encoders:
Mar 26 15:26:55 imac kernel: [drm]     LCD1: INTERNAL_UNIPHY
[... cut ...]
Mar 26 15:26:55 imac kernel: [drm]    pitch is 10240
Mar 26 15:26:55 imac kernel: VT: Replacing driver "efifb" with new "fb".
Mar 26 15:26:55 imac kernel: start FB_INFO:
Mar 26 15:26:55 imac kernel: type=11 height=1440 width=2560 depth=32
Mar 26 15:26:55 imac kernel: pbase=0xc034d000 vbase=0xfffff800c034d000
Mar 26 15:26:55 imac kernel: name=drmn0 flags=0x0 stride=10240 bpp=32
Mar 26 15:26:55 imac kernel: end FB_INFO
Mar 26 15:26:55 imac kernel: [drm] Initialized radeon 2.50.0 20080528 for drmn0 on minor 0
</code>
</pre>
</details>

andrewhotlab, 2024-04-17:
Still not being able to solve the issue in FreeBSD 13.3, but in the mean time I realized that screen freezing at `radeonkms.ko` load in FreeBSD 14.0 was due to my mistake: I missed to rebuild all the DRM-related Ports after upgrade to that latest OS major release.

Any suggestion to help me troubleshooting the issue on FreeBSD 13.3 will be well accepted! :)

thesunexpress, 2024-12-26:
> After upgrading FreeBSD from 13.2 to 13.3, the screen goes blank when radeonkms module loads. Running on iMac10,1 with Radeon RV730 GPU.
> 
> **FreeBSD version**
> 
> ```
> FreeBSD imac 13.3-RELEASE FreeBSD 13.3-RELEASE releng/13.3-n257428-80d2b634ddf0 GENERIC amd64 1303001 1303001
> ```
> 
> **PCI Info**
> pciconf -lv
> 
> **DRM KMOD version**
> 
> ```
> drm-510-kmod 5.10.163_8
> drm-kmod 20220907_1
> ```
> 
> **To Reproduce** Load radeonkms.ko on FreeBSD 13.3-RELEASE with a Radeon RV730 GPU.
> 
> **Additional context** I obviously rebuilt all DRM related Ports, `graphics/gpu-firmware-kmod` and `x11-drivers/xf86-video-ati`. Thanks to Boot Environments, I tried the same steps with FreeBSD 14.0 (which brings `graphics/drm-515-kmod`), and the screen freezes when loading radeonkms.ko. The system is reachable via SSH with both releases. No problem with FreeBSD 13.2
> 
> I'm not very experienced with this type of issue... there is a way to make the module logging debug info when it loads?

If I am not mistaken, doesn't radeonkms.ko (and some loader.conf shenanigans) require booting in Legacy BIOS mode rather than UEFI? That is a 16 year old iMac/GPU combo, isn't it?
Bit late to the party here, I realize...

andrewhotlab, 2024-12-26:
In the mean time I started using the 14 branch as my daily driver, so this issue became less relevant to me and, I guess, to the most FreeBSD users.
At this very time I upgraded from 14.1 to 14.2, and all vty consoles become unaccessible as soon as `radeonkms.ko` loads… :’(