freebsd/drm-kmod #296: HDMI sound lost after upgrade to drm-61-kmod

mzary, 2024-04-03:
**Describe the bug**
Sound worked with drm-51{0,5}-kmod on the laptop, but it's broken now.

**FreeBSD version**
FreeBSD bsdondell.lab.pwste.edu.pl 14.0-STABLE FreeBSD 14.0-STABLE #14 stable/14-n267036-e87a6f886046-dirty: Sun Mar 24 10:36:12 CET 2024     root@hamal.dom.potoki.eu:/usr/obj/usr/src/amd64.amd64/sys/MINTAKA amd64 1400510 1400510
**PCI Info**

<details>
  <summary>
% pciconf -lv
hostb0@pci0:0:0:0:	class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
ig4iic1@pci0:0:21:1:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d61 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:	class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
none2@pci0:0:22:3:	class=0x070002 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3d subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Active Management Technology - SOL'
    class      = simple comms
    subclass   = UART
ahci0@pci0:0:23:0:	class=0x010400 rev=0x21 hdr=0x00 vendor=0x8086 device=0x282a subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = '82801 Mobile SATA Controller [RAID mode]'
    class      = mass storage
    subclass   = RAID
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:2:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d12 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d14 subvendor=0x1028 subdevice=0x0817
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
[... 118 lines cut ...]
hdac0: Command 0x20270d00 timeout on address 2
hdac0: Command 0x20270600 timeout on address 2

**Additional context**
Add any other context about the problem here.

drmn0: successfully loaded firmware image 'i915/kbl_dmc_ver1_04.bin'
drmn0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
drmn0: [drm] [ENCODER:102:DDI B/PHY B] is disabled/in DSI mode with an ungated DDI clock, gate it
drmn0: [drm] [ENCODER:113:DDI C/PHY C] is disabled/in DSI mode with an ungated DDI clock, gate it
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
lkpi_iic3: <LinuxKPI I2C> on drm1
iicbus3: <Philips I2C bus> on lkpi_iic3
iic3: <I2C generic I/O> on iicbus3
lkpi_iic4: <LinuxKPI I2C> on drm2
iicbus4: <Philips I2C bus> on lkpi_iic4
iic4: <I2C generic I/O> on iicbus4
lkpi_iic5: <LinuxKPI I2C> on drm3
iicbus5: <Philips I2C bus> on lkpi_iic5
iic5: <I2C generic I/O> on iicbus5
[drm] Initialized i915 1.6.0 20201103 for drmn0 on minor 0

I hope it helps to make it work before 14.1 RELEASE.

Marek

evadot, 2024-04-03:
Thanks for the report.
It will probably be hard to fix for me as it never worked on my kabylake laptop ...

rosti-il, 2024-05-22:
Actually both versions of this driver have bugs in its sound capabilities with influence even to its video capabilities:
https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=278930

grahamperrin, 2024-12-01:
```text
root@fourteen-pkgbase:~ # freebsd-version -kru ; uname -bmvKU
14.2-RELEASE
14.2-RELEASE
14.2-RELEASE
FreeBSD 14.2-RELEASE releng/14.2-n269506-c8918d6c7412 GENERIC amd64 1402000 1402000 4f83741c36da87fe6226d4043930a4ea79801b8e
root@fourteen-pkgbase:~ # kldload i915kms
root@fourteen-pkgbase:~ # sysctl -d compat.linuxkpi.i915_disable_power_well
compat.linuxkpi.i915_disable_power_well: Disable display power wells when possible (-1=auto [default], 0=power wells always on, 1=power wells disabled when possible)
root@fourteen-pkgbase:~ # 
```

As a possible workaround, please try: `0`

----

Background: 

- [MFC r359777: hdac: show which command timed out · freebsd/freebsd-src@107ecd7](https://github.com/freebsd/freebsd-src/commit/107ecd7b5999c27d90f46609d92a25c541e18d32) (2020-04-25)
- [MFC r359781: hda: if a CODEC is not responding, report which one · freebsd/freebsd-src@b1a7781](https://github.com/freebsd/freebsd-src/commit/b1a778168a9e781b4cbf1127e3ce52e7e2384325) (2020-04-25)
- [229190 – hdac0: Command timeout on address 2 (Intel Kabylake HDA CODEC)](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=229190) (closed (fixed) 2021-05-21)

freebsdcross, 2025-01-04:
I have this problem also, and I can confirm the workaround works for me.  For others trying this, you need to set this at boot time in /boot/loader.conf

mzary, 2025-03-28:
This issue now affects regular 14.2-RELEASE. The report below refers to different hardware, it's a PC with Intel Kaby Lake-H HDA where sound over HDMI worked perfectly under 14.1-RELEASE and earlier versions. 

Hardware details: 
Mar 28 08:55:37 pc-s47-10-8-47-17 kernel: hdac0: <Intel Kaby Lake-H HDA Controller> mem 0xdf140000-0xdf143fff,0xdf120000-0xdf12ffff irq 16 at device 31.3 on pci0
Mar 28 08:55:37 pc-s47-10-8-47-17 kernel: hdacc1: <Intel Kaby Lake HDA CODEC> at cad 2 on hdac0
Mar 28 08:55:37 pc-s47-10-8-47-17 kernel: hdaa1: <Intel Kaby Lake Audio Function Group> at nid 1 on hdacc1
Mar 28 08:55:37 pc-s47-10-8-47-17 kernel: pcm2: <Intel Kaby Lake (HDMI/DP 8ch)> at nid 3 on hdaa1
Mar 28 08:55:37 pc-s47-10-8-47-17 kernel: ichsmb0: <Intel Kaby Lake SMBus controller> port 0xf040-0xf05f mem 0xdf14a000-0xdf14a0ff irq 16 at device 31.4 on pci0

vgapci1@pci0:0:2:0:	class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x5912 subvendor=0x1462 subdevice=0x7a72
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics 630'
    class      = display
    subclass   = VGA
    bar   [10] = type Memory, range 64, base 0xde000000, size 16777216, enabled
    bar   [18] = type Prefetchable Memory, range 64, base 0xc0000000, size 268435456, enabled
    bar   [20] = type I/O Port, range 32, base 0xf000, size 64, enabled

The PC has installed 14.2-RELEASE-p1 FreeBSD 14.2-RELEASE-p1 GENERIC amd64 with packages:
 drm-61-kmod-6.1.128.1401000_1,  drm-kmod-20220907_3 and   gpu-firmware-kmod-20241114,1. Sound over HDMI is broken; moreover, this driver build on 14.1 doesn't allow switching to the text console, so I am eagerly waiting for 14.1 builders to be killed and eventually replaced by 14.2 to fix at least this console issue. 

We had university open day today, and this issue is really painful. We have here HDMI projectors with speakers, and everything worked fine when I had FreeBSD 14.1-RELEASE with drm-510 or drm-515 kmods last time. 

In the kernel message buffer, there are tens of messages like the one below: 
hdacc1: Unexpected unsolicited response with tag 63: ffffffff

In the year of FreeBSD on the desktop, this issue has to be solved. How can I help with debugging this ?

grahamperrin, 2025-03-28:
> drm-61-kmod-6.1.128.1401000_1

Builds for 14.2 began last year. 

[FreeBSD Project-provided repositories for kernel modules in the ports collection: usage](https://blendit.bsd.cafe/post/821622)

mzary, 2025-03-29:
I can confirm that compat.linuxkpi.i915_disable_power_well set to 0 solves the issue in case of both Kaby Lake systems described above. Plase, consider making it default setting.

> > drm-61-kmod-6.1.128.1401000_1
> 
> Builds for 14.2 began last year.
> 
> [FreeBSD Project-provided repositories for kernel modules in the ports collection: usage](https://blendit.bsd.cafe/post/821622)

Thank you for this tip. It fixed the problem of blank text consoles after the upgrade to 14.2-RELEASE.

ant-222, 2026-01-01:
I confirm that the bug is present on
`FreeBSD 14.3-RELEASE-p5`
and the workaround still works and is required.

Unixware, 2026-09-12:
.,..as above on 15.1R-p3  / i3 7th

zarMarco, 2026-10-08:
Same problem with 
`  󰆧  freebsd-version -kru && uname -a
15.1-RELEASE-p4
15.1-RELEASE-p4
15.1-RELEASE-p4
FreeBSD beastiegram 15.1-RELEASE-p4 FreeBSD 15.1-RELEASE-p4 releng/15.1-n283627-db6066678b33 GENERIC amd64
`

and this drm
`  󰆧  pkg info -g '*drm*'
drm-latest-kmod-6.12.85.1501000
faac-backup-libfaac_drm.so.0-20260810215821
libdrm-2.4.133,1`