freebsd/drm-kmod #298: drm 6.1: FreeSync/VRR doesnt work correctly.

Alexander88207, 2024-04-07:
**Describe the bug**
Since VRR/Freesync support came with drm 5.4 it has always worked beautifully.

Now it only works partially.

I was able to observe the following

- FreeSync kicks in and changes but stays stuck at a fixed refresh rate or want return to normal refresh rate after leaving the game.

- Sometimes it doesn't kick in and all and you have to play around with Vsync leaver or turn on/off from VRR from the Display to wake it up.

**FreeBSD version**
FreeBSD Home 14.0-STABLE FreeBSD 14.0-STABLE #0 stable/14-n267114-72c3d91294c4: Thu Apr  4 05:52:43 UTC 2024     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1400510 1400510

**PCI Info**
vgapci0@pci0:3:0:0:	class=0x030000 rev=0xc5 hdr=0x00 vendor=0x1002 device=0x73df subvendor=0x1043 subdevice=0x05cb
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT]'
    class      = display
    subclass   = VGA

</details>

**DRM KMOD version**
 
6.1.69_2

**To Reproduce**

Install minetest_game and run it in fullscreen mode and set the FPS in your VRR range.

The refresh rate should change constantly if it works correctly if not then its stuck or doesnt change at all.

**Screenshots**
WIP

Alexander88207, 2024-04-08:
If someone else wants to try out it aswell here is an small article about FreeSync/VRR on FreeBSD: https://wiki.freebsd.org/AlexanderVereeken/FreeSync

Alexander88207, 2024-05-05:
Also happens with https://github.com/freebsd/drm-kmod/pull/283

nevillehay, 2025-04-15:
Have you tried setting 
hw.amdgpu.freesync_video="1"
in /boot/loader.conf