freebsd/drm-kmod #301: gma500 support

hhartzer, 2024-05-03:
Hi,

It'd be nice to be able to use Linux' gma500 driver on FreeBSD.

From what I understand, it's PowerVR-based. Includes GMA3600/GMA3650. Useful largely for older netbooks.

I have no idea what this would take to pull off, if it's relatively trivial or extremely difficult.

https://github.com/torvalds/linux/tree/3d25a941ea5013b552b96330c83052ccace73a48/drivers/gpu/drm/gma500

Could someone take a look and give me an idea? I don't mind helping do this if it seems like I might have some chance of success.

Thank you!

evadot, 2024-06-04:
Unlikely to happen because :
1/ As you said it's for old hardware that no one really cares about
2/ It's GPL.

Closing.

closed 2024-06-04 by evadot

hhartzer, 2024-06-04:
2. Isn't i915, and most of this, GPL? Curious if I'm wrong about that.

Thanks!