freebsd/drm-kmod #303: DRM 6.1: Transient screen corruption and rare freezes (Intel UHD 620)

OlCe2, 2024-05-22:
**Description and reproduction**

With DRM 6.1, in a freshly started KDE graphical session (Plasma X11), transient screen corruption, i.e., random pixels with all possible colors for some millisecond or less (may not be completely random, but rather some evolution of the regular displayed image; corruption doesn't last enough to be able to tell), can be observed when performing some actions. The easiest is to hover the pointer over applications in the task bar, which triggers display of thumbnails with some transparency. Other desktop effects seem to trigger it, as well as using Firefox. Usually, corruption is transient, but rarely the display freezes forever, with the original image decaying (with colors becoming darker and darker).

Though perhaps an unrelated problem (but playing a role here, as will be seen below), eventually `plasmashell` freezes, and sometimes `kwin_x11` as well (this happens with every DRM version, 5.10, 5.15 and 6.1). After such a freeze, going to the text console and restarting `kwin_x11` and `plasmashell` (with `--replace`) rejuvenates the desktop once and for all (no freeze then ever happens, except after suspend/resume, which seems to somehow reset the state).

Going back to the current bug, after a restart of KWin and Plasmashell as just explained, the screen corruption seems to be gone, and never happens again.

**System Information**

*FreeBSD version*

FreeBSD 14.1-STABLE #0 n267671-9a8a26aefb36: Mon May 13 13:39:56 CEST 2024     MYCONFIG 1401500 1401500
Kernel MYCONFIG is a stripped-down version of GENERIC close to MINIMAL.

*PCI Info*

vgapci0@pci0:0:2:0:     class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x9b41 subvendor=0x1028 subdevice=0x0962
    vendor     = 'Intel Corporation'
    device     = 'CometLake-U GT2 [UHD Graphics]'
    class      = display
    subclass   = VGA
    bar   [10] = type Memory, range 64, base 0x604a000000, size 16777216, enabled
    bar   [18] = type Prefetchable Memory, range 64, base 0x4000000000, size 268435456, enabled
    bar   [20] = type I/O Port, range 32, base 0x3000, size 64, enabled
    cap 09[40] = vendor (length 12) Intel cap 0 version 1
    cap 10[70] = PCI-Express 2 root endpoint max data 128(128) FLR
                 max read 128
    cap 05[ac] = MSI supports 1 message enabled with 1 message
    cap 01[d0] = powerspec 2  supports D0 D3  current D0
    ecap 001b[100] = Process Address Space ID 1
    ecap 000f[200] = ATS 1
    ecap 0013[300] = Page Page Request 1

*DRM KMOD version*

Problem reproduced with:
drm-61-kmod 6.1.69_2

No problem with:
drm-510-kmod 5.10.163_9
drm-515-kmod 5.15.118_4

The KWin/Plasmashell freeze secondary problem evoked above occurs with any of the DRM KMODs listed above.

Feel free to ask for more information.

(Part of the time spent on this report was sponsored by the FreeBSD Foundation.)

OlCe2, 2024-05-23:
After a search over the DRM bugs database, this issue might be related to:
https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/5969
even if the kind of visual corruption I'm getting is quite different than that visible in the video attached to that bug.

I'll try some hand-patching in the lines of what is suggested there to see if this makes any difference here.

evadot, 2024-05-28:
I will love to help here but I don't have this problem at all on all my machines.
My main machines are skylake and whiskeylake.
Also could be only triggered by something wrt kde as I run sway on all of them.

OlCe2, 2024-09-02:
All the problems I mentioned above were fixed by switching to Xorg's `modesetting` driver (instead of `intel`).

So closing this.

closed 2024-09-02 by OlCe2