freebsd/drm-kmod #304: Power off not working after upgrade drm-61-kmod-6.1.69_2 -> drm-61-kmod-6.1.92

jalvermark, 2024-06-05:
**Describe the bug**
My laptop does not power off when issuing "poweroff" or "shutdown -p now" when amdgpu.ko from drm-61-kmod-6.1.92 is loaded.
It worked before with drm-61-kmod-6.1.69_2

**FreeBSD version**
15.0-CURRENT FreeBSD 15.0-CURRENT #32 main-n270598-41dfea24eec2-dirty: Wed Jun  5 09:52:45 CEST 2024     root@bob:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64 1500019 1500019

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14e8 subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14e9 subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:2:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14ee subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:2:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14ee subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:3:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14ef subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 19h USB4/Thunderbolt PCIe tunnel'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:3:2:	class=0x000000 rev=0x00 hdr=0x00 vendor=0x0000 device=0x0000 subvendor=0x0000 subdevice=0x0000
    class      = old
    subclass   = non-VGA display device
none2@pci0:0:3:3:	class=0x000000 rev=0x00 hdr=0x00 vendor=0x0000 device=0x0000 subvendor=0x0000 subdevice=0x0000
    class      = old
    subclass   = non-VGA display device
none3@pci0:0:3:4:	class=0x000000 rev=0x00 hdr=0x00 vendor=0x0000 device=0x0000 subvendor=0x0000 subdevice=0x0000
    class      = old
    subclass   = non-VGA display device
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib4@pci0:0:4:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14ef subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 19h USB4/Thunderbolt PCIe tunnel'
    class      = bridge
    subclass   = PCI-PCI
hostb5@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14eb subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
[... 109 lines cut ...]
none11@pci0:197:0:5:	class=0x0c0340 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1668 subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Pink Sardine USB4/Thunderbolt NHI controller'
    class      = serial bus
    subclass   = USB
none12@pci0:197:0:6:	class=0x0c0340 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1669 subvendor=0x17aa subdevice=0x50d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Pink Sardine USB4/Thunderbolt NHI controller'
    class      = serial bus
    subclass   = USB

</details>

**DRM KMOD version**
drm-61-kmod 6.1.92

**To Reproduce**
Steps to reproduce the behavior:
Run "poweroff" or "shutdown -p now" while amdgpu.ko from drm-61-kmod 6.1.92 is loaded.

**Screenshots**
N/A

**Additional context**
Add any other context about the problem here.

evadot, 2024-06-05:
No problems here on my side, can you git bisect the 6.1-lts branch ?
Thanks.

jalvermark, 2024-06-06:
Bisecting points to this commit:
commit 2aaea5d18c21608d2959dd5e5bffa2370068a544 (HEAD)
Author: Mario Limonciello <mario.limonciello@amd.com>
Date:   Thu May 2 13:32:17 2024 -0500

    dm/amd/pm: Fix problems with reboot/shutdown for some SMU 13.0.4/13.0.11 use
rs
    
    [ Upstream commit cd94d1b182d2986378550c9087571991bfee01d4 ]
    
    Limit the workaround introduced by commit 31729e8c21ec ("drm/amd/pm: fixes
    a random hang in S4 for SMU v13.0.4/11") to only run in the s4 path.
    
    Cc: Tim Huang <Tim.Huang@amd.com>
    Fixes: 31729e8c21ec ("drm/amd/pm: fixes a random hang in S4 for SMU v13.0.4/11")
    Closes: https://gitlab.freedesktop.org/drm/amd/-/issues/3351
    Signed-off-by: Mario Limonciello <mario.limonciello@amd.com>
    Acked-by: Alex Deucher <alexander.deucher@amd.com>
    Signed-off-by: Alex Deucher <alexander.deucher@amd.com>
    Signed-off-by: Sasha Levin <sashal@kernel.org>

diff --git a/drivers/gpu/drm/amd/pm/swsmu/smu13/smu_v13_0_4_ppt.c b/drivers/gpu/drm/amd/pm/swsmu/smu13/smu_v13_0_4_ppt.c
index 21b374d121..5de3196131 100644
--- a/drivers/gpu/drm/amd/pm/swsmu/smu13/smu_v13_0_4_ppt.c
+++ b/drivers/gpu/drm/amd/pm/swsmu/smu13/smu_v13_0_4_ppt.c
@@ -222,7 +222,7 @@ static int smu_v13_0_4_system_features_control(struct smu_context *smu, bool en)
        struct amdgpu_device *adev = smu->adev;
        int ret = 0;
 
-       if (!en && !adev->in_s0ix) {
+       if (!en && adev->in_s4) {
                /* Adds a GFX reset as workaround just before sending the
                 * MP1_UNLOAD message to prevent GC/RLC/PMFW from entering
                 * an invalid state.

bsdcode, 2024-07-06:
I can confirm:

My minipc EliteMini UM780 XTX from Minisforum uses AMD Radeom 780M graphics (probably the same as @jalvermark because of `device = Phoenix1` from `pciconf -lv`). I'm using FreeBSD 14.1-RELEASE-p2 with packages from the latest repository and I compiled the drm-61-kmod port from the main branch of the ports tree. The system hangs after `poweroff` and has to be shutdown via the power button on the minipc.

Reverting [2aaea5d](https://github.com/freebsd/drm-kmod/commit/2aaea5d18c21608d2959dd5e5bffa2370068a544) fixes this issue.

wulf7, 2024-07-11:
Probably fixed in Linux 7da44257e64d

jalvermark, 2024-07-12:
> Probably fixed in Linux 7da44257e64d

I couldn't find that.

I did find https://github.com/torvalds/linux/commit/267cace556e8a53d703119f7435ab556209e5b6a

I tested it and it works.

bsdcode, 2024-07-18:
> I did find [torvalds/linux@267cace](https://github.com/torvalds/linux/commit/267cace556e8a53d703119f7435ab556209e5b6a)
> 
> I tested it and it works.

I also tested this and I can confirm that it works.