freebsd/drm-kmod #305: Black screen and system freezing on AMD Ryzen 9 7950X

daniloegea, 2024-06-08:
**Describe the bug**

I'm running FreeBSD 15 on a Ryzen 9 7950X (codename Raphael) and trying to get the amdgpu driver working. I've found at least two problems with the drm-61-kmod (6.1.92) driver.

One is that if the required firmwares are not loaded when you load the amdgpu driver the system will panic. It took me a while to figure out what firmwares I needed. There are some clues in the panic itself but what helped me was this documentation here: https://docs.kernel.org/gpu/amdgpu/driver-misc.html

There is a list of what versions each processor family needs.

After loading the required firmwares the system stopped panic'ing but now all I get is a black screen and the system hangs. I attached below the logs I get when I load the module.

IF YOU GOT HERE WITH THE SAME PROBLEM: the `scfb` driver + Xorg works really well. To get a high resolution use the `gop set` command in the bootloader and set the resolution you want.

**FreeBSD version**
`FreeBSD capeta 15.0-CURRENT FreeBSD 15.0-CURRENT #0 main-n270672-2b887687edc2: Fri Jun  7 13:40:19 IST 2024     danilo@capeta:/usr/obj/usr/src/amd64.amd64/sys/CAPETA amd64 1500019 1500019`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14d8 subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14d9 subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14db subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib2@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14db subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib10@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14dd subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
pcib11@pci0:0:8:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14dd subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:20:0:	class=0x0c0500 rev=0x71 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
[... 253 lines cut ...]
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring sdma0 uses VM inv eng 12 on hub 0
Jun  7 13:23:05 capeta kernel: drmn0: ring vcn_dec_0 uses VM inv eng 0 on hub 1
Jun  7 13:23:05 capeta kernel: drmn0: ring vcn_enc_0.0 uses VM inv eng 1 on hub 1
Jun  7 13:23:05 capeta kernel: drmn0: ring vcn_enc_0.1 uses VM inv eng 4 on hub 1
Jun  7 13:23:05 capeta kernel: drmn0: ring jpeg_dec uses VM inv eng 5 on hub 1
Jun  7 13:23:05 capeta kernel: vgapci0: child drmn0 requested pci_get_powerstate
Jun  7 13:23:05 capeta kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
Jun  7 13:23:05 capeta kernel: lkpi_iic2: <LinuxKPI I2C> on drm2
Jun  7 13:23:05 capeta kernel: iicbus3: <Philips I2C bus> on lkpi_iic2
Jun  7 13:23:05 capeta kernel: iic3: <I2C generic I/O> on iicbus3
Jun  7 13:23:05 capeta kernel: [drm] Initialized amdgpu 3.49.0 20150101 for drmn0 on minor 0
Jun  7 13:23:05 capeta kernel: VT: Replacing driver "efifb" with new "drmfb".
Jun  7 13:23:05 capeta kernel: [drm] DSC precompute is not needed.
Jun  7 13:23:05 capeta kernel: [drm] REG_WAIT timeout 1us * 100000 tries - optc31_disable_crtc line:138
Jun  7 13:25:50 capeta syslogd: kernel boot file is /boot/kernel/kernel
Jun  7 13:25:50 capeta kernel: ---<<BOOT>>---

```

evadot, 2024-06-10:
For the firmware part we need to update fwget(8) to match the pci id to firmware names, can you cook up a patch similar to https://cgit.freebsd.org/src/commit/usr.sbin/fwget/pci/pci_video_amd?id=2db2566f55b06d80e28c932ec30b5730815d95a2 ?
For the black screen I have no idea but I'm not sure if 6.1 have everything ready for Raphael GPUs.

chrislongros, 2024-06-22:
Have the exact same issue using 14.1 release. amdgpu without the firmware package produces kernel panic. After installing it and loading of amdgpu module at booting, I get a black screen. Ctrl+Alt+F2 does not function and I must reboot into single user mode.

daniloegea, 2024-06-30:
I created a PR adding the firmware for this GPU to fwget https://github.com/freebsd/freebsd-src/pull/1308

jwp, 2025-03-02:
Seeing similar logs with a Ryzen 7 9700X on 14.2-RELEASE-p1 (drm-61-kmod installed from pkg+https://pkg.freebsd.org/FreeBSD:14:amd64/kmods_latest_2).

Loads the same firmware image, but ends with two REG_WAIT timeouts and a blank screen. optc1_wait_for_state line 819 in addition to optc31_disable_crtc line 138. However, no panic or freeze; can just execute reboot from the still active shell.

Using scfb as well for Xorg.

levelad, 2025-05-17:
I've got 2 questions:

1. I could not find that exact information but I guess AMD Raphael is only supported since Linux kernel 6.2 and since FreeBSD only seems to use LTS kernel drivers, this would mean AMD Raphael will be earliest supported in drm-66-kmod (based on Linux kernel 6.6 drivers).

2. Which FreeBSD version would be the first one where the drm-kmod metaport is at Linux kernel version 6.6 and supports AMD Raphael out of the box, would that already be FreeBSD 14.3?

Here is my device id from an EPYC 4344P (basically a Ryzen 7 7000) AMD Raphael IGPU:

`vgapci0@pci0:15:0:0:    class=0x030000 rev=0xc5 hdr=0x00 vendor=0x1002 device=0x164e subvendor=0x15d9 subdevice=0x1c97`

https://www.techpowerup.com/gpu-specs/amd-raphael.g1021

daniloegea, 2026-01-30:
Just a heads up here,

I just tested the recently merged driver from Linux 6.10 with an updated FreeBSD 16 and it worked!

I'm running it for just one hour and only tested the Display port connection but it seems to be working.

Graphics acceleration seems to be working. I tested it with Chromium and WebGL rendering stuff.

Firefox though, when hardware acceleration is enabled, will be almost unusable with lots of artifacts. No idea what the problem could be.

But anyway, you may want to test the new driver :)

daniloegea, 2026-01-31:
As my problem seems to be solved, I'll be closing this issue. Thanks for all this work!