freebsd/drm-kmod #307: Raptor Lake

JasonSmith, 2024-06-11:
This is just a general question on the support status of Raptor Lake.

I installed FreeBSD 14.1 on my Framework 13 laptop with a 13th gen intel processor. I then built from the ports tree the drm-61-kmod package. After adding my user to the video group I added kld_list="i915kms" to my rc.conf file.

On a reboot I am greeted with a black screen with a insert cursor white rectangle in the top left. The only interaction that seemed to work was ctrl+alt+delete. I rebooted into single user mode to revert my changes.

I then checked the ports tree graphics/gpu-firmware-intel-kmod and the latest listing seems to be for Sky Lake.

So is Raptor Lake currently still unsupported or am I just not understanding how to install it? Is it only available in FreeBSD 15 current?

Thanks in advance for any help you can provide.

rosorio, 2024-06-11:
AFAIK, only 15 has official binary packages packages for Raptor Lake cards,
14 apparently 14-CURRENT also has it but since packages are built per major release, the 14 is probably based on 14.0.

I really encourage you to test the current, specially after the last drm-kmod update done last week,
I'm running this version since this afternoon and I'm really happy with it.

JasonSmith, 2024-06-11:
Thanks for answering so quickly.

I'll have to ponder about switching to freebsd current. I'm just not sure if I can run a system that is that volatile on my main machine.

huanghwh, 2024-06-11:
14.1 is works fine with drm-61-kmod, you may need to compile the port by yourself.

JasonSmith, 2024-06-12:
I did compile drm-61-kmod from the source tree. It was marked as ignore so there was no pre built package for it. Still, I am unable to have X11 use the intel driver or have glxinfo show anything other than mesa LLVM pipe.

Let me know if you have any suggestions. I'd love for there to be a documented answer for any other framework users.

huanghwh, 2024-06-12:
should not ignore?

```
more drm-61-kmod/Makefile
.if ${OPSYS} == FreeBSD && !( ${OSVERSION} >= 1500008 || ( ${OSVERSION} >= 1400508 && ${OSVERSION} < 1500000 ))
IGNORE=         not supported on older than 14-STABLE 1400508, no kernel support
.endif
.if ${OPSYS} != FreeBSD
IGNORE=         not supported on anything but FreeBSD (missing linuxkpi functionality)
.endif

% uname -aKU
FreeBSD mbxp.jopens.cn 14.1-RELEASE FreeBSD 14.1-RELEASE #5 n267679-10e31f0946d8-dirty: Tue Jun 11 11:17:29 CST 2024     root@mbxp.jopens.cn:/usr/obj/usr/src/amd64.amd64/sys/MBXP amd64 1401000 1401000
```

JasonSmith, 2024-06-12:
My apologies, when i say it is marked as ignored. I was going off of the FreshPorts page.

[image]

I did, however, try to install the binary package and it was not found in the repositories.

[image]

Which is why I tried to build it from the ports tree.

Sorry about the screen shots, I don't have access to another computer at the moment that I can use to ssh into my laptop so I can copy paste.

evadot, 2024-06-13:
It's not yet available as a package for 14.1 as packages are still built under 14.0 so yes for now you have to build it yourself.
I don't have a RaptorLake device but based on feedback it should work fine with 6.1 (And I have feedback that it worked on this specific laptop iirc).
Make sure you have the firmwares installed (use fwget(8), if it doesn't install anything please install all the gpu-firmware-intel-* packages).

mzperx7, 2024-07-01:
@evadot 
Regarding the firmswares to be installed: if I make  search for "firmware-intel" in FreeBSD Ports Search page I get results for different GPUs but none of them is Raptor Lake.

JasonSmith, 2024-07-07:
@evadot Sorry, I got really busy for a bit.

Starting with a fresh install of 14.1-release. I do the following.

```

pkg update
pkg install vim git
git clone https://git.FreeBSD.org/ports.git /usr/ports
git clone https://git.FreeBSD.org/src.git /usr/src
cd /usr/ports/graphics/gpu-firmware-intel-kmod
make install
cd /usr/ports/graphics/drm-61-kmod
make install
sysrc kld_list+=i915kms
shutdown -r now

```

When the system restarts it is unable to load the driver.

`fwget -n`
```

No package found for device 0xa7a0
No firmware packages to install.

```

`pciconf -lv | grep -A 3 a7a0`
```

vgapci0@pci0:0:2:0:     class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0xa7a0 subvendor=0xf111 subdevice=0x0003
     vendor     = 'Intel Corporation'
     device     = 'Raptor Lake-P [Iris Xe Graphics]'
     class       = display

```

Either I'm missing something, perfectly possible, or their is no firmware for Raptor Lake to use.

It is kind of interesting that when I tell it to load the i915kms module it seems to knock out my wifi driver(intel ax210) from loading as well.

decke, 2024-07-07:
You are using kernel sources for 15-current right now (main branch). Checkout releng/14.1 will work for you.

JasonSmith, 2024-07-08:
@decke that seems to have done it. Thank you very much. Now I get:

`es2_info | grep GL_RENDERER`
```

GL_RENDERER: Mesa Intel(R) Graphics (RPL-P)

```

Now that everything can use the hardware acceleration the computer's fan runs a lot less.

Thanks everyone for all your hard work making FreeBSD function.

closed 2024-07-08 by JasonSmith