freebsd/drm-kmod #308: Hyprland crash at start since 0.41.1

Canvis-Me, 2024-06-15:
**Describe the bug**
Since the port `hyprland` updated to 0.41.1, I can't run it at start. The situation
is the same with 0.41.1_1 and 0.41.1_2. Other ports are up to date.

**FreeBSD version**
FreeBSD zfd.mechot.canvis 14.1-RELEASE FreeBSD 14.1-RELEASE releng/14.1-n267679-10e31f0946d8 GENERIC amd64 1401000 1401000

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
amdiommu0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1481 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib11@pci0:0:3:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:7:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib12@pci0:0:7:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1043 subdevice=0x8808
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
[... 370 lines cut ...]
#10 0xffffffff828889d5 in __nv_drm_gem_nvkms_handle_vma_fault () from /boot/modules/nvidia-drm.ko
#11 0xffffffff80908ea2 in linux_cdev_pager_populate (vm_obj=0xfffff803aa7c7d00, pidx=<optimized out>,
    fault_type=<optimized out>, max_prot=<optimized out>, first=0xfffffe01929dbc00,
    last=0xfffffe01929dbc18) at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:516
#12 0xffffffff809bba67 in vm_pager_populate (object=0x75e3ba3d79a3ba51, object@entry=0xfffffe01929dbb80,
    pidx=16, fault_type=1024, first=0xfffffe01929dbc00, last=0xfffffe01929dbc18,
    max_prot=<optimized out>) at /usr/src/sys/vm/vm_pager.h:187
#13 vm_fault_populate (fs=0xfffffe01929dbca0) at /usr/src/sys/vm/vm_fault.c:515
#14 vm_fault_allocate (fs=fs@entry=0xfffffe01929dbca0) at /usr/src/sys/vm/vm_fault.c:1240
#15 0xffffffff809ba456 in vm_fault_object (fs=0xfffffe01929dbca0, behindp=0xfffffe01929dbc94,
    aheadp=0xfffffe01929dbc98) at /usr/src/sys/vm/vm_fault.c:1504
#16 vm_fault (map=<optimized out>, vaddr=36335472640, fault_type=2 '\002', fault_flags=<optimized out>,
    m_hold=m_hold@entry=0x0) at /usr/src/sys/vm/vm_fault.c:1638
#17 0xffffffff809b9fcd in vm_fault_trap (map=<optimized out>, vaddr=<optimized out>,
    fault_type=<optimized out>, fault_flags=fault_flags@entry=0, signo=0xfffffe01929dbf00,
    ucode=0xfffffe01929dbf04) at /usr/src/sys/vm/vm_fault.c:712
#18 0xffffffff80a602de in trap_pfault (frame=frame@entry=0xfffffe01929dbf40, usermode=true,
    signo=0xfffffe01929dbc00, signo@entry=0xfffffe01929dbf00, ucode=0xfffffe01929dbc18,
    ucode@entry=0xfffffe01929dbf04) at /usr/src/sys/amd64/amd64/trap.c:845
#19 0xffffffff80a5f90b in trap (frame=0xfffffe01929dbf40) at /usr/src/sys/amd64/amd64/trap.c:384
#20 <signal handler called>
#21 0x0000000834f3b33a in ?? ()
Backtrace stopped: Cannot access memory at address 0x82157cfe8
(kgdb) q
</details>

closed 2024-06-16 by Canvis-Me

rhelmot, 2025-04-16:
Were you ever able to diagnose this more precisely? I am encountering a very similar backtrace on FreeBSD 14.2 + drm-kmod 6.1 + nvidia-drm-kmod 570.124.04, but with kwin_wayland instead of hyprland.

Canvis-Me, 2025-04-17:
If I remember right, it was because `Hyprland` switch to their own wayland implementation ( get rid off wlroots). But I didn't dig into the bug, and switch to `sway` after that. KDE wayland on FreeBSD is still buggy now, it's better to wait for more developments.