freebsd/drm-kmod #310: drmn0: [drm] *ERROR* Fault errors on pipe A: 0x00000080

freebsdcross, 2024-06-27:
**Describe the bug**
While running ports compiled from git checkout on June 21 on FreeBSD 14.1-Stable running gnome desktop and firefox the screen repeatedly reset and flashed becoming unusable.  The error "drmn0: [drm] *ERROR* Fault errors on pipe A: 0x00000080" displayed repeatedly in the console while switched to ttyv0.  Interestingly the mouse cursor on the screen was unaffected, and between flashes I could see the screen rendered correctly

**FreeBSD version**
FreeBSD HOSTNAME 14.1-RELEASE FreeBSD 14.1-RELEASE releng/14.1-n267679-10e31f0936d8 GENERIC amd64 1401000 1401000

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
Paste the output of `pciconf -lv`

</details>

**DRM KMOD version**
Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`

**To Reproduce**
start gdm, log in, open firefox, go to upper left of screen to get to app selector.

*I will fill in pciconf -v shortly, i am trying a rebuild to test something.*

hurzl, 2024-07-22:
It happens when videos are played full screen. But not with `mpv` on the console.

knaggsy2000, 2024-09-19:
Been getting quite a few issues myself, see my posted issues.  I've played videos from YouTube from Firefox without any issues for many many months while fullscreen.

Plus the DRM version of the package has not been defined as per post should have mentioned: ````Paste the output of pkg query -x '%n %v' 'drm.*kmod'````

But there does seems to be an issue with v14.1, many others are experiencing issues.