freebsd/drm-kmod #311: FreeBSD 14.1-RELEASE-p2 panics with amdgpu when SDDM starts

sc68cal, 2024-07-25:
**Describe the bug**

My desktop FreeBSD machine was working great with KDE-5 plasma and `amdgpu` kmod until I used `freebsd-update` to update to `FreeBSD 14.1-RELEASE-p2`

During multiuser boot, SDDM starts and then the kernel panics.

**FreeBSD version**
FreeBSD 14.1-RELEASE-p2 (bad)

`FreeBSD thor 14.1-RELEASE FreeBSD 14.1-RELEASE releng/14.1-n267679-10e31f0946d8 GENERIC amd64 1401000 1401000` (good)

**PCI Info**

<summary>
<details>
hostb0@pci0:0:0:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1849 subdevice=0x1480
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
hostb1@pci0:0:1:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:4:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:7:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib8@pci0:0:7:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1022 subdevice=0x1484
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
    class      = bridge
    subclass   = PCI-PCI
hostb7@pci0:0:8:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib9@pci0:0:8:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1022 subdevice=0x1484
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
    class      = bridge
    subclass   = PCI-PCI
[... 153 lines cut ...]
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
ahci3@pci0:11:0:0:      class=0x010601 rev=0x51 hdr=0x00 vendor=0x1022 device=0x7901 subvendor=0x1849 subdevice=0xffff
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
</details>
</summary>

**DRM KMOD version**
```
drm-515-kmod 5.15.160
drm-kmod 20220907_3
```

**To Reproduce**
Steps to reproduce the behavior:

Start SDDM

**Additional context**

I have a kernel panic log that I can link

sc68cal, 2024-07-26:
So, what I noticed is that if I booted back into the older `14.1-RELEASE` kernel, I could get SDDM to start. I could then boot into the newer `14.1-RELEASE-p2` kernel and it would run and load SDDM, but upon reboot, it would keep crashing. I would have to boot into the older kernel to get it back into a state where it would not crash.

There was also some weirdness around the boot environment (I use ZFS) where at one point I had to boot into the `-p2` userland but use the `kernel.old` to get things working, and then I set my boot environment to boot to the older boot environment and now everything works.

There's something about the `-p2` kernel that really messes things up.

sc68cal, 2024-07-26:
I will try and attach the full `core.txt` files somewhere, gist.github.com is not liking it.

sc68cal, 2024-07-26:
```
Tracing command "Xorg", '\000' <repeats 15 times> pid 1649 tid 100507 (CPU 2)
#0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:57
#1  doadump (textdump=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:405
#2  0xffffffff80b324f7 in kern_reboot (howto=260)
    at /usr/src/sys/kern/kern_shutdown.c:523
#3  0xffffffff80b329ce in vpanic (
    fmt=0xffffffff81115a4f "%s: fault on nofault entry, addr: %#lx", 
    ap=ap@entry=0xfffffe0158a90540) at /usr/src/sys/kern/kern_shutdown.c:967
#4  0xffffffff80b32823 in panic (fmt=<unavailable>)
    at /usr/src/sys/kern/kern_shutdown.c:891
#5  0xffffffff80ea670f in vm_fault_lookup (fs=0xfffffe0158a905b0)
    at /usr/src/sys/vm/vm_fault.c:912
#6  vm_fault (map=<optimized out>, vaddr=18446741880512184320, 
    fault_type=2 '\002', fault_flags=<optimized out>, m_hold=m_hold@entry=0x0)
    at /usr/src/sys/vm/vm_fault.c:1569
#7  0xffffffff80ea5081 in vm_fault_trap (map=<optimized out>, 
    vaddr=<optimized out>, fault_type=<optimized out>, 
    fault_flags=<unavailable>, fault_flags@entry=0, signo=0x0, ucode=0x0)
    at /usr/src/sys/vm/vm_fault.c:712
#8  0xffffffff80fffade in trap_pfault (frame=0xfffffe0158a90730, 
    usermode=false, signo=<unavailable>, ucode=<unavailable>)
    at /usr/src/sys/amd64/amd64/trap.c:845
#9  <signal handler called>
#10 0xffffffff83a34d74 in amdgpu_bo_add_to_shadow_list ()
   from /boot/modules/amdgpu.ko
#11 0xffffffff83a5c200 in amdgpu_vm_pt_create () from /boot/modules/amdgpu.ko
#12 0xffffffff83a5bc91 in amdgpu_vm_init () from /boot/modules/amdgpu.ko
#13 0xffffffff83a32064 in amdgpu_driver_open_kms ()
   from /boot/modules/amdgpu.ko
#14 0xffffffff8391164a in drm_file_alloc () from /boot/modules/drm.ko
#15 0xffffffff83911af4 in drm_open () from /boot/modules/drm.ko
#16 0xffffffff83907522 in drm_stub_open () from /boot/modules/drm.ko
#17 0xffffffff80d996d1 in linux_dev_fdopen (dev=<optimized out>, 
    fflags=<optimized out>, td=0x0, file=0xfffff80037af2740)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:762
#18 0xffffffff809c18ce in devfs_open (ap=0xfffffe0158a90ab0)
    at /usr/src/sys/fs/devfs/devfs_vnops.c:1294
#19 0xffffffff810d3b8d in VOP_OPEN_APV (
    vop=0xffffffff818ab788 <devfs_specops>, a=a@entry=0xfffffe0158a90ab0)
    at vnode_if.c:434
#20 0xffffffff80c2d543 in VOP_OPEN (vp=0xfffff801b3e13a80, mode=1048579, 
    cred=0xfffff80012349a00, fp=0xfffff80012a84d20, td=<optimized out>)
    at ./vnode_if.h:218
#21 vn_open_vnode (vp=0xfffff801b3e13a80, fmode=fmode@entry=1048579, 
    cred=cred@entry=0xfffff80012349a00, td=td@entry=0xfffff80037af2740, 
    fp=0xfffff80012a84d20) at /usr/src/sys/kern/vfs_vnops.c:450
#22 0xffffffff80c2d080 in vn_open_cred (ndp=ndp@entry=0xfffffe0158a90c90, 
    flagp=flagp@entry=0xfffffe0158a90da4, cmode=cmode@entry=0, 
    vn_open_flags=vn_open_flags@entry=16, cred=0xfffff80012349a00, 
    fp=0xfffff80012a84d20) at /usr/src/sys/kern/vfs_vnops.c:337
#23 0xffffffff80c236a8 in openatfp (td=0xfffff80037af2740, dirfd=-100, 
    path=0x15b33a92d2c0 <error: Cannot access memory at address 0x15b33a92d2c0>, pathseg=pathseg@entry=UIO_USERSPACE, flags=1048579, mode=<optimized out>, 
    fpp=0x0) at /usr/src/sys/kern/vfs_syscalls.c:1173
#24 0xffffffff80c2341d in kern_openat (dirfd=1531523744, 
    path=0xfffffe015b4002a0 "", pathseg=UIO_USERSPACE, flags=312331808, 
    mode=304464416, td=<optimized out>)
    at /usr/src/sys/kern/vfs_syscalls.c:1278
#25 sys_openat (td=<optimized out>, uap=<optimized out>)
    at /usr/src/sys/kern/vfs_syscalls.c:1111
#26 0xffffffff810001c0 in syscallenter (td=0xfffff80037af2740)
    at /usr/src/sys/amd64/amd64/../../kern/subr_syscall.c:188
#27 amd64_syscall (td=0xfffff80037af2740, traced=0)
    at /usr/src/sys/amd64/amd64/trap.c:1194
#28 <signal handler called>
#29 0x00000008283374aa in ?? ()
Backtrace stopped: Cannot access memory at address 0x820d1d208

```

donaldcallen, 2024-07-26:
I've been bitten by this as well, on an AMD-based machine that is my primary computer. I ran 14.0 happily on this machine and it was dead reliable. Since installing 14.1, I am seeing a full system crash or two per week. This is a show-stopper for me -- too disruptive. I am re-installing NixOS as we speak.

sc68cal, 2024-07-28:
I have started getting kernel panics even with the "known good" ZFS boot environment for `14.1-RELEASE`.

I have fiddled with my `rc.conf` and removed `fusefs` from my `kld_loadlist` and also removed `linux_enable=YES`.

Without the `fusefs` driver loaded and the linux compatibility later not running, I am able to boot into SDDM with the `14-1-RELEASE-p2` boot environment.

sc68cal, 2024-07-28:
Another crash after resuming from suspend, no linux compat loaded, was able to post a full crash text file to gist.

https://gist.github.com/sc68cal/0d71c26141300a8807b27f92005003fe

FloppyKing, 2024-09-09:
I've also been encountering similar issues after upgrading to 14.1.  I have 5 machines all with the A12-9800E APU.  The graphics driver throws a fit 100% of the time after sddm hands off to KDE5 (x11) when autologin is enabled.  If I don't have it do autologin, then it loads into KDE fine.  Also since updating to 14.1 the display will no longer "wake back up" after the screen turns off from being idle.  As is, i've had to disable autologin and display sleep.  Would very much like to see this fixed.

knaggsy2000, 2024-09-17:
Getting a similar issues too [myself](https://github.com/freebsd/drm-kmod/issues/317).

knaggsy2000, 2024-09-18:
But the drm-61 packaged is NOT available by pkg or via source - the src fails as I've mentioned in my post.

ClausAndersen, 2024-09-20:
Ports is not enough to build `/usr/ports/graphics/drm-61-kmod` you need to have `usr/src` as well.

donaldcallen, 2024-09-20:
> Ports is not enough to build /usr/ports/graphics/drm-61-kmod you need to
> have /usr/src as well.

Yes, that's true, as I discovered. So I installed the sources, did a 'make
install' of drm-61-kmod, which now completed without error, rebooted my
system and ran into the problem I reported -- no graphic devices found.
After the multiple problems I encountered with FreeBSD, I am now running
Arch Linux on the system without any issues.

ClausAndersen, 2024-09-20:
Sorry! Did not mean to imply! That remark was aimed at @knaggsy2000 who in his PR seems to be missing `/usr/src/`.

Did NixOS cause issues as well or are you just distro hopping to Arch?

As for @sc68cal original problem there has been kernel changes as he sussed out. 

`ports/graphics/drm-61-kmod/Makefile`

    IGNORE=         not supported on older than 14-STABLE 1400508, no kernel support

Current 14.1p2 kernel version is 1401000.

So with the p2 kernel or later I would go with drm-61. Or at least test it. For @FloppyKing (see above) it has been a success as reported on the [forum](https://forums.freebsd.org/threads/amd-gpu-driver-has-a-serious-bug-introduced-in-14-1.94338/#post-671127) as well.

knaggsy2000, 2024-09-20:
@ClausAndersen Yes, src is installed - always install it as sometimes I need to make a custom kernel, not BUT in this case: -

````
# ll /usr/src
total 598
-rw-r--r--    1 root wheel    148 Nov 23  2023 .arcconfig
-rw-r--r--    1 root wheel    448 Feb 13  2022 .arclint
drwxr-xr-x    2 root wheel      3 May 19  2022 .cirrus-ci/
-rw-r--r--    1 root wheel   5462 Jun 16 23:14 .cirrus.yml
-rw-r--r--    1 root wheel   4827 Nov 23  2023 .clang-format
-rw-r--r--    1 root wheel    991 Jun 16 23:14 .git-blame-ignore-revs
-rw-r--r--    1 root wheel    205 Feb 13  2022 .gitattributes
drwxr-xr-x    4 root wheel      5 Nov 23  2023 .github/
-rw-r--r--    1 root wheel    297 Nov 23  2023 .gitignore
drwxr-xr-x    4 root wheel      8 Feb 13  2022 .svn/
-rw-r--r--    1 root wheel   7903 Jun 16 23:14 CONTRIBUTING.md
-rw-r--r--    1 root wheel   6109 Nov 23  2023 COPYRIGHT
-rw-r--r--    1 root wheel    490 Nov 23  2023 LOCKS
-rw-r--r--    1 root wheel   7175 Nov 23  2023 MAINTAINERS
-rw-r--r--    1 root wheel  31522 Jun 16 23:14 Makefile
-rw-r--r--    1 root wheel 131226 Jun 16 23:14 Makefile.inc1
-rw-r--r--    1 root wheel   3621 Nov 23  2023 Makefile.libcompat
-rw-r--r--    1 root wheel   1939 Nov 23  2023 Makefile.sys.inc
-rw-r--r--    1 root wheel 931131 Jun 16 23:14 ObsoleteFiles.inc
-rw-r--r--    1 root wheel   2837 Nov 23  2023 README.md
-rw-r--r--    1 root wheel  12396 Jun 16 23:14 RELNOTES
-rw-r--r--    1 root wheel  87182 Jun 16 23:14 UPDATING
drwxr-xr-x   44 root wheel     46 Nov 23  2023 bin/
drwxr-xr-x   11 root wheel     13 Nov 23  2023 cddl/
drwxr-xr-x   90 root wheel     90 Jun 16 23:32 contrib/
drwxr-xr-x    5 root wheel      6 Jun 16 23:25 crypto/
drwxr-xr-x    8 root wheel     13 Jun 16 23:26 etc/
drwxr-xr-x    5 root wheel      9 Nov 23  2023 gnu/
drwxr-xr-x   10 root wheel    112 Jun 16 23:26 include/
drwxr-xr-x    8 root wheel     11 Nov 23  2023 kerberos5/
drwxr-xr-x  146 root wheel    148 Jun 16 23:26 lib/
drwxr-xr-x   40 root wheel     42 Jun 16 23:28 libexec/
drwxr-xr-x   12 root wheel     24 Jun 16 23:28 release/
drwxr-xr-x    4 root wheel      6 Nov 23  2023 rescue/
drwxr-xr-x   87 root wheel     93 Nov 23  2023 sbin/
drwxr-xr-x    8 root wheel     11 Nov 23  2023 secure/
drwxr-xr-x   31 root wheel     33 Nov 23  2023 share/
drwxr-xr-x   24 root wheel     33 Jun 16 23:29 stand/
drwxr-xr-x   54 root wheel     56 Nov 23  2023 sys/
drwxr-xr-x    3 root wheel      6 Nov 23  2023 targets/
drwxr-xr-x    8 root wheel     15 Nov 23  2023 tests/
drwxr-xr-x   19 root wheel     23 Nov 23  2023 tools/
drwxr-xr-x  278 root wheel    284 Jun 16 23:31 usr.bin/
drwxr-xr-x  227 root wheel    235 Nov 23  2023 usr.sbin/
````

See you have also mentioned something on my OG post.  Let me look at that...

knaggsy2000, 2024-09-20:
@sc68cal Tell us more about that panics.

knaggsy2000, 2024-09-20:
@FloppyKing That was the article that I found too.  Originally, wasn't able to compile from ports.  Turns out my src folder was quite out-of-date and wiping it and cloning from new, allowed me to compile drm-61.

knaggsy2000, 2024-09-20:
drm-61 *IS* able to built from src, please look at my issue for info #317

donaldcallen, 2024-09-21:
> Sorry! Did not mean to imply! That remark was aimed at @knaggsy2000
> <https://github.com/knaggsy2000> who in his PR seems to be missing
> /usr/src/.
No offense taken.

> Did NixOS cause issues as well or are you just distro hopping to Arch?
NixOS works as advertised. No particular problems. But it does impose
performance and complexity penalties that became tiresome. Arch has been my
Linux distribution of choice for years, I had a Clonezilla backup of it
for the machine in question, so after the FreeBSD problems surfaced, I just
restored the tried and true Arch and updated it.

ClausAndersen, 2024-09-21:
Why is this happening? If others in the future finds the issue it might be interesting with some background for the uninitiated.

It takes a lot of resources to write a driver from scratch. And some informationen needed to do it might not be publicly available. That is why FreeBSD is actually using Linux drivers with some patches. FreeBSD is a full operating system (kernel+userland) and does not share code with the Linux kernel nor the typical GNU userland utilities found in many Linux Distributions (Linux kernel+distribution specific userland). FreeBSD does however have a high degree or Linux compatibility and is able to run nativer Linux binaries.

The graphics drivers (in this case amdgpu) comes (mainly) from the Linux world and they depend on a Linux kernel feature called DRM ([Direct Rendering Manager](https://en.wikipedia.org/wiki/Direct_Rendering_Manager)). DRM evolves with the Linux Kernel.

In FreeBSD this is handled with the DRM kernel module. Notice the major and minor version numbers on this module: 5.10, 5.15 and now 6.10. This corresponds to the the versions of the Linux Kernel it tries to match. With the transition from 5 to 6 we are now moving a full major version forward.

I do not know the finer details but "something" happened around the 14.1-RELEASE-p2 kernel. Most likely what has been considered a bugfix in the Linux compatibility layer. You would not see new features introduced in a p1 -> p2 point update. This has then unfortunately triggered some side issues with the amdgpu driver. Exactly what and who is to blame I will not speculate on. Ít is typical for users to upgrade their packages at the same time as the system. So they might get a newer amdgpu driver. Or the issue has been latent with the driver for a longer time. This then only becomes visible with the bugfixed kernel. It might be an issue which was fixed in the Linux world but relies on newer DRM features. There is many small devils hidden here.

Would this not be rather trivial if we had the drm-61 as a binary package ready to go? Yes! But. 

ports/graphics/drm-61-kmod/Makefile

    IGNORE=         not supported on older than 14-STABLE 1400508, no kernel support
    
All the 14 pre-built pacakages must work on all kernels - even 1400000. That is why you cannot find a pre-built package and need to build it yourself. This is fairly easy if you understand how to build ports. This then tricks some as this port is a little special. It needs some of the Linux compatibility headers from the FreeBSD kernel. Regular users do normally not need them of have them installed. So that is an extra still simple but unusual step. And remember that when you have built your own version you have a binary version which you can use on other hosts.

So bugs do happen. And fixes are available. In my mind this could have been handled much more smoothly if the project had made a binary package available from an official location for download. I understand why it was not simply added to the package repository because of backwards compatibility. But a direct download would have been nice for regular users.

It might also be worth reading the [DRM Kmod Primer](https://freebsddesktop.github.io/2018/12/08/drm-kmod-primer.html).

donaldcallen, 2024-09-21:
This is a very useful explanation.

My difficulty would not have been solved by a 61 package. As I've said
previously, I got drm-61-kmod built and installed as a port and it did not
work on my system, as I've described elsewhere, I provided a dmesg that
shows that my hardware is vanilla, the
gpu on the processor chip. And 515 crashes the system periodically (a few
times a week with constant use).

I have also experienced consistent failures to properly reawaken a mounted
USB drive after use of zzz, resulting in the need to fsck the drive.

I also experienced a softdep-related issue with my root file-system as a
result of a 515 crash. This required a bit of work to get the system back
on its feet, at which point I backed up the files I'd been working on,
which were intact, and restored Arch Linux.

I must say I'm more than a bit surprised to encounter softdep problems at
this point in its history. The same for the USB issue. The USB driver was a
huge problem for FreeBSD many years ago and I believe it was re-written.
I've had no problems with it
using FreeBSD in recent years until now.

I realize that FreeBSD doesn't have the resources or the user community
that Linux has. But I used both 13-series versions of FreeBSD as well as
14.0 without significant issues. So hopefully my experience is not an
indication that the complexity
of providing FreeBSD for the desktop is beyond the capacity of the project
and what I've encountered is not indicative of a systemic problem.

knaggsy2000, 2024-09-22:
@ClausAndersen I will state that he (Claus) gives very verbose and detailed replies to people.  I like that - and should be valued for that fact.  Thank you, Claus - please keep it up.

@donaldcallen I've used ZFS ever since it was available in FreeBSD for all my drives, and glad that I did.  Have you considered to use ZFS?  Just wondering.  Don't get me wrong, I've used UFS for a long time.

donaldcallen, 2024-09-22:
> @ClausAndersen <https://github.com/ClausAndersen> I will state that he
> (Claus) gives very verbose and detailed replies to people. I like that -
> and should be valued for that fact. Thank you, Claus - [...]

If my softdep problem was the only issue, I'd consider using ZFS instead of
UFS. But the graphics driver bug is a show-stopper for me and Arch Linux
has replace FreeBSD on my system.

I should also note that I used ZFS in the past, including on the backup
drive that I"ve just reported
not remaining mount after awakening from system suspension. I had issues
with that, too, about which I filed a bug report almost exactly two years
ago. Its status is still 'new'.

I have had an on-again off-again relationship with
FreeBSD for about 20 years, mostly off-again. I run it for awhile, I prefer
it to Linux, and then I run into show-stopping bugs. It has happened every
time I've tried to use the system over a very long period of time. I hope
I'm wrong, but I fear
that the project just does not have the critical mass to do something as
complex as providing a reliable desktop operating system on PC hardware.

yukiteruamano, 2024-09-23:
Hi! 

Right now I'm using 14.1-RELEASE-p5, hw specs Ryzen 7 3800X and RX 580, and I have hard crashes with drm-515-kmod (drm-510-kmod is good), I share my crash file for add more info to this issue:

[crash-rx580.txt](https://github.com/user-attachments/files/17100688/crash-rx580.txt)

knaggsy2000, 2024-09-23:
@yukiteruamano Have you seen my reported issue?  #317

yukiteruamano, 2024-09-23:
> @yukiteruamano Have you seen my reported issue?  #317 

No, but I can test build a drm-610-kmod using poudriere  I have this infra for my own packages and for future collaboration with ports packages

knaggsy2000, 2024-09-23:
@donaldcallen I remember those USB issues several years ago, sometimes even unplugging a USB device could cause a kernel panic.  But no longer see that anymore, for a long time now.

I am a Linux user too, and Arch seemed very appealing to me - but currently settled on Debian for that.

My personal preference of OS will ALWAYS be FreeBSD though, for many reasons - outside the scope of this issue.

But I don't understand why you don't use ZFS (did I miss something?) - I know it has additional hardware requirements, especially more RAM.  Been using FreeBSD since v5 to v14.1 continuously for my main PC.  Seen lots happen during that time.

It's funny how an OS that has originated many years before Linux, appears to get less support: -

````
Copyright (c) 1992-2023 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
        The Regents of the University of California. All rights reserved
````

1979 - wow, the earliest mention.  I'm sure v5 was released around 2003 (please correct me if I'm wrong), so a little more time for me.  The only reason I got into FreeBSD is because at that time, my current laptop couldn't install *ANY* distro of Linux (remember them being on MULTIPLE CDs?).  But when I installed FreeBSD v5, everything worked - so I stuck with it ever since.

Feel like I've gone off-subject a little, I'm sorry.

knaggsy2000, 2024-09-23:
> > @yukiteruamano Have you seen my reported issue?  #317
> 
> No, but I can test build a drm-610-kmod using poudriere I have this infra for my own packages and for future collaboration with ports packages

Please try drm-610-kmod, ensure your src folder and ports folder are completely up-to-date - see my post for details on that.

knaggsy2000, 2024-09-23:
@yukiteruamano As you have a RX 580, which is similar to my card mentioned in my post.  Please update on MY post and not here.

knaggsy2000, 2024-09-26:
The issue with @yukiteruamano has been resolved - see my post for that info,

knaggsy2000, 2024-09-26:
@sc68cal Any updates?  As lots has happened.

sc68cal, 2024-10-09:
@knaggsy2000 unfortunately my system was unusable to the point where I installed Ubuntu so that I could continue to use the system day to day :(

yukiteruamano, 2024-10-09:
The only option in FreeBSD 14 for use AMD GPUs is stick to drm-510-kmod or
install drm-61-kmod, 515-kmod is very unstable with various AMD cards

donaldcallen, 2024-10-09:
> The only option in FreeBSD 14 for use AMD GPUs is stick to drm-510-kmod or
> install drm-61-kmod, 515-kmod is very unstable with various AMD cards

Yes, as it was with mine:

30:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI]
Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c9)

Version 61 didn't work at all with my system.

I'm running Arch Linux on the same system now. No problems whatsoever.

I will add that my experience with FreeBSD, over about 20 years, is that,
for whatever reason, the project is glacially slow to respond to serious
issues like this, even
when they are properly reported to the bug database. This adds to my
theory, that I hope/wish I'm wrong about,  that supporting FreeBSD on the
desktop is just too resource-intensive given the
person-power the project has.  This is Linux's big advantage over the BSDs.
You can cite the technical superiority of the BSDs over Linux from now
until the end of time, but if they don't get the details right, make the
systems run reliably, all that citing is moot.

knaggsy2000, 2024-11-02:
@sc68cal I compiled drm-61-kmod from a fresh src folder and that solved my issue.  That's why I've locked that package now.

The "fresh" part is VERY important, as they have been very many src updates.  I can tell you how to update both the ports and src in FreeBSD now, as that's changed - even since I've being using FreeBSD since v5.

To update your ports (ensure the folder is EMPTY): -

```
# pkg install git
# git clone --depth 1 https://git.FreeBSD.org/ports.git /usr/ports
```

Then to update it: -

`# git -C /usr/ports pull`

If you need to get the latest src folder: -

```
# cd /usr
# git clone https://git.freebsd.org/src.git /usr/src
# git checkout releng/14.1
```

Then to update src: -

`# git -C /usr/src pull`

Appears none of the FreeBSD documentation is stating that these folders are going towards "git" updates.

These are gathered from my previous posts - hope it helps people.

knaggsy2000, 2024-11-02:
@donaldcallen Have you tried the options I've mentioned?  Like yourself, I've used FreeBSD for over two decades but I never left it.  Always stuck though things.

donaldcallen, 2024-11-02:
> @donaldcallen <https://github.com/donaldcallen> Have you tried the
> options I've mentioned? Like yourself, I've used FreeBSD for over two
> decades but I never left it. Always stuck though things.

FreeBSD is a tool that I used to do my work. It disrupted my ability to do
that work, so I replaced it with a more reliable tool, Arch Linux.

knaggsy2000, 2024-11-02:
@donaldcallen What kind of work was that?  If you don't me asking.

Yes, I've had my issues with FreeBSD but *NEVER* enough to change to another OS *EVER*.  All the Linux distros...  No, this is NOT the place to talk about that.

jeannekamikaze, 2024-11-12:
Does it work for you if you `startx` with the following `.xinitrc` instead of `sddm_enable=YES` in `/etc/rc.conf`?

```
# ~/.xinitrc
exec dbus-launch --exit-with-x11 ck-launch-session startplasma-x11
```

I am on 15-CURRENT and my SDDM also hangs when it starts at boot on an AMD Phoenix (RDNA 3000) laptop. For some reason, it works if I don't start SDDM at boot and instead `startx`.

Most of the times, the system becomes unresponsive, but it once dumped into the kernel debugger and showed what looked to be a look-after-free. If I am able to catch this again, I'll try to share more details.

jeannekamikaze, 2024-11-12:
By the way, how can I start and attach a debugger to SDDM? Any tips to narrow down this thing would help.