freebsd/drm-kmod #312: Resume failing on latest CURRENT for both 6.6 and 6.1-lts

orbitz, 2024-08-19:
**Describe the bug**
When I resume from suspend, the screen stays black and I must hard-reboot.

**FreeBSD version**

`FreeBSD bix 15.0-CURRENT FreeBSD 15.0-CURRENT #23 main-n271694-0a9d1da6e6ce: Sun Aug 18 19:35:33 +00 2024     root@bix:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1500023 1500023`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  

hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x4621 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x46a6 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P GT2 [Iris Xe Graphics]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x461d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake Innovation Platform Framework Processor Participant'
    class      = dasp
pcib1@pci0:0:6:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x464d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor PCI Express x4 Controller'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x466e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:7:1:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x463f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:7:2:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x462f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:7:3:	class=0x060400 rev=0x02 hdr=0x01 vendor=0x8086 device=0x461f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:8:0:	class=0x088000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x464f subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = '12th Gen Core Processor Gaussian & Neural Accelerator'
    class      = base peripheral
none2@pci0:0:10:0:	class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x467d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Platform Monitoring Technology'
    class      = dasp
xhci0@pci0:0:13:0:	class=0x0c0330 rev=0x02 hdr=0x00 vendor=0x8086 device=0x461e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 USB Controller'
    class      = serial bus
    subclass   = USB
none3@pci0:0:13:2:	class=0x0c0340 rev=0x02 hdr=0x00 vendor=0x8086 device=0x463e subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
none4@pci0:0:13:3:	class=0x0c0340 rev=0x02 hdr=0x00 vendor=0x8086 device=0x466d subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P Thunderbolt 4 NHI'
    class      = serial bus
    subclass   = USB
xhci1@pci0:0:20:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ed subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH USB 3.2 xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none5@pci0:0:20:2:	class=0x050000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51ef subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Shared SRAM'
    class      = memory
    subclass   = RAM
ig4iic0@pci0:0:21:0:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e8 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
ig4iic1@pci0:0:21:1:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e9 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
ig4iic2@pci0:0:21:3:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51eb subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH Serial IO I2C Controller'
    class      = serial bus
none6@pci0:0:22:0:	class=0x078000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e0 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH HECI Controller'
    class      = simple comms
none7@pci0:0:22:3:	class=0x070002 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51e3 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake AMT SOL Redirection'
    class      = simple comms
    subclass   = UART
pcib6@pci0:0:29:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x51b0 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x5182 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:	class=0x040380 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51c8 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a3 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake PCH-P SMBus Host Controller'
    class      = serial bus
    subclass   = SMBus
none8@pci0:0:31:5:	class=0x0c8000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x51a4 subvendor=0xf111 subdevice=0x0002
    vendor     = 'Intel Corporation'
    device     = 'Alder Lake-P PCH SPI Controller'
    class      = serial bus
nvme0@pci0:1:0:0:	class=0x010802 rev=0x01 hdr=0x00 vendor=0x15b7 device=0x5017 subvendor=0x15b7 subdevice=0x5017
    vendor     = 'Sandisk Corp'
    device     = 'WD Black SN770 / PC SN740 256GB / PC SN560 (DRAM-less) NVMe SSD'
    class      = mass storage
    subclass   = NVM
iwlwifi0@pci0:166:0:0:	class=0x028000 rev=0x1a hdr=0x00 vendor=0x8086 device=0x2725 subvendor=0x8086 subdevice=0x0024
    vendor     = 'Intel Corporation'
    device     = 'Wi-Fi 6E(802.11ax) AX210/AX1675* 2x2 [Typhoon Peak]'
    class      = network
</details>

**DRM KMOD version**

6.1-lts 

61e5ade08b996bd17c09c1b54ffb37a025e059d5

**To Reproduce**
Steps to reproduce the behavior:

Install the latest CURRENT and either 6.1-lts or 6.6, suspend the system and try to resume.

**Screenshots**
No screenshots

**Additional context**

I created a ticket in FreeBSD bug tracker as well:

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=280923

grahamperrin, 2024-08-22:
> 1500023

Reproducible with `1500022`?

orbitz, 2024-08-22:
After upgrading to latest kernel commits, resume is working

```
FreeBSD bix 15.0-CURRENT FreeBSD 15.0-CURRENT #24 main-n271769-0875f3cd74b2: Tue Aug 20 19:38:14 +00 2024     root@bix:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64 1500023 1500023
```

closed 2024-08-22 by orbitz

grahamperrin, 2024-08-24:
Thanks. 

<https://cgit.freebsd.org/src/log/?qt=range&q=0a9d1da6e6ce..0875f3cd74b2>

grahamperrin, 2024-09-04:
<https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=280923#c3> reopened. 

When _not_ bugged: 

- does the display flash on, then off, for split-seconds, before reappearing with console text?

orbitz, 2024-10-07:
Closed gain, I have not had any issues with this in awhile.