freebsd/drm-kmod #316: RFE: Xe driver support

DemiMarie, 2024-09-12:
Lunar Lake, Battlemage, and newer Intel graphics cards are Xe2 and require the Xe driver from Linux.

thesunexpress, 2024-12-26:
...aaand also requires Linux 6.6+ at least. We are still at Linux 6.1-lts at best.

mtlll, 2025-06-13:
Now that drm-kmod master is at 6.8, I guess it's at least imaginable to try to bring the Xe driver over. Is anyone working or planning to work on this? I'm interested in trying, since I have an A770 I'm not able to use at the moment. And although I'm a decent C programmer, I have no FreeBSD kernel experience, or any other kernel experience really. Does anyone know if there is anything currently in LinuxKPI or drm-kmod that's required by Xe(the one from 6.8)?

etienne-defenso, 2026-01-14:
> We will relatively soon be at parity with Linux 6.12 and will then carry on to 6.18 and beyond, but porting Xe is not yet started. There's also an issue with the license on some of the Xe files (some are GPL-only) that needs to be resolved.

Thank you for your answer. Do you know if it is possible to use Lunar lake with the i915 driver ?

koinec, 2026-10-04:
Hello.

I own a Lunar Lake laptop and am considering contributing to the porting of the Xe driver. 
Is anyone currently working on this porting effort based on the comments mentioned above?
I cannot gauge the current status from the project's branches or pull requests, so I would appreciate any information you might have.

For my own experiments, I have extracted the xe driver source tree from Linux v6.12 into the `drivers/gpu/drm/xe` directory, created a `bmake` Makefile, and am attempting to build the `xe_**.c` files one by one.
So far, I haven't yet encountered the specific issues reported in the comments mentioned above.
However, it appears that `linuxkpi` is missing at least the following items in addition to those, and I am attempting to add them:
 - Certain functions related to `mempool` functionality
 - `should_fail()` function from `fault-inject.h`
 - `iosys_map_memcpy_from()` function
 - `devm_release_action()` function
 - `string_get_size()` function
 - `bitmap_set_value8()` and `bitmap_shift_left()` functions
 - `pci_dev_msi_enabled()` function
 - `drmm_mutex_init()`, `__drmm_mutex_release()`, `drmm_kmalloc()` functions (for drm-kmod's drm driver)

I have also identified several other issues, such as those involving `tracepoint.h` macros; I anticipate that continuing this work will require a lot of time.
If no one is currently assigned to this task and you are open to accepting pull requests from external contributors, I would like to propose a submission once I have successfully completed the build.

Thank you for your time and consideration.

wulf7, 2026-10-04:
You may start with @lutzbichler work: https://github.com/lutzbichler/drm-kmod/. At least he imported source files and did some initial porting work. I do not know how far hi has advanced.

koinec, 2026-10-05:
Thank you very much for your prompt reply.

@lutzbichler seems to have already ported not only the Xe driver but also drm-kmod and the necessary linuxkpi to Linux 7.3.
Modifications for FreeBSD to support driver loading for Lunar Lake were also added.

I would like to ask @lutzbichler about the plans for future pull requests.

lutzbichler, 2026-10-05:
I have no plans for pull requests. This was only meant to have something for my (LunarLake) laptop until the "real thing" arrives.
The current implementation has (at least) two problems: First, the firmware data is overwritten after loading. I did not manage to find out how/why and worked around it by rewriting it. Second, there are some distortions after startup on LunarLake. Thus, in its current state it is better than nothing, but definitely needs more work.

I have done the updates to see whether this changes something regarding the distortions, but it does not.

koinec, 2026-10-06:
Thank you very much for your reply and for explaining the situation!
Thanks to you, I have gained a good understanding of the current situation and the issues involved.

I was proceeding with the porting work because I wanted to test building and loading the xe driver, but I have absolutely no confidence that I could contribute to investigating or resolving the issues you mentioned.
Therefore, I would like to try reproducing the situation on my end first and then observe the situation for a while.

etienne-bnd, 2026-11-12:
> Thank you very much for your reply and for explaining the situation! Thanks to you, I have gained a good understanding of the current situation and the issues involved.
>
> I was proceeding with the [...]

Hey ! How is it going for you so far on this project ? I'm also considering taking some time to try the portage. Would be nice to start with your knowledge on the topic :)

koinec, 2026-11-15:
To be frank, I am struggling; I have not yet reached the stage where I can verify its operation.

I am attempting a port from scratch, partly as a learning exercise.
So far, I have made 42 commits to linuxkpi and have identified 4 remaining TODO items. For these 4 items, I have created skeleton functions simply to allow the build process to proceed.

Even so, the Xe driver still does not build. Consequently, I am currently working through the compilation errors one by one and refining linuxkpi accordingly.

Once I am able to confirm that it is functioning to some extent - or if I achieve any notable progress - I intend to publish the source tree on GitHub and provide an update here.

However, as the code does not yet build, I am currently working within a local repository.
Please excuse the informal nature of this, but I am sharing a one-line log of the commits from my local linuxkpi repository for your reference.
I would be happy to provide further details if you are interested.

[linuxkpi_drmkmod-xe_commitlog_20261115.txt](https://github.com/user-attachments/files/34871823/linuxkpi_drmkmod-xe_commitlog_20261115.txt)

Please note that I can only dedicate about four hours a week to this project, so I expect it will take quite some time to complete. I apologize for this.
Please let me know if there is any other information you might need.

mtlll, 2026-11-15:
Bravo! I applaud your efforts! I'm too busy currently to provide meaningful assistance writing code, but I'm sure I could find some time to stare at compiler errors/warnings. And I have an A770 and 13900k(which drm-kmod already supports, I know, but the more unique chips can be tested, the better) for testing once that becomes possible.

I'd like to make a suggestion though. Why not push what you have currently to github forks right away, and throw the build errors in this issue? Get more eyeballs on it that way.

etienne-bnd, 2026-11-16:
> To be frank, I am struggling; I have not yet reached the stage where I can verify its operation.
>
> I am attempting a port from scratch, partly as a learning exercise. So far, I have made 42 commits [...]

Amazing work ! How do you manage the problems related to the GPL licence ? 
I am also trying to contribute, if it is something possible can you share your local repository ?