freebsd/drm-kmod #317: Since FreeBSD 14.1 upgrade, getting amdgpu issues with RX 550

knaggsy2000, 2024-09-17:
**Describe the bug**
Ever since I've upgraded from v14.0 to v14.1 been getting kernel panics when the X environment gets loaded - BUT, not every time - it seems to when something needs to access DRM, like in my case when VLC loads a DVD.

I don't use a graphical login manager, I login via console and then issue "startx" - always done this and that's never going to change.

With my mentioned card (in the title - RX 550) it's 100% fails now.  So I swapped it out for my backup gfx card - which is a RX 6500XT, everything works perfectly!  So I ordered another RX550 thinking my gfx card failed in some way (as from personal experiences, kernel panics have always been hardware issues) - well, it turns out that it wasn't.  Fault still happens with NEW hardware - so the issue is clearly with the drivers/firmware.

Upon some research found [this](https://forums.freebsd.org/threads/amd-gpu-driver-has-a-serious-bug-introduced-in-14-1.94338/).  It states that the solution is to install drm-61 instead - great!  But there's multiple problems with that.  First: No pkg currently available to install.  Second: Building from ports/source fails (for me at least): -

````
--- dma-resv.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/dma-resv.c:36:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-resv.h:42:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/ww_mutex.h:34:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/mutex.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/kernel.h:50:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- dma-fence.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/dma-fence.c:27:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence.h:31:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/wait.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- dma-fence-unwrap.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/dma-fence-unwrap.c:10:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence.h:31:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/wait.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- dma-fence-chain.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/dma-fence-chain.c:29:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence-chain.h:30:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence.h:31:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/wait.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- sync_file.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/sync_file.c:10:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/file.h:39:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/fs.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/wait.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- dma-fence-array.o ---
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/drivers/dma-buf/dma-fence-array.c:29:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence-array.h:30:
In file included from /usr/ports/graphics/drm-61-kmod/work/drm-kmod-drm_v6.1.92_0/linuxkpi/bsd/include/linux/dma-fence.h:31:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/wait.h:37:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/sched.h:40:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/bitmap.h:31:
/usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
   39 | #include <linux/llist.h>
      |          ^~~~~~~~~~~~~~~
--- dma-buf.o ---
[... 277 lines cut ...]
none2@pci0:43:0:1:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1486 subvendor=0x1022 subdevice=0x1486
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Cryptographic Coprocessor PSPCPP'
    class      = encrypt/decrypt
xhci1@pci0:43:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x149c subvendor=0x1462 subdevice=0x7b78
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse USB 3.0 Host Controller'
    class      = serial bus
    subclass   = USB
````

**DRM KMOD version**
````
drm-515-kmod 5.15.160
drm-kmod 20220907_3
````

**To Reproduce**
Already mentioned earlier.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Additional info available upon request.

knaggsy2000, 2024-09-18:
Forgot to include the output of `kgdb /boot/kernel/kernel vmcore.last`: -

````
GNU gdb (GDB) 14.1 [GDB v14.1 for FreeBSD]
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-portbld-freebsd14.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /boot/kernel/kernel...
Reading symbols from /usr/lib/debug//boot/kernel/kernel.debug...

Unread portion of the kernel message buffer:
panic: vm_fault_lookup: fault on nofault entry, addr: 0xfffffe01ce36e000
cpuid = 0
time = 1726594283
KDB: stack backtrace:
#0 0xffffffff80b7fefd at kdb_backtrace+0x5d
#1 0xffffffff80b32bd1 at vpanic+0x131
#2 0xffffffff80b32a93 at panic+0x43
#3 0xffffffff80ea6a0f at vm_fault+0x15af
#4 0xffffffff80ea5381 at vm_fault_trap+0x81
#5 0xffffffff81000ade at trap_pfault+0x1be
#6 0xffffffff80fd6d48 at calltrap+0x8
#7 0xffffffff8425c200 at amdgpu_vm_pt_create+0x3b0
#8 0xffffffff84259148 at amdgpu_vm_bo_update_mapping+0x568
#9 0xffffffff84259da2 at amdgpu_vm_bo_update+0x2c2
#10 0xffffffff84225dcf at amdgpu_gem_va_update_vm+0x4f
#11 0xffffffff84225d33 at amdgpu_gem_va_ioctl+0x253
#12 0xffffffff84059cc6 at drm_ioctl_kernel+0xc6
#13 0xffffffff8405a075 at drm_ioctl+0x2b5
#14 0xffffffff80d9a44d at linux_file_ioctl+0x2fd
#15 0xffffffff80ba0f95 at kern_ioctl+0x255
#16 0xffffffff80ba0cdf at sys_ioctl+0xff
#17 0xffffffff810011c0 at amd64_syscall+0x100
Uptime: 1m15s
Dumping 1431 out of 32635 MB:..2%..11%..21%..31%..41%..51%..61%..71%..81%..91%

Reading symbols from /boot/kernel/cryptodev.ko...
Reading symbols from /usr/lib/debug//boot/kernel/cryptodev.ko.debug...
Reading symbols from /boot/kernel/nmdm.ko...
Reading symbols from /usr/lib/debug//boot/kernel/nmdm.ko.debug...
Reading symbols from /boot/kernel/if_bridge.ko...
Reading symbols from /usr/lib/debug//boot/kernel/if_bridge.ko.debug...
Reading symbols from /boot/kernel/bridgestp.ko...
Reading symbols from /usr/lib/debug//boot/kernel/bridgestp.ko.debug...
Reading symbols from /boot/kernel/amdtemp.ko...
Reading symbols from /usr/lib/debug//boot/kernel/amdtemp.ko.debug...
Reading symbols from /boot/kernel/amdsmn.ko...
Reading symbols from /usr/lib/debug//boot/kernel/amdsmn.ko.debug...
Reading symbols from /boot/kernel/zfs.ko...
Reading symbols from /usr/lib/debug//boot/kernel/zfs.ko.debug...
Reading symbols from /boot/kernel/vmm.ko...
Reading symbols from /usr/lib/debug//boot/kernel/vmm.ko.debug...
Reading symbols from /boot/modules/amdgpu.ko...
(No debugging symbols found in /boot/modules/amdgpu.ko)
Reading symbols from /boot/modules/drm.ko...
(No debugging symbols found in /boot/modules/drm.ko)
Reading symbols from /boot/kernel/iic.ko...
Reading symbols from /usr/lib/debug//boot/kernel/iic.ko.debug...
Reading symbols from /boot/kernel/linuxkpi_hdmi.ko...
[... 49 lines cut ...]
Reading symbols from /usr/lib/debug//boot/kernel/ums.ko.debug...
Reading symbols from /boot/kernel/usbhid.ko...
Reading symbols from /usr/lib/debug//boot/kernel/usbhid.ko.debug...
Reading symbols from /boot/kernel/hms.ko...
Reading symbols from /usr/lib/debug//boot/kernel/hms.ko.debug...
Reading symbols from /boot/kernel/hidmap.ko...
Reading symbols from /usr/lib/debug//boot/kernel/hidmap.ko.debug...
Reading symbols from /boot/kernel/pflog.ko...
Reading symbols from /usr/lib/debug//boot/kernel/pflog.ko.debug...
Reading symbols from /boot/kernel/pf.ko...
Reading symbols from /usr/lib/debug//boot/kernel/pf.ko.debug...
Reading symbols from /boot/kernel/fdescfs.ko...
Reading symbols from /usr/lib/debug//boot/kernel/fdescfs.ko.debug...
__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:57
57		__asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
````

Some additional info: -

1. Disabled linux emulation
2. Removed the proc entry in /etc/fstab

As I saw Linux in the trace, but it didn't any difference.

What has changed in FreeBSD 14.1 which is causing this?

knaggsy2000, 2024-09-18:
Have been a longtime FreeBSD user (since v5), not had to deal with crash dumps so much before.  So throw me bone if needed.

ClausAndersen, 2024-09-20:
I am not sure this is a "bone" but.... To me it looks like your might not have `/usr/src` populated correctly (The FreeBSD kernel source). This is what [FloppyKing](https://forums.FreeBSD.org/threads/amd-gpu-driver-has-a-serious-bug-introduced-in-14-1.94338/post-671127) hinted at in the thread you referenced. Credit to T-Daemon as I got it from [here.](https://forums.FreeBSD.org/threads/intermittent-scanline-flickering-with-amdgpu-in-both-vt-and-x.93159/post-662354) 

The following worked for me:

```
cd /usr
git clone
git clone https://git.freebsd.org/src.git
git checkout releng/14.1
cd /usr/ports/graphics/drm-61-kmod
make
make install
```

knaggsy2000, 2024-09-20:
@ClausAndersen "Throw me a bone" is an expression that equates to something like "help me out" or "give me a hint".

Used "freebsd-update" to update the src and system bins before, have been through quite a few major OS versions so far.

Now on p5 version of kernel as of today.

Maybe the src folder needs re-populating as you mentioned (as I've said, done multiple major OS upgrades), let me wipe it and follow your instructions to re-populate it.

Watch this space...

knaggsy2000, 2024-09-20:
@ClausAndersen

Updated src folder but not as you mentioned (the first "git clone" command just dumped the usage): -

````
# cd /usr
# git clone https://git.freebsd.org/src.git
Cloning into 'src'...
remote: Enumerating objects: 4549076, done.
remote: Counting objects: 100% (383654/383654), done.
remote: Compressing objects: 100% (29008/29008), done.
remote: Total 4549076 (delta 377922), reused 354646 (delta 354646), pack-reused 4165422 (from 1)
Receiving objects: 100% (4549076/4549076), 1.66 GiB | 29.16 MiB/s, done.
Resolving deltas: 100% (3616018/3616018), done.
Updating files: 100% (101258/101258), done.
# git checkout releng/14.1
Updating files: 100% (27338/27338), done.
branch 'releng/14.1' set up to track 'origin/releng/14.1'.
Switched to a new branch 'releng/14.1'
# ll /usr/src
total 598
-rw-r--r--    1 root wheel    148 Sep 20 23:31 .arcconfig
-rw-r--r--    1 root wheel    448 Sep 20 23:31 .arclint
drwxr-xr-x    2 root wheel      3 Sep 20 23:31 .cirrus-ci/
-rw-r--r--    1 root wheel   5462 Sep 20 23:31 .cirrus.yml
-rw-r--r--    1 root wheel   4827 Sep 20 23:31 .clang-format
drwxr-xr-x    8 root wheel     13 Sep 20 23:31 .git/
-rw-r--r--    1 root wheel    991 Sep 20 23:31 .git-blame-ignore-revs
-rw-r--r--    1 root wheel    205 Sep 20 23:31 .gitattributes
drwxr-xr-x    4 root wheel      5 Sep 20 23:31 .github/
-rw-r--r--    1 root wheel    297 Sep 20 23:31 .gitignore
-rw-r--r--    1 root wheel   7903 Sep 20 23:31 CONTRIBUTING.md
-rw-r--r--    1 root wheel   6109 Sep 20 23:31 COPYRIGHT
-rw-r--r--    1 root wheel    490 Sep 20 23:31 LOCKS
-rw-r--r--    1 root wheel   7175 Sep 20 23:31 MAINTAINERS
-rw-r--r--    1 root wheel  31522 Sep 20 23:31 Makefile
-rw-r--r--    1 root wheel 131226 Sep 20 23:31 Makefile.inc1
-rw-r--r--    1 root wheel   3621 Sep 20 23:31 Makefile.libcompat
-rw-r--r--    1 root wheel   1939 Sep 20 23:31 Makefile.sys.inc
-rw-r--r--    1 root wheel 931131 Sep 20 23:31 ObsoleteFiles.inc
-rw-r--r--    1 root wheel   2837 Sep 20 23:31 README.md
-rw-r--r--    1 root wheel  12396 Sep 20 23:31 RELNOTES
-rw-r--r--    1 root wheel  88839 Sep 20 23:31 UPDATING
drwxr-xr-x   44 root wheel     46 Sep 20 23:31 bin/
drwxr-xr-x   11 root wheel     13 Sep 20 23:31 cddl/
drwxr-xr-x   90 root wheel     90 Sep 20 23:31 contrib/
drwxr-xr-x    5 root wheel      6 Sep 20 23:31 crypto/
drwxr-xr-x    8 root wheel     13 Sep 20 23:31 etc/
drwxr-xr-x    5 root wheel      9 Sep 20 23:31 gnu/
drwxr-xr-x   10 root wheel    112 Sep 20 23:31 include/
drwxr-xr-x    8 root wheel     11 Sep 20 23:31 kerberos5/
drwxr-xr-x  146 root wheel    148 Sep 20 23:31 lib/
drwxr-xr-x   39 root wheel     41 Sep 20 23:31 libexec/
drwxr-xr-x   12 root wheel     24 Sep 20 23:31 release/
drwxr-xr-x    4 root wheel      6 Sep 20 23:31 rescue/
drwxr-xr-x   86 root wheel     92 Sep 20 23:31 sbin/
drwxr-xr-x    8 root wheel     11 Sep 20 23:31 secure/
drwxr-xr-x   31 root wheel     33 Sep 20 23:31 share/
drwxr-xr-x   26 root wheel     35 Sep 20 23:31 stand/
drwxr-xr-x   53 root wheel     55 Sep 20 23:31 sys/
drwxr-xr-x    3 root wheel      6 Sep 20 23:31 targets/
drwxr-xr-x    8 root wheel     15 Sep 20 23:31 tests/
drwxr-xr-x   19 root wheel     23 Sep 20 23:31 tools/
drwxr-xr-x  278 root wheel    284 Sep 20 23:31 usr.bin/
drwxr-xr-x  227 root wheel    235 Sep 20 23:31 usr.sbin/
````

Let me try re-building drm-61 now...

knaggsy2000, 2024-09-20:
@ClausAndersen Yes, can confirm drm-61 now compiles from ports with my clean src folder.

Will reboot later on and first test if still works my my 6500XT, if that works I will retry the 550.

Watch this space.

knaggsy2000, 2024-09-20:
@ClausAndersen Speaking of the ports (this is a little off-subject) - but I update ports via "portsnap" and now that is saying it's going be EoL'ed soon.  So what is the official method of updating the ports?

ClausAndersen, 2024-09-21:
If you look at the compiler error you will see:

    /usr/src/sys/compat/linuxkpi/common/include/linux/slab.h:39:10: fatal error: 'linux/llist.h' file not found
       39 | #include <linux/llist.h>

That means that `/usr/src/sys/compat/linuxkpi/common/include/linux/llist.h` simply is not there....

The most likely culprit is that it was not installed originally using `bsdinstall`. If it was then it *should* be upgraded using `freebsd-update`. The update process tells you what installed system components are ready to update and `src` should be mentioned here.

I normally do not rebuild FreeBSD myself so I usually never install this on my system. And it is very few ports which depends on this so you can normally live a happy life without.

That would be the common cause. In your case my guess is that you installed `src` later and then simply did not track the changes. `freebsd-update` will not do that if it was not installed with `bsdinstall`.

The official method for both src and ports is using git. They are as usual documented in the Handbook.

You now have an unholy mix of binary packages and ports (as do I) which as you might know is not recommended. If `drm-61` works for you then consider doing a `pkg lock drm-61`. If not it will be rolled back on your next `pkg upgrade`. Should that happen it is an easy fix: Just got back and do a `make install` after the upgrade.

Links:

[26.6.3. Updating the Source](https://docs.freebsd.org/en/books/handbook/cutting-edge/#updating-src-obtaining-src)

[4.4.12. Locking and Unlocking Packages](https://docs.freebsd.org/en/books/handbook/ports/#pkg-locking-unlocking)

[4.5.1. Installing the Ports Collection](https://docs.freebsd.org/en/books/handbook/ports/#ports-using-installation-methods)

[portsnap being retired - what's the alternative?](https://forums.freebsd.org/threads/portsnap-being-retired-whats-the-alternative.76463/)

knaggsy2000, 2024-09-22:
@ClausAndersen I always install the src when installing from ISO.  So "freebsd-update" didn't update my src folders correctly for some reason.

Oh, I've been thought many upgrades before.  Including one where I would only install packages by building from src (ports).

I have locked the package for drm-61 - with command ````pkg lock drm-61-kmod```` - you forgot the "kmod" part.

Do you believe that my RX550 will work now?

And see that using GIT is now the way to update both the ports and src.  But I've noticed some about a new package called "pkgbase" - do you know about this?

ClausAndersen, 2024-09-23:
`pkgbase` is the effort to distribute the base operating system as regular `pkg` packages. It has been brewing for a long time. Originally intended for 12. I see planning documents[1] which state it as a "need" for 15. But if that will materialize is unknown to me.

This will probably be a transistion where we can value we have ZFS snapshots as well. 😄 

You might enjoy the video from EuroBSDcon 2019 as well[2].

[1] [DevSummit planning document for 15](https://github.com/bsdjhb/devsummit/blob/main/15.0/planning.md)
[2] [Are we there yet ????](https://papers.freebsd.org/2019/eurobsdcon/vadot-pkgbase_are_we_there_yet/)

ClausAndersen, 2024-09-23:
I forgot the main page: https://wiki.freebsd.org/PkgBase

knaggsy2000, 2024-09-23:
@ClausAndersen I found that main page quite a while ago.  Just don't understand why the handbook hasn't been updated to say that GIT is the way to update src and ports.  As the message from "pkg update" just states it's going to die soon, yet doesn't mention any alternatives.

knaggsy2000, 2024-09-23:
@ClausAndersen But on a more positive note - the drm-61-kmod firmware is STILL working fine on my system, that is with my backup gfx card which is a RX6500XT which is completely overkill for this PC.  Hence, why I want to put back my RX550.

Do you think things will work now?  As I've had another user report that a RX580 is producing the same issues.  Details to come.

knaggsy2000, 2024-09-23:
@yukiteruamano Please report here.

knaggsy2000, 2024-09-23:
At this time, to update the ports: -

`````
# pkg install git
# git clone --depth 1 https://git.FreeBSD.org/ports.git /usr/ports
````

Then from now on to update them: -

````
git -C /usr/ports pull
````

knaggsy2000, 2024-09-23:
To deal with /usr/src - wipe that folder completely, before starting: -

````
cd /usr
git clone https://git.freebsd.org/src.git
git checkout releng/14.1
````

After that, to update them: -

````
git -C /usr/src pull
````

knaggsy2000, 2024-09-23:
````
git clone https://git.freebsd.org/src.git /usr/src
````

Would be a better command.

knaggsy2000, 2024-09-24:
Tried the RX550 again, but now not even getting a console - switched back the RX6550XT.  Maybe this is an BIOS issue?

yukiteruamano, 2024-09-24:
Right now building with poudriere, drm-61-kmod build successful on FreeBSD 14.1-p5

yukiteruamano, 2024-09-24:
> @yukiteruamano Please report here.

Build, all ok, with drm-61-kmod I don't have problems. X run fine, mpv with VAAPI and GLSL shaders using Vulkan and OpenGL context work fine.

knaggsy2000, 2024-09-26:
> Tried the RX550 again, but now not even getting a console - switched back the RX6550XT. Maybe this is an BIOS issue?

Think my RX550 might have died.  As can't even POST with it.  But there is another BIOS for my motherboard which I have now updated too despite it still being "beta" - well I need that to support my CPU.  My X470 motherboard, originally had an 2700X CPU installed but with the beta BIOS I was able to upgrade to a 5800X and gained at least 60% in performance (with my own internal tests).

I will have to re-test with the new BIOS - watch this space.

knaggsy2000, 2024-09-26:
> > @yukiteruamano Please report here.
> 
> Build, all ok, with drm-61-kmod I don't have problems. X run fine, mpv with VAAPI and GLSL shaders using Vulkan and OpenGL context work fine.

So everything mentioned here has resolved YOUR issue?

knaggsy2000, 2024-09-26:
@ClausAndersen @yukiteruamano Did the 61 version resolve your issues?

yukiteruamano, 2024-09-26:
> @ClausAndersen @yukiteruamano Did the 61 version resolve your issues?

Yes, 61 without problems

knaggsy2000, 2024-09-26:
@yukiteruamano That is fantastic - glad my post helped you resolve your issue ^^.

yukiteruamano, 2024-09-26:
> @yukiteruamano That is fantastic - glad my post helped you resolve your issue ^^.

Yeah, 510 and 61-kmod work fine, but 515 is a disaster with RX580

knaggsy2000, 2024-09-26:
@yukiteruamano But you're running with 61 at this point in time?

knaggsy2000, 2024-09-26:
As I will retry the gfx card again very soon, as I don't want such an expensive card in use for my main PC - it is completely OVERKILL.

Ice-IX, 2024-09-28:
I threw my HD 7850 (pitcairn) into one of my machines to test since that it supposed to work with both radeonkms and amdgpu and I got a kernel panic loading either of them, with both drm-510-kmod installed from pkg and drm-61-kmod compiled from ports.
I also tried to boot from a debian liveusb and got a kernel panic as well.  Thinking at this point that my card had just failed in storage, I booted windows and that worked fine...

knaggsy2000, 2024-09-29:
@Ice-IX That's quite a different generation of card compared to mine.  Codename "Pitcairn Pro" (released 2012) vs "Polaris 12" (released 2017) of mine.  Maybe raise a new issue on here for that?  As that also needs to be noted and resolved.

However, you getting kernel panics with MULTIPLE OS's generally (in my experiences at least) typically is a hardware fault.  I bet Windows just used a generic driver and nothing actually accelerated, hence why it appeared to work at first.  Did you try to run something like [FurMark](https://geeks3d.com/furmark/) in Windows?

knaggsy2000, 2024-09-29:
I've just re-installed my RX550 and my PC now POSTs with it (after the BIOS upgrade - yet it was fine before that, but anyway).  It appears from `dmesg`: -

````
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] initializing kernel modesetting (POLARIS12 0x1002:0x699F 0x1002:0xBD20 0xC7).
[drm] register mmio base: 0xFCF00000
[drm] register mmio size: 262144
[drm] add ip block number 0 <vi_common>
[drm] add ip block number 1 <gmc_v8_0>
[drm] add ip block number 2 <tonga_ih>
[drm] add ip block number 3 <gfx_v8_0>
[drm] add ip block number 4 <sdma_v3_0>
[drm] add ip block number 5 <powerplay>
[drm] add ip block number 6 <dm>
[drm] add ip block number 7 <uvd_v6_0>
[drm] add ip block number 8 <vce_v3_0>
drmn0: Fetched VBIOS from ROM BAR
amdgpu: ATOM BIOS: HAOCHEN.001
[drm] UVD is enabled in VM mode
[drm] UVD ENC is enabled in VM mode
[drm] VCE enabled in VM mode
drmn0: Trusted Memory Zone (TMZ) feature not supported
drmn0: PCIE atomic ops is not supported
[drm] vm size is 128 GB, 2 levels, block size is 10-bit, fragment size is 9-bit
drmn0: successfully loaded firmware image 'amdgpu/polaris12_mc.bin'
drmn0: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
drmn0: GART: 256M 0x000000FF00000000 - 0x000000FF0FFFFFFF
[drm] Detected VRAM RAM=4096M, BAR=256M
[drm] RAM width 128bits GDDR5
[drm] amdgpu: 4096M of VRAM memory ready
[drm] amdgpu: 16317M of GTT memory ready.
[drm] GART: num cpu pages 65536, num gpu pages 65536
[drm] PCIE GART of 256M enabled (table at 0x000000F400900000).
drmn0: successfully loaded firmware image 'amdgpu/polaris12_pfp_2.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_me_2.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_ce_2.bin'
[drm] Chained IB support enabled!
drmn0: successfully loaded firmware image 'amdgpu/polaris12_rlc.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_mec_2.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_mec2_2.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_sdma.bin'
drmn0: successfully loaded firmware image 'amdgpu/polaris12_sdma1.bin'
amdgpu: hwmgr_sw_init smu backed is polaris10_smu
drmn0: successfully loaded firmware image 'amdgpu/polaris12_uvd.bin'
[drm] Found UVD firmware Version: 1.130 Family ID: 16
drmn0: successfully loaded firmware image 'amdgpu/polaris12_vce.bin'
[drm] Found VCE firmware Version: 53.26 Binary ID: 3
drmn0: successfully loaded firmware image 'amdgpu/polaris12_smc.bin'
[drm] Display Core initialized with v3.2.207!
lkpi_iic0: <LinuxKPI I2C> on drmn0
iicbus0: <Philips I2C bus> on lkpi_iic0
iic0: <I2C generic I/O> on iicbus0
lkpi_iic1: <LinuxKPI I2C> on drmn0
iicbus1: <Philips I2C bus> on lkpi_iic1
iic1: <I2C generic I/O> on iicbus1
lkpi_iic2: <LinuxKPI I2C> on drmn0
iicbus2: <Philips I2C bus> on lkpi_iic2
iic2: <I2C generic I/O> on iicbus2
[drm] UVD and UVD ENC initialized successfully.
[drm] VCE initialized successfully.
drmn0: SE 2, SH per SE 1, CU per SH 5, active_cu_number 8
````

The firmware now gets loaded successfully.  Let me test more over the next day or so - to ensure that the 61 version resolved my issue.

knaggsy2000, 2024-09-29:
Do have hardware acceleration as tried my time-tested game - [Neverputt](https://www.freshports.org/games/neverball/).  Haha.

yukiteruamano, 2024-09-30:
> @yukiteruamano But you're running with 61 at this point in time?

Yes I'm running 61 actually

knaggsy2000, 2024-10-03:
Upon re-sync'ing my /usr/src folder and ports /usr/ports - was able to compile the 61 kmod.

Now after a several days of testing - the issue I originally reported is now RESOLVED.  So I will close this issue shortly.

@Ice-IX Please raise your issue on here (not in this post, but as another issue) as my original issue is now resolved.

@yukiteruamano Remember to lock those packages too.  My RX550 is working great now.  But also know that my backup RX6500XT also now works - so bonus.

knaggsy2000, 2024-10-03:
To update your ports (ensure folder is empty to start with): -

````
# pkg install git
# git clone --depth 1 https://git.FreeBSD.org/ports.git /usr/ports
````

Then after that to update it: -

````
# git -C /usr/ports pull
````

knaggsy2000, 2024-10-03:
To update your src (again, ensure folder is empty) and this for 14.1: -

````
# git clone https://git.freebsd.org/src.git /usr/src
# git checkout releng/14.1
````

Then after that to update it: -

````
# git -C /usr/src pull
````

knaggsy2000, 2024-10-03:
Both of these points NEED to be updated in the handbook.

knaggsy2000, 2024-10-03:
Closing issue as now resolved by 61-kmod being built from src.

closed 2024-10-03 by knaggsy2000