freebsd/drm-kmod #319: i915kms from drm-61-kmod causes kernel panic with A380 card

oyvinht, 2024-10-01:
**Describe the bug**
Kernel panic when loading i915kms, as shown in image below.

**FreeBSD version**
FreeBSD beta-orionis 14.1-RELEASE-p3 FreeBSD 14.1-RELEASE-p3 GENERIC amd64 1401000 1401000

**PCI Info**
<details>
  <summary>
</summary>
  hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14d8 subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14d9 subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14db subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14db subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
pcib13@pci0:0:2:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14db subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14da subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = HOST-PCI
pcib14@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14dd subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
pcib15@pci0:0:8:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x14dd subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = bridge
    subclass   = PCI-PCI
none1@pci0:0:20:0:	class=0x0c0500 rev=0x71 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb6@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x14e0 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
[... 126 lines cut ...]
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = serial bus
    subclass   = USB
xhci3@pci0:15:0:0:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15b8 subvendor=0x1043 subdevice=0x8877
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    class      = serial bus
    subclass   = USB
</details>

**DRM KMOD version**
drm-61-kmod 6.1.92

**To Reproduce**
Install an Intel A380 card. The one used here is an ASRock Intel Arc A380 Low Profile.

Then do either:

sudo kldload i915kms

or start Xorg with the "intel" driver.

**Screenshots**
[image]

**Additional context**

lin72h, 2024-10-02:
I think you need newly merged 6.6-drm-kmod  for intel discrete GPU

oyvinht, 2024-10-11:
Oh, right, thanks. That requires 15-CURRENT to compile, I guess.

oyvinht, 2024-10-29:
For what it's worth, I complied drm-kmod on 14-stable, and did some rudimentary debugging.

For now, I discovered that I could at least replace the normal UEFI framebuffer and switch the console to 3840 x 2160 by commenting out from drivers/gpu/drm/i915/i915_gem.c:

```
@@ -1192,12 +1192,12 @@ int i915_gem_init(struct drm_i915_private *dev_priv)
         * FIXME: break up the workarounds and apply them at the right time!
         */
        intel_clock_gating_init(dev_priv);
-
+        /*
        for_each_gt(gt, dev_priv, i) {
                ret = intel_gt_init(gt);
                if (ret)
                        goto err_unlock;
-       }
+       }*/
```

and add a return before setting `ppgtt->vm.insert_page = gen8_ppgtt_insert_entry;` in drivers/gpu/drm/i915/gt/gen8_ppgtt.c.

I'll see if I can find the time to debug a little deeper.

wulf7, 2024-12-25:
You may try patch from https://github.com/freebsd/drm-kmod/issues/315 as it seems to be the same issue.
Unfortunatelly it didn't help.