freebsd/drm-kmod #320: Feature Request: hw.dri.edid_firmware (drm_edid_load.c)

bileslav, 2024-10-23:
One of the things I missed after switching from Linux to FreeBSD was the ability to override the EDID. I used this as a way to force the system to run at the highest frame rate my monitor supports, starting from the earliest possible occasion. I take my monitor's original EDID and use wxedid to swap the 60 and 165 Hz modelines. This makes the 165 Hz modeline the default one.

This has two major benefits, applicable to both X and Wayland:

- Only one black screen from the OS startup to the graphical session.
- Instant switching between the graphical session and virtual terminals.

I just got this working on FreeBSD, you can see the commits [here](https://github.com/freebsd/drm-kmod/compare/6.1-lts...bileslav:drm-kmod:6.1-lts). It took a minor fix to `drm_edid_load.c` to get it to compile. This turned out to be quite simple, although I had no knowledge in this area, so something was probably missed, but the goal was achieved. The only thing I didn't figure out was how to actually load my modified EDID from the file system (we don't seem to have anything like `/lib/firmware` here, and the firmware files are loaded from `/boot/modules` in a way that isn't obvious to me), so I just hard-coded the output of `hexdump`.

```
$ sysctl -a | grep G2412
<6>[drm] Got built-in EDID base block and 1 extension from "edid/G2412-165Hz.bin" for connector "DP-1"
hw.dri.edid_firmware: edid/G2412-165Hz.bin
compat.linuxkpi.drm_edid_firmware: edid/G2412-165Hz.bin
```

evadot, 2024-10-23:
Hi, this would be a nice feature to have but unfortunatelly drm_edid_load.c is gpl-2.0 licenced so we would need to rewrite it before including it into this repo.

bileslav, 2024-10-23:
Whoa. That's news to me. Even being more careful, I wouldn't expect the license of one of the files is different from the others. First lesson!

OK, I'd be happy to see this feature one day.

thesunexpress, 2024-12-26:
>  I wouldn't expect the license of one of the files is different from the others. First lesson!
> 

The mysterious souls in the Linux world... 

There's a few more of those in the FreeBSD ports collection, ffmpeg comes to mind *I think*, where we only use the non-GPL bits, unless agreed upon & accepted at compile time.