freebsd/drm-kmod #323: drm-61-kmod: Firmware fails to load a few times until its loads correctly

Alexander88207, 2024-11-10:
**Describe the bug**
```
navy_flounder_sos.bin: could not load binary firmware /boot/firmware/navy_flounder_sos.bin either
amdgpu/navy_flounder_sos.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_sos.bin either
amdgpu_navy_flounder_sos.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_sos.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_sos.bin'
navy_flounder_ta.bin: could not load binary firmware /boot/firmware/navy_flounder_ta.bin either
amdgpu/navy_flounder_ta.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_ta.bin either
amdgpu_navy_flounder_ta.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_ta.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_ta.bin'
drmn0: PSP runtime database doesn't exist
drmn0: PSP runtime database doesn't exist
navy_flounder_smc.bin: could not load binary firmware /boot/firmware/navy_flounder_smc.bin either
amdgpu/navy_flounder_smc.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_smc.bin either
amdgpu_navy_flounder_smc.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_smc.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_smc.bin'
drmn0: STB initialized to 2048 entrieslkpi_iic0: <LinuxKPI I2C> on drmn0
iicbus0: <Philips I2C bus> on lkpi_iic0
iic0: <I2C generic I/O> on iicbus0
lkpi_iic1: <LinuxKPI I2C> on drmn0
iicbus1: <Philips I2C bus> on lkpi_iic1
iic1: <I2C generic I/O> on iicbus1
navy_flounder_dmcub.bin: could not load binary firmware /boot/firmware/navy_flounder_dmcub.bin either
amdgpu/navy_flounder_dmcub.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_dmcub.bin either
amdgpu_navy_flounder_dmcub.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_dmcub.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_dmcub.bin'
[drm] Loading DMUB firmware via PSP: version=0x0202001E
navy_flounder_pfp.bin: could not load binary firmware /boot/firmware/navy_flounder_pfp.bin either
amdgpu/navy_flounder_pfp.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_pfp.bin either
amdgpu_navy_flounder_pfp.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_pfp.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_pfp.bin'
navy_flounder_me.bin: could not load binary firmware /boot/firmware/navy_flounder_me.bin either
amdgpu/navy_flounder_me.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_me.bin either
amdgpu_navy_flounder_me.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_me.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_me.bin'
navy_flounder_ce.bin: could not load binary firmware /boot/firmware/navy_flounder_ce.bin either
amdgpu/navy_flounder_ce.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_ce.bin either
amdgpu_navy_flounder_ce.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_ce.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_ce.bin'
navy_flounder_rlc.bin: could not load binary firmware /boot/firmware/navy_flounder_rlc.bin either
amdgpu/navy_flounder_rlc.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_rlc.bin either
amdgpu_navy_flounder_rlc.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_rlc.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_rlc.bin'
navy_flounder_mec.bin: could not load binary firmware /boot/firmware/navy_flounder_mec.bin either
amdgpu/navy_flounder_mec.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_mec.bin either
amdgpu_navy_flounder_mec.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_mec.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_mec.bin'
navy_flounder_mec2.bin: could not load binary firmware /boot/firmware/navy_flounder_mec2.bin either
amdgpu/navy_flounder_mec2.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_mec2.bin either
amdgpu_navy_flounder_mec2.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_mec2.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_mec2.bin'
navy_flounder_sdma.bin: could not load binary firmware /boot/firmware/navy_flounder_sdma.bin either
amdgpu/navy_flounder_sdma.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_sdma.bin either
amdgpu_navy_flounder_sdma.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_sdma.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_sdma.bin'
[drm] use_doorbell being set to: [true]
[drm] use_doorbell being set to: [true]
navy_flounder_vcn.bin: could not load binary firmware /boot/firmware/navy_flounder_vcn.bin either
amdgpu/navy_flounder_vcn.bin: could not load binary firmware /boot/firmware/amdgpu/navy_flounder_vcn.bin either
amdgpu_navy_flounder_vcn.bin: could not load binary firmware /boot/firmware/amdgpu_navy_flounder_vcn.bin either
drmn0: successfully loaded firmware image 'amdgpu/navy_flounder_vcn.bin'
```

**FreeBSD version**
FreeBSD Home 14.2-STABLE FreeBSD 14.2-STABLE stable/14-da80e62f9 GENERIC amd64 1402500 1402500

**PCI Info**
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x05 hdr=0x00 vendor=0x8086 device=0x9b53 subvendor=0x1043 subdevice=0x8694
[... 89 lines cut ...]
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTL8125 2.5GbE Controller'
    class      = network
    subclass   = ethernet
```

</details>

**DRM KMOD version**

```
pkg info drm-61-kmod
drm-61-kmod-6.1.92_2
Name           : drm-61-kmod
Version        : 6.1.92_2
Installed on   : Sat Nov  9 21:09:26 2024 CET
Origin         : graphics/drm-61-kmod
Architecture   : FreeBSD:14:amd64
```

**To Reproduce**
Load the driver.

**Additional context**
Everything works fine didn't notice it otherwise just saw it while booting up.

wulf7, 2024-11-10:
It is an expected behavior. drm-kmod tries to load firmware as a file than fallbacks to load it as kmod with success.

Should be closed as NOT-A-BUG

P.S.: The messages are really confusing

Alexander88207, 2024-11-10:
Alright, closing this.

Thanks for your quick answer.

wulf7, 2024-11-10:
Reopen NOT-A-BUG as I expect more bug reports like this one. The things are really very confusing.

evadot, 2024-11-10:
Both main and stable/14 (so 14.2 too) can load binary files now, I plan to add support for this in the near future (I just need to adapt the port for 13 as it doesn't have this feature and never will).
We can leave this bug open for now.

thesunexpress, 2024-12-26:
Also NOT-A-BUG, it happens with i915-kmod as well, using 6.1-lts.
Seems the routine is to verify various GPU-specific firmware bits are not yet loaded, reporting "could not load binary firmware" (which is confusing...), followed by immediately loading the correct firmware. Happens with 14.2-STABLE & 15-CURRENT.