freebsd/drm-kmod #325: Is drm-66-kmod port planned ?

as400l, 2024-11-19:
I'd like to ask whether drm-66-kmod port is planned for 14.1/14.2 ? I'm on meteor lake so 6.1 does not allow me to use my gpu.

Alternatively - is it possible to compile drm-v6.6 on 14.1 ?

Appreciate any help.

kenrap, 2024-11-20:
If you're willing to use CURRENT until it reaches `stable/14`, you can use my "ghetto" version:

```diff
diff --git a/graphics/drm-66-kmod/Makefile b/graphics/drm-66-kmod/Makefile
new file mode 100644
index 000000000000..154624a2b06c
--- /dev/null
+++ b/graphics/drm-66-kmod/Makefile
@@ -0,0 +1,55 @@
+PORTNAME=	drm-66-kmod
+PORTVERSION=	6.6
+CATEGORIES=	graphics kld
+
+MAINTAINER=	x11@FreeBSD.org
+COMMENT=	DRM drivers modules
+WWW=		https://github.com/freebsd/drm-kmod/
+
+LICENSE=	BSD2CLAUSE MIT GPLv2
+LICENSE_COMB=	multi
+
+ONLY_FOR_ARCHS=	amd64 powerpc64 powerpc64le
+ONLY_FOR_ARCHS_REASON=	the new KMS components are only supported on amd64 and powerpc64
+
+CONFLICTS_INSTALL=	drm-510-kmod \
+			drm-515-kmod \
+			drm-61-kmod
+
+USES=		kmod uidfix compiler:c++11-lang
+
+USE_GITHUB=	yes
+GH_ACCOUNT=	freebsd
+GH_PROJECT=	drm-kmod
+GH_TAGNAME=	d077996bd442e9410de95aa9c19e55b7c19a54d8
+
+.include <bsd.port.options.mk>
+
+.if ${OPSYS} == FreeBSD && !( ${OSVERSION} >= 1500008 || ( ${OSVERSION} >= 1400508 && ${OSVERSION} < 1500000 ))
+IGNORE=		not supported on older than 14-STABLE 1400508, no kernel support
+.endif
+.if ${OPSYS} != FreeBSD
+IGNORE=		not supported on anything but FreeBSD (missing linuxkpi functionality)
+.endif
+
+.if ${ARCH} == "amd64"
+PLIST_SUB+=	AMDGPU=""
+PLIST_SUB+=	I915=""
+.elif ${ARCH} == "i386"
+PLIST_SUB+=	AMDGPU="@comment "
+PLIST_SUB+=	I915=""
+.elif ${ARCH} == "aarch64" || ${ARCH:Mpowerpc*}
+PLIST_SUB+=	AMDGPU=""
+PLIST_SUB+=	I915="@comment "
+.else
+PLIST_SUB+=	AMDGPU="@comment "
+PLIST_SUB+=	I915="@comment "
+.endif
+
+MAKE_ENV+=	MAKEOBJDIRPREFIX=${WRKSRC}/obj
+
+pre-build:
+		${MKDIR} ${WRKSRC}/obj
+		(cd ${WRKSRC} ; ${SETENV} ${MAKE_ENV} ${MAKE_CMD} obj)
+
+.include <bsd.port.mk>
diff --git a/graphics/drm-66-kmod/distinfo b/graphics/drm-66-kmod/distinfo
new file mode 100644
index 000000000000..6f1e46906318
--- /dev/null
+++ b/graphics/drm-66-kmod/distinfo
@@ -0,0 +1,3 @@
+TIMESTAMP = 1731531695
+SHA256 (freebsd-drm-kmod-6.6-d077996bd442e9410de95aa9c19e55b7c19a54d8_GH0.tar.gz) = b741eca36e3128dc6fd5d97dc10e011d4794e1fb8c65041896d326f5fbf686f6
+SIZE (freebsd-drm-kmod-6.6-d077996bd442e9410de95aa9c19e55b7c19a54d8_GH0.tar.gz) = 38487339
diff --git a/graphics/drm-66-kmod/pkg-descr b/graphics/drm-66-kmod/pkg-descr
new file mode 100644
index 000000000000..a2e7a2515a87
--- /dev/null
+++ b/graphics/drm-66-kmod/pkg-descr
@@ -0,0 +1,4 @@
+amdgpu, i915, and radeon DRM drivers modules.
+Currently corresponding to Linux 6.1 DRM.
+This version is for FreeBSD 14-STABLE 1400508
+and above.
diff --git a/graphics/drm-66-kmod/pkg-message b/graphics/drm-66-kmod/pkg-message
new file mode 100644
index 000000000000..995379cd0d7d
--- /dev/null
+++ b/graphics/drm-66-kmod/pkg-message
@@ -0,0 +1,18 @@
+[
+{ type: install
+  message: <<EOM
+The drm-61-kmod port can be enabled for amdgpu (for AMD
+GPUs starting with the HD7000 series / Tahiti) or i915kms (for Intel
+APUs starting with HD3000 / Sandy Bridge) through kld_list in
+/etc/rc.conf. radeonkms for older AMD GPUs can be loaded and there are
+some positive reports if EFI boot is NOT enabled.
+
+For amdgpu: kld_list="amdgpu"
+For Intel: kld_list="i915kms"
+For radeonkms: kld_list="radeonkms"
+
+Please ensure that all users requiring graphics are members of the
+"video" group.
+EOM
+}
+]
diff --git a/graphics/drm-66-kmod/pkg-plist b/graphics/drm-66-kmod/pkg-plist
new file mode 100644
index 000000000000..42f54df26cff
--- /dev/null
+++ b/graphics/drm-66-kmod/pkg-plist
@@ -0,0 +1,6 @@
+/%%KMODDIR%%/dmabuf.ko
+%%AMDGPU%%/%%KMODDIR%%/amdgpu.ko
+/%%KMODDIR%%/drm.ko
+%%I915%%/%%KMODDIR%%/i915kms.ko
+/%%KMODDIR%%/radeonkms.ko
+/%%KMODDIR%%/ttm.ko

```

as400l, 2024-11-20:
"ghetto" - you made my Wednesday :)

Thanks

Closing.

closed 2024-11-20 by as400l