freebsd/drm-kmod #337: Cannot build drm-kmod on 14.2 aarch64

csgordon, 2025-01-09:
**Describe the bug**
No version of drm-kmod builds on 14.2 for aarch64

**FreeBSD version**
FreeBSD bsd 14.2-RELEASE FreeBSD 14.2-RELEASE releng/14.2-n269506-c8918d6c7412 GENERIC arm64 1402000 1402000

**PCI Info**

<details>
It's a lot to manually re-type, this is from a VM with no graphics. I've attached a screenshot below.
[image]

</details>

**DRM KMOD version**
N/A because it doesn't build; using ports installed with 14.2 RELEASE installation

**To Reproduce**
Steps to reproduce the behavior:
1. Install 14.2-RELEASE for arm64, including ports
2. ```cd /usr/ports/graphics/drm-kmod```
3. ```make install```

The build fails with a message that drm-510-kmod is not supported on FreeBSD 14.2 or later. drm-515-kmod and drm-61-kmod fail with messages that newer KPI is not supported on aarch64.

**Screenshots**
[image]

**Additional context**
This is being attempted in part to get graphics on a VM running in UTM (a QEMU wrapper) with the virtio GPU (intending to use https://reviews.freebsd.org/D40094, which was apparently tested on arm64).

csgordon, 2025-01-10:
Yes, I tried 510 by itself; the attached screenshot uses doas, but I did it yesterday as root
[image]

Here's what happens for 61:
[image]

and 515:
[image]

csgordon, 2025-01-14:
The GENERIC kernel already includes all of the virtio drivers (it's in the config, and kldload confirms that it's already built-in to the kernel in question). (Though I hand't seen that article before, and it's useful to me in another context, so thanks!)

In any case, if the issue were a missing driver I'd expect to encounter a later build error or a complaint when loading a successfully-compiled kernel module, but the blocker here is that the port just doesn't appear to support the latest stable release on arm64.

dsalychev, 2025-02-20:
This situation with drm-kmod drivers puzzles me as well. This is a workaround of mine to build drm-510-kmod on Honeycomb LX2 running -CURRENT and GENERIC-NODEBUG:
```
diff --git a/graphics/drm-510-kmod/Makefile b/graphics/drm-510-kmod/Makefile
index 048c913ab..4ca57db56 100644
--- a/graphics/drm-510-kmod/Makefile
+++ b/graphics/drm-510-kmod/Makefile
@@ -30,10 +30,10 @@ GH_TAGNAME=	${DRM_KMOD_GH_TAGNAME}
 SUB_FILES=	pkg-message
 SUB_LIST=	OPSYS=${OPSYS} OSREL=${OSREL}
 
-.if ${OPSYS} == FreeBSD && ${OSVERSION} >= 1401501
-IGNORE=		not supported on FreeBSD 14.2 and higher
-.endif
-IGNORE_FreeBSD_15=	not supported anymore
+#.if ${OPSYS} == FreeBSD && ${OSVERSION} >= 1401501
+#IGNORE=		not supported on FreeBSD 14.2 and higher
+#.endif
+#IGNORE_FreeBSD_15=	not supported anymore
 
 .if ${OPSYS} != FreeBSD
 IGNORE=		not supported on anything but FreeBSD (missing linuxkpi functionality)
diff --git a/graphics/drm-510-kmod/files/patch-drivers_gpu_drm_amd_display_dc_core_dc.c b/graphics/drm-510-kmod/files/patch-drivers_gpu_drm_amd_display_dc_core_dc.c
new file mode 100644
index 000000000..2c3290eab
--- /dev/null
+++ b/graphics/drm-510-kmod/files/patch-drivers_gpu_drm_amd_display_dc_core_dc.c
@@ -0,0 +1,28 @@
+--- drivers/gpu/drm/amd/display/dc/core/dc.c      2025-02-11 13:24:04.722048000 +0100
++++ drivers/gpu/drm/amd/display/dc/core/dc.c	  2025-02-11 13:24:12.993431000 +0100
+@@ -731,23 +731,23 @@
+ 	dc->clk_mgr = dc_clk_mgr_create(dc->ctx, dc->res_pool->pp_smu, dc->res_pool->dccg);
+ 	if (!dc->clk_mgr)
+ 		goto fail;
+ #ifdef CONFIG_DRM_AMD_DC_DCN3_0
+ 	dc->clk_mgr->force_smu_not_present = init_params->force_smu_not_present;
+ #endif
+ 
+ 	dc->debug.force_ignore_link_settings = init_params->force_ignore_link_settings;
+ 
+ 	if (dc->res_pool->funcs->update_bw_bounding_box) {
+-		DC_FP_START();
++		//DC_FP_START();
+ 		dc->res_pool->funcs->update_bw_bounding_box(dc, dc->clk_mgr->bw_params);
+-		DC_FP_END();
++		//DC_FP_END();
+ 	}
+ 
+ 	/* Creation of current_state must occur after dc->dml
+ 	 * is initialized in dc_create_resource_pool because
+ 	 * on creation it copies the contents of dc->dml
+ 	 */
+ 
+ 	dc->current_state = dc_create_state(dc);
+ 
+ 	if (!dc->current_state) {
```