freebsd/drm-kmod #343: amdgpu: kernel panic on load for 890M

vrachnis, 2025-02-21:
**Describe the bug**
I recently got my hands on an AMD Ryzen AI 9 HX 370 laptop and tried to see if FreeBSD works on it. At the moment I'm having problems with a reproducible kernel panic when running `kldload amdgpu` on CURRENT. I'm guessing that I need a newer version to make this work, but I tried the 6.6 branch without any success. Is there any chance of getting this to work under FreeBSD?

**FreeBSD version**
Paste the output of `uname -aKU`
```
FreeBSD gp4 15.0-CURRENT FreeBSD 15.0-CURRENT main-n275584-21b5b8b38b1c GENERIC amd64 1500033 1500033
```

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1507 subvendor=0x1022 subdevice=0x1507
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1508 subvendor=0x1022 subdevice=0x1508
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150a subvendor=0x1022 subdevice=0x150a
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo PCIe USB4 Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150a subvendor=0x1022 subdevice=0x150a
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo PCIe USB4 Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:2:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
[... 170 lines cut ...]
amdgpu_device_fini_hw() at amdgpu_device_fini_hw+0x190/frame 0xfffffe01d54d3870
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0x84/frame 0xfffffe01d54d38a0
amdgpu_pci_probe() at amdgpu_pci_probe+0x290/frame 0xfffffe01d54d38e0
linux_pci_attach_device() at linux_pci_attach_device+0x430/frame 0xfffffe01d54d3930
device_attach() at device_attach+0x45b/frame 0xfffffe01d543980
bus_generic_driver_added() at bus_generic_driver_added+0x90/frame 0xfffffe01d54d39a0
devclass_driver_added() at devclass_driver_added+0x29/frame 0xfffffe01d54d39d0
devclass_add_driver() at devclass_add_driver+0x138/frame 0xfffffe01d54d3a10
_linux_pci_register_driver() at _linux_pci_register_driver+0xc1/frame 0xfffffe01d54d3a40
amdgpu_evh() at amdgpu_evh+0x80/frame 0xfffffe01d54d3a50
module_register_init() at module_register_init+0xb0/frame 0xfffffe01d54d3a80
linker_load_module() at linker_load_module+0xc51/frame 0xfffffe01d54d3d80
kern_kldload() at kern_kldload+0x16e/frame 0xfffffe01d54d3dd0
sys_kldload() at sys_kldload+0x59/frame 0xfffffe01d54d3e00
amd64_syscall() at amd64_syscall+0x15a/frame 0xfffffe01d54d3f30
fast_syscall_common() at fast_syscall_common+0xf8/frame 0xfffffe01d54d3f30
--- syscall (304, FreeBSD ELF64, kldload), rip = 0x179116cecada, rsp = 0x17911418ba78, rbp = 0x17911418bff0 ---
KDB: enter: panic
[ thread pid 3578 tid 100922 ]
Stopped at      kdb_enter+0x33: movq    $0,0x10520b2(%rip)
db>
```

**Additional context**
Add any other context about the problem here.

kenrap, 2025-02-22:
Since your installed drm-61-kmod package has the `__FreeBSD_version` suffix at `1500032` and your CURRENT's kernel version is at `1500033`, I would suggest rebuilding the drm-61-kmod package through ports to match your kernel's API. Or wait for an updated package with version at the `1500033` version suffix.

vrachnis, 2025-02-23:
Ah, good catch! I hadn't realized that the package was outdated. I did just rebuild graphics/drm-61-kmod from ports, but unfortunately it has the exact same behaviour - it complains that `Start of segment isn't less than end`.

Is there anything else that I could try to get this working?

kenrap, 2025-02-24:
Thanks for trying and confirming that.

So, rather than me guessing a solution, I decided lookup about your integrated GPU. Since it's based on RDNA 3.5, AMD had rolled out support for that in Linux 6.7. So, you'd want to subscribe to #332 and show your love for it. 👍

vrachnis, 2025-02-24:
Aha! that's exactly what I'm looking for. I suspected that it's a case of needing a newer version. I'll monitor #332 for updates. Thanks!