freebsd/drm-kmod #360: Arrow Lake processor cannot used in drm-kmod

takawata, 2025-06-21:
**Describe the bug**
Arrow Lake is used in ThinkPad X13 Gen6 and cannot used in FreeBSD drm-kmod.

**FreeBSD version**
FreeBSD shigure 15.0-CURRENT FreeBSD 15.0-CURRENT #0 takawata-wip-n277953-665f8fbadd17: Sat Jun 14 19:32:55 JST 2025     takawata@akatsuki.init-main.com:/usr/obj/usr/home/takawata/projects/freebsd/src/amd64.amd64/sys/GENERIC amd64

**PCI Info**

<details>
  <summary vgapci0@pci0:0:2:0:     class=0x030000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x7d41 subvendor=0x17aa subdevice=0x2344
    vendor     = 'Intel Corporation'
    device     = 'Arrow Lake-U [Intel Graphics]'
    class      = display
    subclass   = VGA</summary>
  

</details>
github head as of today

**To Reproduce**
loading drm.ko　and i915kms.ko

**Additional context**
It seems 2 "!f" s related to Arrow Lake U processor in Linux stable as of today. Any plan to update ?

takawata, 2025-06-21:
PCI info is as follows: 
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x7d41 subvendor=0x17aa subdevice=0x2344
    vendor     = 'Intel Corporation'
    device     = 'Arrow Lake-U [Intel Graphics]'
    class      = display
    subclass   = VGA

centromere, 2025-06-24:
[Here](https://bsd-hardware.info/?probe=fd3536cb97) is a hardware probe of a machine with an Arrow Lake processor.