freebsd/drm-kmod #366: amdgpu: hangs on drm-61

orsolic, 2025-07-27:
**Describe the bug**
A clear and concise description of what the bug is.

**FreeBSD version**
FreeBSD zen-pobro 14.3-RELEASE-p1 FreeBSD 14.3-RELEASE-p1 my14.3-zen-n271436-d39eb9efea60 GENERIC amd64 1403000 1403000
(same issue with kernel from freebsd-update)

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
amdviiommu0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1633 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib13@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb4@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1448 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1449 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Device 24: Function 1'
[... 150 lines cut ...]
xhci2@pci0:13:0:4:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1639 subvendor=0x1043 subdevice=0x87e1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne USB 3.1'
    class      = serial bus
    subclass   = USB
hdac1@pci0:13:0:6:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x1043 subdevice=0x8797
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h/19h/1ah HD Audio Controller'
    class      = multimedia
    subclass   = HDA

</details>

**DRM KMOD version**
drm-61-kmod 6.1.128.1403000_5

**To Reproduce**
Use AMD iGPU and wait? Sometimes triggered with mpv, KiCAD or WINE. Playing music in YouTube in Firefox doesn't have affect.

**logs**
<details>
</details>

<details>
</details>

abishai, 2025-09-09:
I tried reproduction steps and my GPU crashed almost immediately after I put FF in full screen.
I use HDMI.
<details>
<summary>logs</summary>
Sep 10 00:28:30 moonstar kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=48455092, emitted seq=48455094
Sep 10 00:28:30 moonstar kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 320633 thread  pid 320633
Sep 10 00:28:30 moonstar kernel: drmn0: GPU reset begin!
Sep 10 00:28:30 moonstar kernel: drmn0: free PSP TMR buffer
Sep 10 00:28:30 moonstar kernel: drmn0: MODE2 reset
Sep 10 00:28:30 moonstar kernel: drmn0: GPU reset succeeded, trying to resume
Sep 10 00:28:30 moonstar kernel: [drm] PCIE GART of 1024M enabled.
Sep 10 00:28:30 moonstar kernel: [drm] PTB located at 0x000000F43FC00000
Sep 10 00:28:30 moonstar kernel: [drm] PSP is resuming...
Sep 10 00:28:30 moonstar kernel: [drm] reserve 0x400000 from 0xf43f800000 for PSP TMR
Sep 10 00:28:30 moonstar kernel: drmn0: RAS: optional ras ta ucode is not available
Sep 10 00:28:30 moonstar kernel: drmn0: RAP: optional rap ta ucode is not available
Sep 10 00:28:30 moonstar kernel: drmn0: SECUREDISPLAY: securedisplay ta ucode is not available
Sep 10 00:28:30 moonstar kernel: drmn0: SMU is resuming...
Sep 10 00:28:30 moonstar kernel: drmn0: SMU is resumed successfully!
Sep 10 00:28:30 moonstar kernel: [drm] DMUB hardware initialized: version=0x01010027
Sep 10 00:28:30 moonstar kernel: [drm] kiq ring mec 2 pipe 1 q 0
Sep 10 00:28:30 moonstar kernel: [drm] VCN decode and encode initialized successfully(under DPG Mode).
Sep 10 00:28:30 moonstar kernel: [drm] JPEG decode initialized successfully.
Sep 10 00:28:30 moonstar kernel: drmn0: ring gfx uses VM inv eng 0 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
Sep 10 00:28:30 moonstar kernel: drmn0: ring sdma0 uses VM inv eng 0 on hub 1
Sep 10 00:28:30 moonstar kernel: drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
Sep 10 00:28:30 moonstar kernel: drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
Sep 10 00:28:30 moonstar kernel: drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
Sep 10 00:28:30 moonstar kernel: drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
Sep 10 00:28:30 moonstar kernel: drmn0: recover vram bo from shadow start
Sep 10 00:28:30 moonstar kernel: drmn0: recover vram bo from shadow done
Sep 10 00:28:30 moonstar kernel: drmn0: GPU reset(2) succeeded!
Sep 10 00:28:30 moonstar kernel: [drm ERROR :amdgpu_cs_ioctl] Failed to initialize parser -85!
Sep 10 00:28:41 moonstar kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=48455095, emitted seq=48455095
Sep 10 00:28:41 moonstar kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 320633 thread  pid 320633
Sep 10 00:28:41 moonstar kernel: drmn0: GPU reset begin!
</details>

<details>
<summary>pciconf</summary>
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1022 subdevice=0x1630
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1022 subdevice=0x1631
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
[... 93 lines cut ...]
    subclass   = VGA
hdac0@pci0:4:0:1:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0x1637 subvendor=0x1002 subdevice=0x1637
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Renoir/Cezanne HDMI/DP Audio Controller'
    class      = multimedia
    subclass   = HDA
none2@pci0:4:0:2:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15df subvendor=0x1022 subdevice=0x15df
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2/FireFlight/Renoir/Cezanne Platform Security Processor'
    class      = encrypt/decrypt
xhci0@pci0:4:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1639 subvendor=0x1022 subdevice=0x1639
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
[... cut ...]
xhci1@pci0:4:0:4:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1639 subvendor=0x1022 subdevice=0x1639
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne USB 3.1'
    class      = serial bus
    subclass   = USB
none3@pci0:4:0:5:	class=0x048000 rev=0x01 hdr=0x00 vendor=0x1022 device=0x15e2 subvendor=0x1022 subdevice=0x15e2
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Audio Coprocessor'
    class      = multimedia
</details>

abishai, 2025-09-13:
It's Renoir (GCN 5.0) and, finally it crashed with and without loader.conf tunables. Doesn't look like working workaround.

orsolic, 2025-09-15:
Reproduce it again on 15.0-ALPHA1 with drm 6.1. Will try with 6.6.
My slowdowns are possible storage related, I'll write another ticket for that.

orsolic, 2025-09-17:
It seems that my setup needs some uptime (~2 days) for bug to be reproducible.
After fresh reboot, 15-ALPHA1 with drm-61 I am not able to reproduce it (with mpv and Firefox). But, day later, I've started FreeCAD, did some work and it crashed.

<details>
<summary>logs</summary>

Sep 17 17:42:12 zen-pobro kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=11702979, emitted seq=11702981
Sep 17 17:42:12 zen-pobro kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 852494 thread  pid 852494
Sep 17 17:42:12 zen-pobro kernel: drmn0: GPU reset begin!
Sep 17 17:42:12 zen-pobro kernel: [drm] psp gfx command UNLOAD_TA(0x2) failed and response status is (0x117)
Sep 17 17:42:12 zen-pobro kernel: drmn0: free PSP TMR buffer
Sep 17 17:42:12 zen-pobro kernel: drmn0: MODE2 reset
Sep 17 17:42:12 zen-pobro kernel: drmn0: GPU reset succeeded, trying to resume
Sep 17 17:42:12 zen-pobro kernel: [drm] PCIE GART of 1024M enabled.
Sep 17 17:42:12 zen-pobro kernel: [drm] PTB located at 0x000000F4FFC00000
Sep 17 17:42:12 zen-pobro kernel: [drm] PSP is resuming...
Sep 17 17:42:12 zen-pobro kernel: [drm] reserve 0x400000 from 0xf4ff800000 for PSP TMR
Sep 17 17:42:13 zen-pobro kernel: drmn0: RAS: optional ras ta ucode is not available
Sep 17 17:42:13 zen-pobro kernel: drmn0: RAP: optional rap ta ucode is not available
Sep 17 17:42:13 zen-pobro kernel: drmn0: SECUREDISPLAY: securedisplay ta ucode is not available
Sep 17 17:42:13 zen-pobro kernel: drmn0: SMU is resuming...
Sep 17 17:42:13 zen-pobro kernel: drmn0: SMU is resumed successfully!
Sep 17 17:42:13 zen-pobro kernel: [drm] DMUB hardware initialized: version=0x01010027
Sep 17 17:42:13 zen-pobro kernel: [drm] kiq ring mec 2 pipe 1 q 0
Sep 17 17:42:13 zen-pobro kernel: [drm] VCN decode and encode initialized successfully(under DPG Mode).
Sep 17 17:42:13 zen-pobro kernel: [drm] JPEG decode initialized successfully.
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring gfx uses VM inv eng 0 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring sdma0 uses VM inv eng 0 on hub 1
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
Sep 17 17:42:13 zen-pobro kernel: drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
Sep 17 17:42:13 zen-pobro kernel: drmn0: recover vram bo from shadow start
Sep 17 17:42:13 zen-pobro kernel: drmn0: recover vram bo from shadow done
Sep 17 17:42:13 zen-pobro kernel: drmn0: GPU reset(2) succeeded!
Sep 17 17:42:13 zen-pobro kernel: [drm ERROR :amdgpu_cs_ioctl] Failed to initialize parser -85!
Sep 17 17:42:23 zen-pobro kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=11702984, emitted seq=11702984
Sep 17 17:42:23 zen-pobro kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 852494 thread  pid 852494
Sep 17 17:42:23 zen-pobro kernel: drmn0: GPU reset begin!
</details>

After reboot, now 15-ALPHA1 with drm-66 same thing - not possible to easily reproduce it with Firefox and mpv, but it will probably crash in a day or two

tm512, 2025-09-18:
So it does look like on your system it's dedicating 4GB of system memory to the GPU, vs. 2GB on mine. No idea if that has anything to do with the difference in how this manifests. I suppose it's also possible that despite both being "Vega 8" iGPUs, the actual GPU was revised between Picasso (my 3500U) and your Renoir. I doubt your UEFI updates are related to this, since this is happening on 4 completely different machines with iGPUs from 2019-2020ish.

I might play around with the modesetting driver, even though I'd have to start using picom again, but at this point, I'm kinda just hoping that someone more knowledgeable can chime in on what else we could do to actually track this down. I'm not sure if `git bisect` would be practical here, at least for me. Even if bisecting on the master branch would cover all of the relevant commits, the kernel panics I get with the drm-515 code would be a confounding factor (maybe I should get around to actually submitting a separate bug report on that).

abishai, 2025-09-19:
I use xOrg without graphics related configs, i3wm and picom, amdgpu is loaded as kernel driver in rc.conf, just like in manuals.
```
[    32.290] (II) Loading /usr/local/lib/xorg/modules/libglamoregl.so
[    32.342] (II) Module glamoregl: vendor="X.Org Foundation"
[    32.342]    compiled for 1.21.1.18, module version = 1.0.1
[    32.342]    ABI class: X.Org ANSI C Emulation, version 0.4
[    32.446] (II) AMDGPU(0): glamor X acceleration enabled on AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.7, DRM 3.49, 14.3-RELEASE)
[    32.446] (II) AMDGPU(0): glamor detected, initialising EGL layer.
[    32.446] (==) AMDGPU(0): TearFree property default: auto
```

```
abishai@moonstar:~ % glxinfo -B
name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.7, DRM 3.49, 14.3-RELEASE) (0x15e7)
    Version: 24.1.7
    Accelerated: yes
    Video memory: 1024MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 177 MB, largest block: 177 MB
    VBO free aux. memory - total: 15065 MB, largest block: 15065 MB
    Texture free memory - total: 177 MB, largest block: 177 MB
    Texture free aux. memory - total: 15065 MB, largest block: 15065 MB
    Renderbuffer free memory - total: 177 MB, largest block: 177 MB
    Renderbuffer free aux. memory - total: 15065 MB, largest block: 15065 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 1024 MB
    Total available memory: 16825 MB
    Currently available dedicated video memory: 177 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.7, DRM 3.49, 14.3-RELEASE)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.1.7
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.1.7
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 24.1.7
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

tm512, 2025-09-21:
So I did some testing using the modesetting DDX, instead of amdgpu. Despite initially getting my hopes up, it did eventually hang (when using mpv with Firefox open), though the logs are quite a lot different, with "Illegal register access in command stream" and then a page fault during the GPU reset:

<details>
<summary>log</summary>
<pre>
Sep 21 01:25:14 merak kernel: [drm ERROR :gfx_v9_0_priv_reg_irq] Illegal register access in command stream
Sep 21 01:25:14 merak kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=957460, emitted seq=957462
Sep 21 01:25:14 merak kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 129855 thread  pid 129855
Sep 21 01:25:14 merak kernel: drmn0: GPU reset begin!
Sep 21 01:25:14 merak kernel: [drm ERROR :amdgpu_cs_ioctl] Failed to initialize parser -85!
Sep 21 01:25:14 merak kernel: drmn0: free PSP TMR buffer
Sep 21 01:25:14 merak kernel: drmn0: MODE2 reset
Sep 21 01:25:14 merak kernel: drmn0: GPU reset succeeded, trying to resume
Sep 21 01:25:14 merak kernel: [drm] PCIE GART of 1024M enabled.
Sep 21 01:25:14 merak kernel: [drm] PTB located at 0x000000F400A00000
Sep 21 01:25:14 merak kernel: [drm] PSP is resuming...
Sep 21 01:25:14 merak kernel: [drm] reserve 0x400000 from 0xf47fc00000 for PSP TMR
Sep 21 01:25:14 merak kernel: drmn0: RAS: optional ras ta ucode is not available
Sep 21 01:25:15 merak kernel: drmn0: RAP: optional rap ta ucode is not available
Sep 21 01:25:15 merak kernel: [drm] kiq ring mec 2 pipe 1 q 0
Sep 21 01:25:15 merak kernel: [drm] VCN decode and encode initialized successfully(under SPG Mode).
Sep 21 01:25:15 merak kernel: drmn0: ring gfx uses VM inv eng 0 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
Sep 21 01:25:15 merak kernel: drmn0: ring sdma0 uses VM inv eng 0 on hub 1
Sep 21 01:25:15 merak kernel: drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
Sep 21 01:25:15 merak kernel: drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
Sep 21 01:25:15 merak kernel: drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
Sep 21 01:25:15 merak kernel: drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
Sep 21 01:25:15 merak kernel: drmn0: [gfxhub0] no-retry page fault (src_id:0 ring:172 vmid:4 pasid:32774, for process  pid 129855 thread  pid 129855)
Sep 21 01:25:15 merak kernel: drmn0:   in page starting at address 0x0000800100017000 from IH client 0x1b (UTCL2)
Sep 21 01:25:15 merak kernel: drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00440B59
Sep 21 01:25:15 merak kernel: drmn0: 	 Faulty UTCL2 client ID: CPC (0x5)
Sep 21 01:25:15 merak kernel: drmn0: 	 MORE_FAULTS: 0x1
Sep 21 01:25:15 merak kernel: drmn0: 	 WALKER_ERROR: 0x4
Sep 21 01:25:15 merak kernel: drmn0: 	 PERMISSION_FAULTS: 0x5
Sep 21 01:25:15 merak kernel: drmn0: 	 MAPPING_ERROR: 0x1
Sep 21 01:25:15 merak kernel: drmn0: 	 RW: 0x1
Sep 21 01:25:15 merak kernel: drmn0: recover vram bo from shadow start
Sep 21 01:25:15 merak kernel: drmn0: recover vram bo from shadow done
Sep 21 01:25:15 merak kernel: drmn0: GPU reset(2) succeeded!
Sep 21 01:25:25 merak kernel: [drm ERROR :amdgpu_job_timedout] ring gfx timeout, signaled seq=957463, emitted seq=957463
Sep 21 01:25:25 merak kernel: [drm ERROR :amdgpu_job_timedout] Process information: process  pid 129855 thread  pid 129855
Sep 21 01:25:25 merak kernel: drmn0: GPU reset begin!
</pre>
</details>

darbsh, 2025-10-16:
> If those updates do fix this particular issue, then the update is incomplete, because the referenced commit doesn't touch the firmware that my iGPU uses (picasso_vcn.bin). I'm skeptical that it's related to this issue in the first place, though. The dmesg logs attached on the amdgpu gitlab say `ring vcn_dec timeout` before the hang, not `ring gfx timeout` as in this issue. I also seem to recall testing mpv with VAAPI decoding disabled, and the hangs still occurred, though I would have to test again to be sure.

picasso_vcn.bin hasn't been updated in the linux firmware repo since 2023, so who knows if they're still dealing with it. There are specifically 4 firmware commits for AMD's Picasso GPU that FreeBSD's repo doesn't have

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/?qt=grep&q=picasso   

Have you tried merging the linux firmware repo with what FreeBSD has to see if it fixes your issue? 

I did, and fixed my 7950x3d igpu issue, however my PR was left to wither on the vine.

tm512, 2025-10-24:
Using the firmware from https://github.com/freebsd/drm-kmod-firmware/pull/38, which appears to be recent enough to have all of the updates for picasso, and it does not seem to help at all on my end. Same `ring gfx timeout` issue.

orsolic, 2026-01-18:
This is still an issue on 15.0-RELEASE with drm-66-kmod-6.6.25.1500068_8 and drm-latest-kmod-6.9.1500068_1 - multiple freezes per hour are not uncommon.

tm512, 2026-04-05:
Sorry it took me a while to report back about the `git bisect` process. I confirmed that 6.9 hangs on CURRENT, started the bisection, but apparently I made a mistake by upgrading non-base packages during that process. Every single commit it was having me test failed to hang, even as commits that touched anything amdgpu-related dwindled. Started to feel like I was going crazy, I went back to 6.9 to test it again, and it also fails to hang. 15-STABLE with 6.9 (haven't tested other versions yet) is working fine for me now as well.

I'm not sure which package update is the deciding variable here, I initially suspected mpv and downgraded it, to no effect. Maybe ffmpeg, if something it was doing with VAAPI triggered the hangs in the gfx ring, though I thought I reproduced the hangs even after disabling hardware decoding. I'm not sure how much effort I actually want to put into tracking it down at this point. Are the hangs still present (on an up-to-date install) for anyone here?

orsolic, 2026-06-01:
Reproducible on 15.1-RC2, drm-latest-kmod-6.12.1501000. System hangs for a moment, but system and GUI survives.

```
Jun  1 07:51:56 zen-pobro kernel: [drm ERROR :gfx_v9_0_priv_reg_irq] Illegal register access in command stream
Jun  1 07:51:56 zen-pobro kernel: drmn0: Dumping IP State
Jun  1 07:51:56 zen-pobro kernel: drmn0: Dumping IP State Completed
Jun  1 07:51:56 zen-pobro kernel: drmn0: ring gfx timeout, signaled seq=14416844, emitted seq=14416847
Jun  1 07:51:56 zen-pobro kernel: drmn0: Process information: process  pid 167175 thread  pid 167175
Jun  1 07:51:56 zen-pobro kernel: drmn0: GPU reset begin!
Jun  1 07:51:56 zen-pobro kernel: [drm] *ERROR* 
Jun  1 07:51:56 zen-pobro kernel: drmn0: psp gfx command UNLOAD_TA(0x2) failed and response status is (0x117)
Jun  1 07:51:56 zen-pobro kernel: drmn0: MODE2 reset
Jun  1 07:51:56 zen-pobro kernel: drmn0: GPU reset succeeded, trying to resume
Jun  1 07:51:56 zen-pobro kernel: [drm] PCIE GART of 1024M enabled.
Jun  1 07:51:56 zen-pobro kernel: [drm] PTB located at 0x000000F4FFC00000
Jun  1 07:51:56 zen-pobro kernel: drmn0: PSP is resuming...
Jun  1 07:51:56 zen-pobro kernel: drmn0: reserve 0x400000 from 0xf4ff800000 for PSP TMR
Jun  1 07:51:57 zen-pobro kernel: drmn0: RAS: optional ras ta ucode is not available
Jun  1 07:51:57 zen-pobro kernel: drmn0: RAP: optional rap ta ucode is not available
Jun  1 07:51:57 zen-pobro kernel: drmn0: SECUREDISPLAY: securedisplay ta ucode is not available
Jun  1 07:51:57 zen-pobro kernel: drmn0: SMU is resuming...
Jun  1 07:51:57 zen-pobro kernel: drmn0: SMU is resumed successfully!
Jun  1 07:51:57 zen-pobro kernel: [drm] DMUB hardware initialized: version=0x0101002B
Jun  1 07:51:57 zen-pobro kernel: [drm] kiq ring mec 2 pipe 1 q 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring gfx uses VM inv eng 0 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring kiq_0.2.1.0 uses VM inv eng 11 on hub 0
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring sdma0 uses VM inv eng 0 on hub 8
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring vcn_dec uses VM inv eng 1 on hub 8
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 8
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 8
Jun  1 07:51:57 zen-pobro kernel: drmn0: ring jpeg_dec uses VM inv eng 6 on hub 8
Jun  1 07:51:57 zen-pobro kernel: drmn0: GPU reset(2) succeeded!
Jun  1 07:51:57 zen-pobro kernel: drmn0: [gfxhub0] no-retry page fault (src_id:0 ring:40 vmid:0 pasid:0)
Jun  1 07:51:57 zen-pobro kernel: drmn0:   in page starting at address 0x0000000000679000 from IH client 0x1b (UTCL2)
Jun  1 07:51:57 zen-pobro kernel: drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00040A51
Jun  1 07:51:57 zen-pobro kernel: drmn0:         Faulty UTCL2 client ID: CPC (0x5)
Jun  1 07:51:57 zen-pobro kernel: drmn0:         MORE_FAULTS: 0x1
Jun  1 07:51:57 zen-pobro kernel: drmn0:         WALKER_ERROR: 0x0
Jun  1 07:51:57 zen-pobro kernel: drmn0:         PERMISSION_FAULTS: 0x5
Jun  1 07:51:57 zen-pobro kernel: drmn0:         MAPPING_ERROR: 0x0
Jun  1 07:51:57 zen-pobro kernel: drmn0:         RW: 0x1
Jun  1 07:51:57 zen-pobro kernel: [drm ERROR :gfx_v9_0_bad_op_irq] Illegal opcode in command stream
```