freebsd/drm-kmod #372: amdgpu 890M panic: Unregistered use of FPU in kernel

itglob, 2025-08-10:
**Describe the bug**
Panics when loading amdgpu.ko

**FreeBSD version**
FreeBSD fbsd-x1-tst 15.0-PRERELEASE FreeBSD 15.0-PRERELEASE #0 main-084a3e1e0337: Sun Aug 10 23:29:35 MSK 2025     root@fbsd-x1-tst:/usr/obj/usr/src/amd64.amd64/sys/KERNEL_X1_370 amd64 1500059 1500059

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1507 subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1508 subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150a subvendor=0x1022 subdevice=0x150a
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo PCIe USB4 Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150a subvendor=0x1022 subdevice=0x150a
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo PCIe USB4 Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:2:5:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150b subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1509 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib6@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x150c subvendor=0x1f4c subdevice=0xb022
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Strix/Strix Halo Internal GPP Bridge to Bus [C:A]'
    class      = bridge
[... 260 lines cut ...]
dml2_init() at dml2_init+0x7f/frame 0xfffffe0255b92488
dml2_create() at dml2_create+0x3a/frame 0xfffffe0255b924b0
dc_state_create() at dc_state_create+0x7f/frame 0xfffffe0255b924d0
dc_create() at dc_create+0x419/frame 0xfffffe0255b92510
dm_hw_init() at dm_hw_init+0x3b2/frame 0xfffffe0255b92700
amdgpu_device_ip_hw_init_phase2() at amdgpu_device_ip_hw_init_phase2+0x5a/frame 0xfffffe0255b92730
amdgpu_device_ip_init() at amdgpu_device_ip_init+0x3e0/frame 0xfffffe0255b927a0
amdgpu_device_init() at amdgpu_device_init+0x1cc0/frame 0xfffffe0255b92860
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0x16/frame 0xfffffe0255b92890
amdgpu_pci_probe() at amdgpu_pci_probe+0x2dd/frame 0xfffffe0255b928e0
linux_pci_attach_device() at linux_pci_attach_device+0x440/frame 0xfffffe0255b92930
device_attach() at device_attach+0x45b/frame 0xfffffe0255b92980
bus_generic_driver_added() at bus_generic_driver_added+0x90/frame 0xfffffe0255b929a0
devclass_driver_added() at devclass_driver_added+0x29/frame 0xfffffe0255b929d0
devclass_add_driver() at devclass_add_driver+0x138/frame 0xfffffe0255b92a10
_linux_pci_register_driver() at _linux_pci_register_driver+0xc1/frame 0xfffffe0255b92a40
amdgpu_evh() at amdgpu_evh+0x73/frame 0xfffffe0255b92a50
module_register_init() at module_register_init+0xb0/frame 0xfffffe0255b92a80
linker_load_module() at linker_load_module+0xc56/frame 0xfffffe0255b92d80
kern_kldload() at kern_kldload+0x16e/frame 0xfffffe0255b92dd0
sys_kldload() at sys_kldload+0x59/frame 0xfffffe0255b92e00
amd64_syscall() at amd64_syscall+0x169/frame 0xfffffe0255b92f30
fast_syscall_common() at fast_syscall_common+0xf8/frame 0xfffffe0255b92f30
--- syscall (304, FreeBSD ELF64, kldload), rip = 0x2fd38c275c9a, rsp = 0x2fd38a536e68, rbp = 0x2fd38a5373e0 ---
KDB: enter: panic

chris3torek, 2025-09-17:
@ctuffli I can't speak to particular devices, but for using the drm-66-kmod module you also need to install firmware. The from-source method is to use the `graphics/gpu-firmware-amd-kmod` port (which grabs https://github.com/freebsd/drm-kmod/firmware using tag `20230625_8`, which as you can tell by the date is woefully out of date but is the latest in that repository) or using `fwget`.

I'm not sure what you mean by "on `master`" since these port repositories are meant to be used by tag (`GH_TAGNAME`).

itglob, 2025-11-22:
build drm-latest-kmod with enabled DRM_AMD_DC_FP  in KCONFIG, nothing change. Firmware latest from linux-firmware.git, freebsd-15-stable.

chris3torek, 2025-11-22:
@emaste @itglob We don't want to use FP in the kernel (because we don't enable it at the right points) and the current drivers don't anyway. The iGPU crash I got was due to bad firmware. The current freebsd firmware modules are horribly out of date. I have a pending PR that's being studiously ignored (https://github.com/freebsd/drm-kmod-firmware/pull/38).

(Or, if we _do_ want to use the FPU, we'll need to make the setup calls.)

asherikov, 2025-11-23:
1. firmware update did not help in my case (also 890M).
2. it is not stable on Linux as well https://bbs.archlinux.org/viewtopic.php?id=301260

emaste, 2025-11-24:
Oh, I didn't realize we do already set DRM_AMD_DC_FP,
```
.if ${MACHINE_CPUARCH} == "amd64"
...
KCONFIG+=       DRM_AMD_DC_FP
.endif
```

Looking in some more detail this is of course the case -- the `dml2_create` call is inside `#ifdef CONFIG_DRM_AMD_DC_FP` in `dc_state_create`.

So we do in fact need to fix up `DC_FP_START`/`END` locations, whether this is a side effect of moving drm-kmod to be based on newer Linux versions, cherry-picking specific fixes, or with bespoke changes.

itglob, 2025-11-24:
@emaste  dc_fpu_begin / end already in drivers/gpu/drm/amd/display/amdgpu_dm/dc_fpu.c  and use our lkpi_kernel_fpu_begin / end